adafruit/circuitpython #6024: Need Support Board LILYGO® TTGO T-Display RP2040 Raspberry Pi Modul 1,14 Zoll LCD

egs2003, 2022-02-12:
https://de.banggood.com/LILYGO-TTGO-T-Display-RP2040-Raspberry-Pi-Module-1_14-inch-LCD-Development-Board-p-1915490.html?rmmds=myorder&cur_warehouse=CN

ladyada, 2022-02-12:
we don't own one, and a PR to add support would be welcome!
https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

egs2003, 2022-02-14:
Here is the complete documantation with shematics

https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040

erongd, 2022-02-15:
I got the board working with circuitpython. Would follow the guide to create a pull request.

erongd, 2022-02-15:
Have created PR #6037 for initial support. Need help with assigning a PID, I believe the VID is the Lilygo one as reference to the existing ESP32S2 based board.

todbot, 2022-06-23:
If anyone else has this board and wants to get it working before the PR is merged, you can use the standard Pico firmware and the `adafruit_st7789` library.  Here is a gist with a small demo: https://gist.github.com/todbot/a23c8cb28e5925dfa79e6dba5c9d0962

dhalbert, 2023-10-28:
Exists: https://circuitpython.org/board/lilygo_t_display_rp2040/

closed 2023-10-28 by dhalbert