adafruit/circuitpython #6025: print bytearray output garbled and incomplete

CanyonCasa, 2022-02-13:
### CircuitPython version

```python
CircuitPython 7.1.1 on a QT Py RP2040.
```

### Code/REPL

```python
# Write your code here :-)
import board
from adafruit_onewire.bus import OneWireBus
ow_bus = OneWireBus(board.D2)
devices = ow_bus.scan()
for d in devices:
    print("ROM={}\tFamily=0x{:02x}".format(d.rom, d.family_code))
    print("Correct ROM[{}]:".format(len(d.rom)), list(map(lambda x: "0x{:02x}".format(x),d.rom)))
```

### Behavior

I copied the OneWire example directly from (https://github.com/adafruit/Adafruit_CircuitPython_OneWire) as above...

I added the last line because the rom value was not being reported correctly. Based on the serial output below, it clearly shows that the bytearray contains the correct rom serial number but the print statement outputs it incorrectly, garbled and incomplete.

`
code.py output:
ROM=bytearray(b'(\xcd>\x16\xa8\x01<\xd6')	Family=0x28
Correct ROM[8]: ['0x28', '0xcd', '0x3e', '0x16', '0xa8', '0x01', '0x3c', '0xd6']
ROM=bytearray(b'(TE\xff\x00\x00\x00<')	Family=0x28
Correct ROM[8]: ['0x28', '0x54', '0x45', '0xff', '0x00', '0x00', '0x00', '0x3c']

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit QT Py RP2040 with rp2040
`

### Additional information

Same behavior directly from REPL...

>>> b = bytearray([0x28, 0x54, 0x45, 0xff, 0x00, 0x00, 0x00, 0x3c])
>>> print(b)

bytearray(b'(TE\xff\x00\x00\x00<')

ladyada, 2022-02-13:
what do you mean by 'garbled'? note that the print output will convert valid ascii chars to printable characters. you dont get hex data if its valid ascii

jepler, 2022-02-13:
For comparison, here's how the same data displays on a host computer:
```python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> b = bytearray([0x28, 0x54, 0x45, 0xff, 0x00, 0x00, 0x00, 0x3c])
>>> print(b)
bytearray(b'(TE\xff\x00\x00\x00<')
```

CanyonCasa, 2022-02-13:
My bad. I tried a separate sequence on my host that spit out all hex and thought that was the intended output form. I see now that it's a mix of ascii and hex.