adafruit/circuitpython #6036: SCK does not remain idle high with SPI polarity=1

sgauche, 2022-02-15:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-344-ga8d865187-dirty

Custom Board with STM32F412CG
```

### Code/REPL

```python
import time
import board
import digitalio
import busio
from adafruit_bus_device.spi_device import SPIDevice

spi_sns_mag_cs = digitalio.DigitalInOut(board.SPI_SNS_MAG_CS)
spi_sns_mag_cs.direction = digitalio.Direction.OUTPUT
spi_sns_mag_cs.value = True

spi_sns = busio.SPI(board.SPI_SNS_SCK, MOSI=board.SPI_SNS_BIDI, MISO=None, half_duplex=True)
mag_sns = SPIDevice(spi_sns, spi_sns_mag_cs, baudrate=781250, polarity=1, phase=1)

while True:
    time.sleep(5.0)

    with mag_sns as spi:
            spi.write(bytearray([0x4F]))
            result = bytearray(1)
            spi.readinto(result)
    
    print(result)
```

### Behavior

In the scope capture below, SCK Is yellow and chip select is blue.  The SCK should be idle high before chip select goes active low. The SCK signal stays idle high in between the transactions, which is the correct behavior. After the 2nd byte, it looks like the SCK pin is set back to an input and the signal starts to decay.  It should remain high until chip select goes high again.
[image]

Because SCK is not already idle high on the first byte, it goes to high due to polarity=1 after the chip select is active low and creates 9 rising edges of the clock on the first byte.
[image]

sgauche, 2022-02-15:
I'm thinking this has to do with the `__enter__` and `__exit__` for the context manager and using `with mag_sns as spi:`.  Possibly busio is deinitializing. Then when `with` for adafruit_bus_device, it sets chip select, and then busio isn't reinitialized until after? And then vice versa for the end of the transaction.

jepler, 2022-02-15:
Let's (try to) eliminate `SPIDevice` as it relates to the problem.

Here's what SPIDevice's `__enter__` should be doing, [except written in pure Python](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/main/adafruit_bus_device/spi_device.py#L93):
```python
    def __enter__(self) -> SPI:
        while not self.spi.try_lock():
            pass
        self.spi.configure(
            baudrate=self.baudrate, polarity=self.polarity, phase=self.phase
        )
        if self.chip_select:
            self.chip_select.value = self.cs_active_value
        return self.spi
```

so if you change your program to:
```python
spi_sns = busio.SPI(board.SPI_SNS_SCK, MOSI=board.SPI_SNS_BIDI, MISO=None, half_duplex=True)
if not spi_sns.try_lock: raise RuntimeError("could not lock SPI bus")
spi_sns.configure(baudrate=781250, polarity=1, phase=1)
while True:
    time.sleep(1)
```
I believe that in a correct implementation of `busio.SPI`, the SCK output should be in *output* mode for the whole lifetime of the object. (and, when not actively clocked, driven according to the last configured polarity value) However, in your scope traces, it appears to be un-driven, or only temporarily driven, because of how the voltage on it slowly disappears at the end of the SPI write.

If that example alone doesn't reproduce a problem, then try adding the read/write part, but all without introducing SPIDevice:
```python
while True:
    time.sleep(1)
    spi.write(bytearray([0x4F]))
    result = bytearray(1)
    spi.readinto(result)
```
or similar. It may be that it's after the `readinto` operation that the required driving state of the SCK pin is being lost.

Finally, if you have any other CircuitPython boards for testing -- particularly an official adafruit board not using an STM-family microcontroller -- comparative results from them would be interesting. I said at the outset I wouldn't be surprised if there was a problem with the unusual phase/polarity combinations, so I could be barking up the wrong tree with these guesses that it is in the low level implementation of `busio.SPI` for your microcontroller, maybe just for the rare half-duplex mode, or maybe for STM microcontrollers in general, which Adafruit doesn't place on many boards.

(You might also consider trying to either eliminate or implicate half duplex mode by specifying other available SPI pins on the package, even if they're not attached to a useful device or connector. It would seem to be necessary to scope SCK only to see whether this problem exists in normal full duplex mode or simplex mode)

sgauche, 2022-02-15:
Possibly resetting the SPI on the STM on entering adafruit_bus_device.SPIDevice here by calling common_hal_busio_spi_configure:
https://github.com/adafruit/circuitpython/blob/main/shared-module/adafruit_bus_device/spi_device/SPIDevice.c#L57
And then not waiting for configuration to complete and setting chip select low.

common_hal_busio_spi_configure does reset the SPI as shown here:
https://github.com/adafruit/circuitpython/blob/main/ports/stm/common-hal/busio/SPI.c#L281-L290

sgauche, 2022-02-15:
@jepler same result without adafruit_bus_device.  The 16 clocks on the read is a bug in ST's drivers that I know about, has to do with their half-duplex support.
```
import time
import board
import digitalio
import busio
from adafruit_bus_device.spi_device import SPIDevice

spi_sns_mag_cs = digitalio.DigitalInOut(board.SPI_SNS_MAG_CS)
spi_sns_mag_cs.direction = digitalio.Direction.OUTPUT
spi_sns_mag_cs.value = True

spi_sns = busio.SPI(board.SPI_SNS_SCK, MOSI=board.SPI_SNS_BIDI, MISO=None, half_duplex=True)
while not spi_sns.try_lock():
    pass
spi_sns.configure(baudrate=781250, polarity=1, phase=1)
while True:
    time.sleep(10)
    spi_sns_mag_cs.value = False
    spi_sns.write(bytearray([0x4F]))
    result = bytearray(1)
    spi_sns.readinto(result)
    spi_sns_mag_cs.value = True
    print(result)
```
[image]

sgauche, 2022-02-15:
I'll try full-duplex tomorrow with a NC'd MISO pin. I unfortunately don't have another adafruit board around.

sgauche, 2022-02-15:
I could also try this in C with their drivers and see if I get the same SCK behavior.

sgauche, 2022-02-15:
Sounds familiar:
https://electronics.stackexchange.com/a/528328

sgauche, 2022-02-15:
On page 821 of RM0402 STM32F412 reference manual there is this little note:  
`
The idle state of SCK must  correspond to the polarity selected in the SPIx_CR1  register (by pulling up SCK if CPOL=1 or pulling down SCK if CPOL=0).`
Sounds like I need to add a pull up resistor to my SCK. Which I guess would make sense to tri-state the clock when not in use if you had a multi-master setup... Surprised that isn't better documented.

sgauche, 2022-02-15:
But then that begs the question of what do you do on SPI busses with multiple slaves that have different POL and PHA requirements... hmm. I'm going to close this for now as I don't think it is a CPy issue.

closed 2022-02-15 by sgauche