adafruit/circuitpython #6042: Enable _uasyncio

dhalbert, 2022-02-16:
`_uasyncio` is not used for asyncio, right now because the `_uasyncio` implementation of `Task` does not provide`__await___()`. Our internal implementation of `async/await` (done by @WarriorOfWire) requires that. Otherwise code is going to fail. I had to add `__await__()` in the Python version in a couple of places:
https://github.com/adafruit/Adafruit_CircuitPython_asyncio/blob/28ef812f14ef1fde1036eb83a4878bd14aa9a67d/asyncio/task.py#L152
https://github.com/adafruit/Adafruit_CircuitPython_asyncio/blob/28ef812f14ef1fde1036eb83a4878bd14aa9a67d/asyncio/core.py#L52-L53

I started to add `__await__()`, but because `__iter__()` is provided in a special way, I had not yet figured out how to make it work, and so did not use `_uasyncio` initially.

In addition, I would like to rename `_uasyncio` to `_asyncio`, but that is minor.

_Originally posted by @dhalbert in https://github.com/adafruit/circuitpython/pull/6041#pullrequestreview-884596976_

See #6041 for initial work on this (thanks @t-ikegami).

t-ikegami, 2022-02-22:
I have reopened https://github.com/adafruit/circuitpython/pull/6041, where `__await__` method is supplemented to Task class.

dhalbert, 2022-10-23:
Fixed by #6041.

closed 2022-10-23 by dhalbert