adafruit/circuitpython #6044: Expose control of stop bit on i2c write

dmccue128, 2022-02-16:
I must use an I2C device that wants a write that is not followed by a stop bit or a (immediate) read.
  
The underlying Wire library offers clients the ability to send or suppress the stop bit after a write on the I2C bus.

The CircuitPython busio library always sets this flag to True in busio,writeto, forcing a stop bit after a write (except when using write_then_read).

Please expose the underlying control of Stop Bit behavior by adding an optional (keyword?) parameter to busio.writeto allowing clients to choose whether or not to send a stop bit following a write e.g., 
          writeto(self, address:[int, buffer: ReadableBuffer, *, start: int = 0, end: int = sys.maxsize, stop=True) → [None]

The capability is available in other languages and already supported by the underlying library used in CircuitPython, but hidden by the CircuitPython interface.  The effort to enable it would be small and fully backward compatible for all users.  Without it, some devices cannot be accessed using CircuitPython.

ladyada, 2022-02-16:
you can totally do that, just as two separate calls: one write then one read. if you write, by default it will add a stop. if you write-then-read it will not add a stop.  we do this because linux kernel doesn't allow a write without a stop in a single call without a followup read. this way we can make sure drivers will work on linux as well as microcontrollers.

https://circuitpython.readthedocs.io/en/latest/shared-bindings/busio/

please try it :)

dmccue128, 2022-02-16:
Thank you for the prompt response!  Unfortunately, a write followed by a
read does not work in circuitpython.  The write is terminated by a stop
bit, which causes the peripheral device to abandon the operation.
I need to write data from the controlling microprocessor (without a
following stop bit), wait for a signal from the peripheral device
indicating data is ready, then read from the peripheral device then send a
stop bit/condition (then release the bus lock).
I can do it in c++, but not in circuitpython.  Am I missing something?
Thanks,
Dan

ladyada, 2022-02-16:
what is the device you are writing a driver for?  so far we've been able to support all the i2c chips we've seen that follow i2c standard.

dhalbert, 2022-02-16:
Did you try `writeto_then_readfrom()` ?

dmccue128, 2022-02-16:
I did not, but I will.
The specification from the device manufacturer says I must wait until a
digital signal from the device indicates that the data is available before
reading.  In my preliminary tests, there seemed to be a significant delay
between the completion of the write and the signal indicating "data ready"
to read.  I expected that writeto_then_readfrom would attempt the read
before the data was ready, but I can try it to see if it works to ignore
the data-ready signal...
Thanks,
Dan

dmccue128, 2022-02-16:
The device is a lighting control system.  If I had access to/control of the
peripheral device software, I might be able to change it to look more
conventional, but I do not.
The specification of the peripheral device says I must write, then wait for
a data ready signal before reading.  I will try writeto_then_readfrom, but
I suspect it will attempt the read before the data is ready...  It may be
inelegant, but I believe the peripheral device conforms to the I2C
specification, but I'm not an I2C expert.

dmccue128, 2022-02-16:
> Please tell us the device so we can look at the datasheet. The usual way
> of doing this is that the device does clock-stretching until it is ready.
> Is the signal you are talking about another pin, or is it a part of the I2C
> transaction.

The signal is coming from another pin.
I cannot send the datasheet as we are bound by an NDA from the vendor.
I will try writeto_then_readfrom.  Perhaps the clock-stretching is enabled
and the digital IO pin for "data ready" is superfluous.

ladyada, 2022-02-16:
a data ready line in between write and read is really completely outside the i2c spec, please contact the vendor cause they're doing something very weird.

dmccue128, 2022-02-16:
OK, thanks.  I will definitely contact the vendor about this.
I have had great success accessing other I2C devices using CircuitPython,
but this one is killing me.

bludin, 2022-02-17:
just an idea: on the hardware-level and depending on the timing of the data ready signal, you might be able to (ab)use it to stretch the clock.

dmccue128, 2022-02-17:
Thanks for the idea!
I'm looking at all options...

dhalbert, 2022-03-09:
@dmccue128 Did you get to a resolution on this?

dmccue128, 2022-03-09:
Dan,

Thanks for following up! I apologize for not responding sooner.

I think I have convinced the device manufacturer to produce a version of
their firmware that conforms to the practice of sending a stop bit
following the write. This will solve my problem so I will not need to
convert my code from circuitpython to c++.  You can close this issue now.

However, after my discussions with the vendor and review of other I2C
devices, I am less certain that it benefits circuitpython to enforce this
rule.  The reason switching to c++ solves the problem for me is that the
Arduino Wire library makes specific accommodation for this protocol variant
(See Arduino - WireEndTransmission
<https://www.arduino.cc/en/Reference/WireEndTransmission>).
A similar, backward-compatible change to the circuitpython library would be
straightforward and would open the possibility of using circuitpython to
other developers with similar devices that they cannot force into
compliance.

Thanks for your help and consideration of my issue!  I hope I can repay the
favor at some point by joining the effort to make circuitpython even better.

With respect,
Dan

dmccue128, 2022-03-09:
> Thanks for the reply. As @ladyada <https://github.com/ladyada> mentioned:
>
> we do this because linux kernel doesn't allow a write without a stop in a
> single call without a followup read. this way we can make sure drivers will
> work on linux as well as microcontrollers.
>
> We did used to support full control of the stop bit, and found it that we
> had more support problems supplying it than not. People would write drivers
> that would sort of work, but not well, and not on Linux. So we made the
> capabilities adhere to the I2C standard.
>
> If there were many I2C devices that violate the standard, we might throw
> up our hands, but they are extremely rare, and we consider them buggy. The
> question is whether to accommodate 99% or 100%, I think. I do see your
> point about flexibility, but for us, it is at the expense of reduced
> compatibility with Linux, and increased support needed.

Dan,
Makes sense, thanks.  I value the compatibility with Linux and I can
appreciate the burden of attempting to support non-compliant devices.  This
is a good decision.  Thanks for the explanation.
Cheers,
Dan

ladyada, 2022-03-09:
whats missing from the api to do a block read call?

deshipu, 2022-03-09:
When you do an I2C read in CircuitPython, you have to specify the number of bytes to read up front, and it's always a whole transactions. But with SMBus, the first byte you read specifies how many more bytes to read. So we would need some way to do a read without ending the transaction, and then a second read that ends it.

ladyada, 2022-03-09:
oh thats freaky - ive never actually seen it done in a device before! i think for now we'd have to just read MAX_EXPECTED, then look to see what the first byte is

ladyada, 2022-03-09:
PN532 is one of the wierdest i2c devices - https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/main/adafruit_pn532/i2c.py#L72
they use a status byte to indicate whether to read the rest, but thankfully not repeated start. i guess if something had both we'd be out of luck, but so far that zebra hasnt appeared

deshipu, 2022-03-09:
Sadly, if a combination is possible, someone is going to use it sooner or later.

https://xkcd.com/1172/

ladyada, 2022-03-09:
sounds like something you should take up with the linux kernel mailing list ;)

deshipu, 2022-03-09:
Oh, I've read those threads. I'm not getting anywhere near that community.

Thraxxos, 2022-07-06:
I have a programmable power supply using I2C and a custom PMbus protocol (similar to SMBus).  A Diligent Analog Discovery unit can produce the "repeated start" on the protocol analyzer, and the PSU responds as expected.

Below is a protocol capture of a GOOD read:
Start, 20 [ 10 | WR ], 00, 
Restart, 21 [ 10 | RD ], F5 NAK, Stop

The circuitpython busio device can perform one of two options::
1) write with stop=False, then readfrom_into
Looks like this on the protocol anaylzer:
Start,20 [ 10 | WR ], 00, 21, FF, Stop

2) writeto_then_readfrom
Start,20 [ 10 | WR ], 00, 21, FF, Stop

Any ideas how to tweak it to produce the "restart" in the protocol viewer?

Thraxxos, 2022-07-06:
[image]

tannewt, 2022-07-07:
> 2\. writeto_then_readfrom
>         Start,20 [ 10 | WR ], 00, 21, FF, Stop

How are you calling this? What microcontroller are you using? It should do a repeated start when switching from write to read.

Thraxxos, 2022-07-07:
This is a PC (Fedora) with Blinka FT232H break out board via USB running python busio commands.
`
import board
import busio
i2c = busio.I2C(board.SCL, board.SDA, 500) 
while not i2c.try_lock():
    pass
device = 0x10
result1 = bytearray(1)
for itemcount, item in enumerate(commands):
    if item[0]=='RD':
        #i2c.writeto(device,bytes([item[1]]), stop=False)
        #i2c.readfrom_into(device, result1)
        i2c.writeto_then_readfrom(device,bytes([item[1]]),result1)
        print("Item expected was ",str(hex(item[2])),". Item read was ", hex(result1[0]), ".")

..........
`

What micro/board are you using which has the correct expected behavior?
I have a PYNQ Z1 I could try this on as well as an EFM PG22 (Silicon Labs).

Braon

tannewt, 2022-07-08:
@Thraxxos This sounds like a Blink FT232 bug. Please file a new issue here: https://github.com/adafruit/Adafruit_Blinka/issues

I'd expect correct behavior on any of the boards here: https://circuitpython.org/downloads

Unfortunately, neither of the two you have are supported.