adafruit/circuitpython #6061: PWM not properly relased on reload

Neradoc, 2022-02-18:
Running on a Clue, I get "All timers in use" error when using PWMOut after a few reloads.
I originally noticed that on the Xiao NRF52 using adafruit_rgbled, and reproduced it on a Feather NRF52.
Couldn't reproduce on other ports.

This code takes 4 reloads to error. Less if more pins are used.
```py
import board
import time
import pwmio
pins = [board.D0, board.D1, board.D2, board.D3, board.D4,]
pwms = []
for pin in pins:
    print(pin)
    pwm = pwmio.PWMOut(pin)
    pwms.append(pwm)
while True:
    time.sleep(1)
```
```
[...]
Code stopped by auto-reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
board.P0
board.P1
board.P2
board.P3
board.P4
Traceback (most recent call last):
  File "code.py", line 13, in <module>
RuntimeError: All timers in use
```

tannewt, 2022-02-22:
This usually happens when we don't fully reset the PWM peripheral or how it's tracked as in use between runs. I think this is usually done from `reset_port()`.

Neradoc, 2022-04-05:
I can confirm the code now passes without error with #6233

PR #6233 (2022-04-05): nrf: Correctly mark PWM unused during reset
This register is ALSO used to tell if the PWM channel should be considered in use by software.

Closes #6061.
closed 2022-04-05 by Neradoc