adafruit/circuitpython #6063: `git describe` needs `--long` when used in setup() in setup.py

dhalbert, 2022-02-19:
The fix in #6033 to use `git describe --first-parent ...` does not quite work when there is a tag.
https://github.com/adafruit/circuitpython/runs/5254687864?check_suite_focus=true#step:7:151

`tools/describe` does
```sh
~/repos/adafruit/circuitpython$ git describe --first-parent --dirty --tags --always --match "[1-9].*"
7.2.0-rc.0
```
The default describe used by `setup()` is
https://github.com/pypa/setuptools_scm/blob/bdcbe2ee9da8f422c80b0090bfd95162401a7c5d/src/setuptools_scm/git.py#L19
```py
DEFAULT_DESCRIBE = [
    "git",
    "describe",
    "--dirty",
    "--tags",
    "--long",
    "--match",
    "*[0-9]*",
]
```
which adds a `--long`, so that you get something like:
```
7.2.0-rc.0-0-g291a33521
```
always. `setup()` depends on this form.

jepler, 2022-02-20:
@dhalbert can we close this up?

dhalbert, 2022-02-22:
Fixed by #6064.

PR #6064 (2022-02-20): use --long for describe in setup.py-stubs
- Fixes #6063.

This will use `--long` so `setup()` in `setup.py` will be happy when running on a tag.
closed 2022-02-22 by dhalbert