adafruit/circuitpython #6066: core adafruit_bus_device module not working with latest build

jerryneedell, 2022-02-19:
When I use the latest build for the feather_m0_rfm9x it fails when importing  the bus_ddevce module
```

Adafruit CircuitPython 7.3.0-alpha.0 on 2022-02-19; Adafruit Feather M0 RFM9x with samd21g18
>>> import adafruit_rfm9x
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_rfm9x.py", line 17, in <module>
ImportError: no module named 'adafruit_bus_device.spi_device'
>>> 
```

If I upload the python version of the library from the 7.x bundle, it works .

```
jerryneedell@jerryneedell-ubuntu-macmini:~/bundle$ cp -r 7.x/lib/adafruit_bus_device /media/jerryneedell/CIRCUITPY/
```
then
```

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.0-alpha.0 on 2022-02-19; Adafruit Feather M0 RFM9x with samd21g18
>>> 
>>> import adafruit_rfm9x
>>> 

```

jerryneedell, 2022-02-19:
This very strange:
I can import adafruit_bme680  (an I2c device,  but not adafruit_rfm9x (an spi device)
but both use adafruit_bus_device....
I am confused

dhalbert, 2022-02-19:
Do you see this problem with 7.2.0-rc.0?

7.3.0-alpha.0 merged in some new code from MicroPython for freezing modules as part of the merge of MicroPython 1.18. The merged code may be broken in some way.

Try building off the 7.2.x branch for a bit if you want to experiment with freezing things to avoid the merge.

jerryneedell, 2022-02-19:
I'll try that -- but do you have any idea why I can import bme680 but not rfm9x?

jerryneedell, 2022-02-19:
Also it the core bus device module supposed to be useable any more?

dhalbert, 2022-02-19:
>I'll try that -- but do you have any idea why I can import bme680 but not rfm9x?

bme680 does `from adafruit_bus_device import spi_device` conditionally, in the constructor. rfm9x does `import adafruit_bus_device.spi_device as spidev` at the top level, and it does the rename as well. I have no idea why this would make a difference, but it is a difference to be investigated.

>Also is the core bus device module supposed to be useable any more?

Sure, it's supposed to be.

PR #6068 (2022-02-19): Fix import of builtin submodules
Originally I was going to figure out where to restore our added behavior for submodule importing, but I found that as long as the submodule is registered as `foo.bar` it can be imported without other code changes.

Closes: #6066 

Note that allowing dotted submodules like this requires a CircuitPython extension: _dot_ in a QSTR name can represent a ".". This isn't possible in general in micropython, except if the string is listed in qstrdefs.h (that's how QSTR__dot_frozen works in MP)
commit 5d213e53 (2022-02-19): bus device: properly register submodules

Closes: #6066 (except for similar work that needs to be done in ulab)
closed 2022-02-19 by dhalbert