adafruit/circuitpython #6070: QT Py ESP32-S2 having issues with I2C

muendlein, 2022-02-20:
### CircuitPython version

```python
7.1.1
```

### Code/REPL

```python
# SPDX-FileCopyrightText: Copyright (c) 2020 ladyada for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import board
import adafruit_sht4x

print("test")
i2c = board.I2C()  # uses board.SCL and board.SDA
sht = adafruit_sht4x.SHT4x(i2c)
print("Found SHT4x with serial number", hex(sht.serial_number))

sht.mode = adafruit_sht4x.Mode.NOHEAT_HIGHPRECISION
# Can also set the mode to enable heater
# sht.mode = adafruit_sht4x.Mode.LOWHEAT_100MS
print("Current mode is: ", adafruit_sht4x.Mode.string[sht.mode])

while True:
    temperature, relative_humidity = sht.measurements
    print("Temperature: %0.1f C" % temperature)
    print("Humidity: %0.1f %%" % relative_humidity)
    print("")
    print("")
    time.sleep(1)
```

### Behavior

It seems like the QT Py ESP32-S2 has some issues to get I2C running with the existing libraries. 

RuntimeError: No pull up found on SDA or SCL; check your wiring

### Description

- Sensor works fine with other devices
- Same issue when using the Arduino code

Neradoc, 2022-02-20:
Hi,
The Stemma QT port on that QT PY is not on board.I2C, which uses the SDA and SCL pins marked on the side of the board.
It is on board.SDA1 and board.SCL1.
```py
import busio
i2c = busio.I2C(scl=board.SCL1, sda=board.SDA1)
```
With Circuitpython 7.2.0 (currently in realease candidate) you can use:
```py
i2c = board.STEMMA_QT()
```
Guides will surely be updated when the release is stable.

ladyada, 2022-02-20:
yep please check
https://learn.adafruit.com/adafruit-qt-py-esp32-s2/i2c
which has some example code/warnings!

muendlein, 2022-02-20:
Thanks for the hint! Seems like I have overlooked the obvious stuff...

closed 2022-02-20 by muendlein