adafruit/circuitpython #6071: Screen is cleared after code.py exits

rsbohn, 2022-02-20:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-rc.0 on 2022-02-19; Adafruit FunHouse with ESP32S2
```

### Code/REPL

```python
# trigger a syntax error
import None
```

### Behavior

After code.py exits the FunHouse display is empty except the Blinka image in the top left corner. After a soft restart (Ctrl-D) the only text on the display is:

    Press any key to enter the REPL. Use CTRL-D to reload.

### Description

Expected: The text after running should appear on the display.
Actual: The display is cleared when code.py exits.

**Any error messages that would be on the display are gone.**

### Additional information

Same behavior:

    Adafruit CircuitPython 7.2.0-alpha.2 on 2022-02-11; Adafruit FunHouse with ESP32S2

Works as expected:

    Adafruit CircuitPython 7.2.0-alpha.1 on 2021-12-29; Adafruit FunHouse with ESP32S2

dhalbert, 2022-02-20:
Maybe we should fix this for 7.2.0, since it's a recent regression.

dhalbert, 2022-02-20:
If I revert #5954, the screen is not cleared. @kmatch98, I don't immediately see why the terminal gets cleared on restart, but you probably know. Do you see a way to fix this? I thought it might be due an VM object going away, but I don't see that is the problem.

kmatch98, 2022-02-20:
@dhalbert Is there something special with the funhouse display where the upper left corner is not (0,0)?  Whenever `display.show(None)` is executed the terminal is moved back to (0,0). If for some reason the display setting don’t match that, then it could pace the REPL terminal out of view. 

I think the best way to debug is to test on the hardware and boot straight to REPL and check the root_group display element (x,y) locations on the screen and then trigger the bug and see the differences. 

Next step if we want to fix the bug will be to understand if there are board settings that need to be looked up to place the elements in the proper location. If you have hints about that, let me know and I’ll take a look. 

Unfortunately I don’t have the hardware to test.

dhalbert, 2022-02-20:
@kmatch98 I tested this on a PyPortal, and though the screen does not blank completely on restart, it does erase the error message. Do you have this or a similar board to test?
E.g.:
```py
print("Hello World!")
3/0    # raises ZeroDivisionError
```
7.1.1:
[image]
7.2.0-rc.0:
[image]

dhalbert, 2022-02-20:
I think the FunHouse and the PyPortal may differ in the initial rotation. I remember we had to do something tricky with the FunHouse to get the orientation right.

rsbohn, 2022-02-20:
@FoamyGuy noticed this in his stream yesterday (2/19) -- he was using a PyPortal.
I also see it on a featherS2+FeatherWing TFT 3.5" combo. After the script sets up the display, any error messages are lost.

    import board
    import displayio
    from adafruit_featherwing.tft_featherwing_35 import TFTFeatherWing35

    if "feathers2" in board.board_id:
        featherwing = TFTFeatherWing35(
            cs=board.D5, dc=board.D6, 
            ts_cs=board.D21, 
            sd_cs=board.D20)
    print("ovo")
    3/0

FoamyGuy, 2022-02-20:
I think maybe the terminal is in the right position but gets the text from it cleared out. If you start using the REPL you do see it begin to show things on the display.

Do the start/stop terminal here result in clearing the text that was previously in the terminal away? https://github.com/adafruit/circuitpython/pull/5954/files#diff-edac12d728f2a3db7c325f4b3d9865f5fe9feb69e34302d626dc2cd11b2bd695R172-R173

FoamyGuy, 2022-02-20:
@kmatch98 I did a brief test with split screen REPL / displayio on a build that removes those `supervisor_stop_terminal()` and `supervisor_start_terminal()` statements. I was still able to reset_terminal to half the display size, add a Label to the displayio group and move it around on it's half of the display. Everything seemed to behave normally, but I was not super thorough.

Do you think there is potential harm in removing these?

dhalbert, 2022-02-20:
Is it `supervisor_stop_terminal()` or `supervisor_start_terminal()` that does the clearing? Can you leave one out? I wonder if you need the `supervisor_start_terminal()` in case the dimensions change. Maybe it can check if the dimensions are actually changing and only do it if necessary.

FoamyGuy, 2022-02-20:
Having only one or the other doesn't lead to a better outcome in either case than removing both of them.

With only `supervisor_start_terminal()` the error message is still not shown. But I don't see other adverse effects at first glance.

With only `supervisor_stop_terminal()` the error message is also not shown, but the REPL is no longer shown on the display either only the blinka icon and no other text..

kmatch98, 2022-02-20:
As for including `supervisor_start_terminal()` and `supervisor_stop_terminal()`, I think I just blindly copied what was already in there.  

If we want to keep the old error message around: [One thing I noted in the PR ](https://github.com/adafruit/circuitpython/pull/5954#issuecomment-1026322206) was whether we wanted to keep the text data in the terminal when we resized it.  The challenge with that is that if the terminal is resized, first you have to allocate a new Terminal, then copy the old Terminal data into it, then deallocate the old terminal.  Initial discussion in the PR suggested to just start with a clean slate. If we want to keep the data from the previous terminal, I can revisit.

kmatch98, 2022-02-20:
> Having only one or the other doesn't lead to a better outcome in either case than removing both of them.
>
> With only `supervisor_start_terminal()` the error message is still not shown. But I don't [...]

Now I’m remembering  `stop_terminal` deallocates the memory for the terminal and `start_terminals reallocates memory for the new terminal. If we leave one or both of these out, we may either cause a memory leak or the terminal may not be the correct size if it was moved or resized.

FoamyGuy, 2022-02-20:
I thinks okay not to retain the data in the terminal when it gets resized. The user presumably would have intentional resized it so they would understand why it's gone I think.

In the case originally reported here though there terminal isn't getting resized, at least not intentionally I don't think. The user has a `code.py` happens to raise an exception which stops it executing and breaks out to "press any key to enter REPL". So have not done anything with resizing the terminal.

I do think we want to preserve the terminal text if possible in this case because it makes troubleshooting easier on devices with displays.

My guess is maybe the system is calling `displayio_display_core_show()` when the exception is encountered in order to switch back to the terminal (assuming user had shown something else) but its appending the error message before doing so and thus it's getting lost when this start/stop occurs.

kmatch98, 2022-02-20:
Ok I’ll take a crack at avoiding the start/stop of the size is not changing.

dhalbert, 2022-02-20:
@kmatch98 If you make a PR, could you make it against `7.1.x`? Thanks.

kmatch98, 2022-02-21:
In looking deeper, if we just run `supervisor_start_terminal`, if the terminal is the same size, it should reuse the tile grid allocation.  However, in testing it I see that the tiles get cleared.  I've got to dig deeper to see where the tiles get cleared.

dhalbert, 2022-02-21:
> @kmatch98 If you make a PR, could you make it against `7.2.x`? Thanks.

I said ~~`7.1.x`~~ by mistake. Corrected.

dhalbert, 2022-02-21:
Fixed by #6076

closed 2022-02-21 by dhalbert