adafruit/circuitpython #6072: readthedocs sphinx doc is not always including `*` from signatures

dhalbert, 2022-02-20:
Asterisks in Sphinx Python type signatures are not showing up in the generated doc, but only sometimes:
`usb_cdc.c:`
```c
//| def enable(*, console: bool = True, data: bool = False) -> None:
//|     """Enable or disable each CDC device. Can be called in ``boot.py``, before USB is connected.
```
[image]

But sometimes it shows up:
`SPI.c:`
```c
//|     def configure(self, *, baudrate: int = 100000, polarity: int = 0, phase: int = 0, bits: int = 8) -> None:
//|         """Configures the SPI bus. The SPI object must be locked.
```
[image]

dhalbert, 2022-02-21:
This is some kind of problem with `autoapi` dropping `*` in signatures; I can see it in the autoapi output. But I don't know why it's doing that, and I don't see any filed issues complaining about this yet. I have a simple test case.

dhalbert, 2022-02-21:
I reproduced this problem with a simple example and opened an an autoapi issue about this: https://github.com/readthedocs/sphinx-autoapi/issues/328

dhalbert, 2022-05-31:
This is still broken in Sphinx 5.0.0.

jepler, 2022-07-06:
We fixed this in https://github.com/adafruit/circuitpython/pull/6449 for ourselves.

closed 2022-07-06 by jepler