adafruit/circuitpython #6078: When the REPL is resized and then code.py completes, the screen may end up black with just a Blinka

kmatch98, 2022-02-21:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-rc.0-3-g6b2266b24-dirty on 2022-02-21; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
import board
import displayio
import supervisor
import time

display=board.DISPLAY

# Create a Group
mygroup = displayio.Group()

print()
print()
print()
print("REPL control test: Taking control of REPL group")
time.sleep(3)

# clear the display to the REPL 
display.show(None)
splash = board.DISPLAY.root_group # this gets the current root_group, the REPL

# Note: You must "display.show" your own group before adding the splash to your own group.
# Reason: When displaying the normal REPL (for example with display.show(None), the splash
# group is already in a group that is displayed.  To remove the splash from the displayed group,
# you first have to display.show some other group, doing that will remove the splash from its group
# and allow you to append it to your own group. 
display.show(mygroup)

# resize the supervisor.splash group pixel dimensions, make it half the display height.
supervisor.reset_terminal(display.width//2, display.height//2)

# relocate the supervisor.splash group on the display, moving it half-way down the display
splash.x=display.width//2
splash.y=display.height//2
print("Resize and move the splash screen")

# append the supervisor.splash group to the displayed group.
mygroup.append(splash)
time.sleep(2)

# demonstrate how print statements scroll on the REPL/console display
print("Add some prints to show terminal scrolling:")
time.sleep(0.3)
for i in range(6):
    print("Line: {}".format(i))
    time.sleep(0.3)
time.sleep(1.5)

print("Ending code.py")

### If there is no serial connection, just the Blinka remains after the code is done.

### Have to reset the display to show(None) to see the "Code done running." text.
# display.show(None)
```

### Behavior

When using the split-screen REPL that was repaired in https://github.com/adafruit/circuitpython/pull/6077, there is some unexpected behavior when `code.py` ends.

Three scenarios and the output observed:
1. No serial connection is available (**This one is disconcerting! Only a Blinka is shown.**)
[image]

2. Serial connection is available
[image]

3. Add`display.show(None)` at the end of `code.py` (see my code above, uncomment the last line)
[image]

### Description

Only happens when `code.py` ends while having the REPL group resized using the new capabilities in the code above.

dhalbert, 2022-02-22:
Fixed by #6077.

dhalbert, 2022-02-22:
@kmatch98 sorry, I thought this was pre-PR, and closed it by mistake.

commit 3a2bcbc5 (2022-08-05): Enable title bar on all builds

* Tweak scroll area position so last line is complete and top is
  under the title bar.
* Pick Blinka size based on the font to minimize unused space in
  title bar. Related to #2791
* Update the title bar after terminal is started. Fixes #6078

Fixes #6668
closed 2022-08-05 by tannewt