adafruit/circuitpython #6080: ESP32-S2 TFT - board.BUTTON attribute no found

UmBsublime, 2022-02-22:
### CircuitPython version

```python
If I understand the pinout of the `ESP32-S2 TFT` correctly I should be able to access to `boot` button through the `board`'s `BUTTON` attribute, but this throws an exception.

Link to pinout PDF: https://github.com/adafruit/Adafruit-ESP32-S2-TFT-Feather-PCB/blob/main/Adafruit%20ESP32-S2%20TFT%20Feather%20Pinout.pdf

I can't find another way to access GPIO0.
```

### Code/REPL

```python
# Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Feather ESP32-S2 TFT with ESP32S2
import board
import digitalio
b = digitalio.DigitalInOut(board.BUTTON)
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'BUTTON'
```

FoamyGuy, 2022-02-22:
I think the BUTTON attribute may have been added after the release of version `7.1.1` I would recomend trying with version `7.2.0-rc.0` and see if `board.BUTTON` exists on that one.

ladyada, 2022-02-22:
its def there now 
https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/adafruit_feather_esp32s2_tft/pins.c#L68

UmBsublime, 2022-02-22:
Will check later today and report back. Thanks for the amazingly quick response !

UmBsublime, 2022-02-23:
Confirmed working as expected on `7.2.0-rc.0`

For those who are waiting for `7.2.0` to land in stable I worked around this with:

```python
import microcontroller
import digitalio
b = digitalio.DigitalInOut(microcontroller.pin.GPIO0)
```

closed 2022-02-23 by UmBsublime