adafruit/circuitpython #6082: Feather STM32F405 intermittent exception on close() and flush()

RealAbsurdity, 2022-02-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Feather STM32F405 Express with STM32F405RG
Board ID:feather_stm32f405_express
```

### Code/REPL

```python
# can't so the entire program... the code in the traceback is here, line 223 is the last line  

            if SDWRITE:
                # save the data to the data_log.txt file
                with open(DFN, "a") as dl:
                    dl.write(ds)
                    dl.write('\n')
                    dl.close()   # error on line 223
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 223, in <module>
OSError: [Errno 5] Input/output error

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

### Description

- Exception after writing data to a file on the local microsd card using sdioio and closing it
- Exception is intermittent and can be caught with an exception handler
- Error occurs intermittently when calling close() on a file
- Same error occurs intermittently when using flush() on same file
- Only occurs on Adafruit Feather STM32F405 Express with STM32F405RG

### Additional information

Can be resolved by removing the call to close() or flush() but possibly an issue with the io.BytesIO methods

tannewt, 2022-02-23:
Please provide a full test program that shows the issue. It's much easier to fix a full example.

RealAbsurdity, 2022-02-24:
I will attempt to recreate with a test program that includes most of the major modules in the original program.

RealAbsurdity, 2022-02-25:
Here is a test program that replicates the issue. The exception occurs on line 33 when .close() is called.

```python
import board
import time
import sdioio
import storage

print("Initializing Data Storage...")
# data storage setup #

# initialize sdcard and storage
sdcard = sdioio.SDCard(clock=board.SDIO_CLOCK,
                        command=board.SDIO_COMMAND,
                        data=board.SDIO_DATA,
                        frequency=25000000)

vfs = storage.VfsFat(sdcard)
# mount the storage device
storage.mount(vfs, "/sd")
print("SD card present and mounted")

DFN = "/sd/data_log.txt"
print("Saving data to:", DFN)

ds = "0000000000000000000000000000000000000000000000000000000000000000"
dc = 0

while True:

    dc += 1

    with open(DFN, "a") as dl:
        dl.write(ds)
        dl.write('\n')
        dl.close()

    print(dc)

    time.sleep(0.1)
```
The code will fail at various times in the sequence while trying to write a 64 character string to the storage device over sdioio. 
Output to serial monitor with traceback:

```bash
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Initializing Data Storage...
SD card present and mounted
Saving data to: /sd/data_log.txt
1
2
3
4
5
6
Traceback (most recent call last):
  File "code.py", line 33, in <module>
OSError: [Errno 5] Input/output error

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

tannewt, 2022-02-25:
Thank you for the test program! That'll be helpful when someone has time to debug this issue.