adafruit/circuitpython #6090: ESP32-S3 exits deep sleep immediately

dgriswo, 2022-02-25:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-9-gf39c2b2e1 on 2022-02-24; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3
```

### Code/REPL

```python
import time
import alarm

time_alarm = alarm.time.TimeAlarm(monotonic_time=(time.monotonic() + 600))
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

[image]

Board does not enter deep sleep.  Code re-executes immediately.

Deep sleep is faked correctly when connected via USB.

### Additional information

Also reproduced on 7.2.0-alpha.2 and 7.2.0-rc.2

dgriswo, 2022-02-25:
I threw some code in to publish microcontroller.cpu.reset_reason to MQTT. (ignore the battery voltage, the voltage divider math is wrong.)

[image]

dgriswo, 2022-02-25:
Also reproduced the issue (and reset_reason) using the Unexpected Maker TinyS3.

prplz, 2022-02-25:
Reproduced on esp32 s3 devkitc n8r8

prplz, 2022-02-25:
With some printf debugging, I tracked it down to this line hanging: https://github.com/adafruit/esp-idf/blob/circuitpython-v4.4/components/esp_hw_support/sleep_modes.c#L509.
If I comment that line out, sleep works as expected.

tannewt, 2022-03-11:
@igrr Has anyone seen issues entering deep sleep from the APP core? It looks like it's hanging when trying to stall the PRO core and then the watchdog resets things.

scarolan, 2022-06-27:
Confirmed on QT PY ESP32-S3

jepler, 2022-07-06:
Maybe there's a way to use IPC to make the sleep execute on the PRO CPU, but we'd really rather have the bug fixed upstream.

moefear85, 2022-07-09:
@prplz 

Line 509 is a closing bracket, so I think the code has changed since then. has the issue been resolved? if not, which line are you referring to? I'm curious, because there is a uart line close nearby, and I wouldn't be surprised if that is part of the problem.

prplz, 2022-07-10:
@moefear85 I think the line was `esp_ipc_isr_stall_other_cpu();`

moefear85, 2022-07-10:
@ prplz

oh then idk... it's just that there's currently this [issue](https://github.com/espressif/esp-idf/issues/9269), and I double checked, it doesn't call esp_ipc_is_stall_other_cpu, so whatever is triggering the wdt there, probably isn't the same thing causing this.

yonaguska, 2022-08-15:
I wish I had seen this before I bought 5 of the UM FeatherS3 boards; I need deep sleep for an IoT project, to conserve battery power. I know my code works on an Adafruit QtPy ESP32-S2, which suffers from another issue that brought me to the FeatherS3. 

The issue with the QtPy system is that it hangs up in what seems to be the SAFE_MODE, which I think is caused by voltage spikes in the supply when they are run off of a battery. I can't filter the spikes adequately, and a QtPy will run for hours or days before it gets hit, but it always gets hit eventually. Resolving this requires a hardware reset to get out of SAFE_MODE, and I built a hardware WDT to issue a reset signal to the processor. Since the QTPy lacks a reset pin, I soldered a wire from the WDT to the tiny reset switch on the QtPy, a real PIB. This is why I bought the UM FeatherS3, it has an exposed reset pin.

I am testing 5 of these QtPy systems, and the hardware WDT has fixed it on all of them. I looked to the FeatherS3 to rid myself of the frangible kluge of the reset wire; this system will be placed under my house (to monitor for water and gas leaks), and I really need it to be uber-reliable. I hate crawling under the house; at 72, it's not fun, trust me.

On a side note, I noticed someone had a similar WDT on eBay going for about $10; I built some extras that I could sell for about half.

But I digress, so back to my main concern. Has there been any progress on fixing this deep sleep issue?

tannewt, 2022-08-16:
> Maybe there's a way to use IPC to make the sleep execute on the PRO CPU, but we'd really rather have the bug fixed upstream.

I tried this with no luck. It still triggers a watchdog. My code is here: https://github.com/tannewt/circuitpython/tree/ipc_deep_sleep

yonaguska, 2022-10-17:
Also reproduced the error with an immediate start after deep sleep

- Unexpected Maker Feather S3
- CircuitPython 8.0.0 beta2
- example from the adafruit documentation (deep_sleep 20 seconds)

dgriswo, 2022-10-17:
There may be more visibility to this bug.  JP's CircuitPython Parsec for Oct 16, 2022 highlighted deep sleep.

dhalbert, 2022-10-27:
As implied above, this appears to be an upstream bug: https://github.com/espressif/esp-idf/issues/8569. Other Espressif users have reported this. It has been "selected for development" by Espressif. We will await an upstream fix for this. I'll remove this from the 8.0.0 issues.

dhalbert, 2022-10-27:
Reproduced on both Feather ESP32-S3 8MB flash no PSRAM and Feather ESP32-S3 4MB flash / 2MB PSRAM, using this test program. NeoPixel flashes three times indicating safe mode when connected to battery.
```py
import digitalio
import board
import alarm
import time

uart = board.UART()
uart.baudrate = 115200

led = digitalio.DigitalInOut(board.LED)
led.switch_to_output(True)
time.sleep(2)
led.value = False
msg = "Alarm was " + str(alarm.wake_alarm) + "\r\n"
print(msg, end = "")
uart.write(msg.encode("utf-8"))

time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 5)

msg = "Entering deep sleep at " + str(time.monotonic()) + "\r\n"
print(msg, end="")
uart.write(msg.encode("utf-8"))

alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

commit a8c70aa7 (2022-11-30): Fix S3 deep sleep

Adding `-u ld_include_highint_hdl` forces the linker to keep the
high priority interrupt handler that calls the ipc_isr handler.

The deep sleep is waiting for this interrupt to be handled on core
0 before sleeping from core 1.

Fixes #6090