adafruit/circuitpython #6091: LED Glasses DisplayIO: unscaled display shows garbage and hits HardFault handler

PaintYourDragon, 2022-02-25:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit LED Glasses Driver nRF52840 with nRF52840
Board ID:adafruit_led_glasses_nrf52840
```

### Code/REPL

```python
import board
import busio
import displayio
import framebufferio
import is31fl3741
from adafruit_is31fl3741.led_glasses_map import glassesmatrix_ledmap
from adafruit_display_shapes.circle import Circle

displayio.release_displays()

i2c = busio.I2C(board.SCL, board.SDA, frequency=1000000)
is31 = is31fl3741.IS31FL3741(i2c=i2c)

# The scaled framebuffer works; circle is expected size and color
#is31_framebuffer = is31fl3741.IS31FL3741_FrameBuffer(
#    is31, 54, 15, glassesmatrix_ledmap, scale=True, gamma=True
#)
#circle = Circle(7, 7, 7, fill=(255, 0, 0))

# Unscaled framebuffer does not. Wrong LEDs are lit, wrong colors,
# usually ends up hitting the HardFault handler. 
is31_framebuffer = is31fl3741.IS31FL3741_FrameBuffer(
    is31, 18, 5, glassesmatrix_ledmap, scale=False, gamma=True
)
circle = Circle(2, 2, 2, fill=(255, 0, 0))

display = framebufferio.FramebufferDisplay(is31_framebuffer, auto_refresh=True)

is31_framebuffer.brightness = 0.2

group = displayio.Group()
group.append(circle)
display.show(group)

while True:
    pass
```

### Behavior

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues
```

### Description

If scaled framebuffer part of above code is enabled (and unscaled is disabled), display works as expected — red circle (antialiased) at left edge of LED matrix. With unscaled enabled, get a blue band across matrix and HardFault handler is encountered. Happens with other primitives too, such as text.

dhalbert, 2022-03-22:
@gamblor21 Do you have an idea of what's wrong here? Do you have time or are you busy at the moment?

gamblor21, 2022-03-22:
Not sure why it’s occurring offhand but I should have a moment to look at it tomorrow or Thursday.  I didn’t test scale being false as much.

PR #6192 (2022-03-24): Fix no scaled framebuffer display
Partially fixes #6091 

Corrects error where non-scaled frame buffers were not displaying properly.

There is still a hard fault occurring that will take more time to figure out (something with memory release and the REPL).
closed 2022-03-24 by tannewt