adafruit/circuitpython #6096: Moved /extmod/ulab/code prevents compiling from fresh checkout

kreier, 2022-02-27:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.1-dirty on 2022-02-11; Blackpill 16MB SSIS edition with STM32F411CE

This version was compiled 16 days ago and works. The new 7.2.0 does not compile because of the missing /extmod/ulab/code
```

### Code/REPL

```python
import ...

We don't get that far :(
```

### Behavior

mk@zbook:~/circuitpython/ports/stm$ make BOARD=stm32f411ve_discovery
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
find: ‘../../extmod/ulab/code’: No such file or directory
make: *** No rule to make target 'lib/tinyusb/src/portable/st/synopsys/dcd_synopsys.c', needed by 'build-stm32f411ve_discovery/genhdr/qstr.split'.  Stop.

### Description

[image]
ulab seems to been moved on February 19th

dhalbert, 2022-02-28:
Did you update the git submodules? Do `make fetch-submodules` from the top level of your repo.

kreier, 2022-03-03:
Sorry, I think this solved it. I missed this step from https://learn.adafruit.com/building-circuitpython/build-circuitpython after a switch to 7.2.0 didn't worked as expected, so I started with a fresh copy. `make BOARD=circuitplayground_express` now works without error message. Just have to confirm it works with the altered `mpconfigboard.h` and `mpconfigboard.mk` for the blackpill with 16MB flash chip, but it looks it was not a bug at all

dhalbert, 2022-03-03:
Glad it's working!

closed 2022-03-03 by dhalbert