adafruit/circuitpython #6102: On ESP32-S2, microcontroller.cpu.temperature hangs when called with WiFi initialised

kattni, 2022-03-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Metro ESP32S2 with ESP32S2
Board ID:adafruit_metro_esp32s2
Also, TFT Feather on 7.2.0
```

### Code/REPL

```python
import time
import board
import neopixel
import wifi
import microcontroller
from secrets import secrets

wifi.radio.connect(secrets["ssid"], secrets["password"])

# Initialize NeoPixel
pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.3)

last = 0
start = time.monotonic()
print("Start: ", time.monotonic())
while True:

    # Obtain the CPU temperature, print it and publish it to Adafruit IO every 10 seconds.
    if (time.monotonic() - last) >= 10:
        temperature = "{:.2f}".format(microcontroller.cpu.temperature)
        print("Current: ", (time.monotonic() - start))
        print("Current CPU temperature: {0} C".format(temperature))
        last = time.monotonic()
```

### Behavior

After a period of time, fairly consistently under 1000 seconds, the serial console will hang. Then you can't interact with the serial console any longer (CTRL+C/CTRL+D do nothing). (The previous code I was working with was also listening for NeoPixel color data from Adafruit IO, which also stopped working.) Access to CIRCUITPY is still present, I can edit code.py and save changes, and on physical reset, the changes are present.

### Description

This issue is ONLY present when WiFi is imported AND initiated, AND `microcontroller.cpu.temperature` is called at any point.

### Additional information

This appears to be an ESP-IDF bug. Here is the open issue: https://github.com/espressif/esp-idf/issues/8088. 

After discussion with Dan, it is agreed that a long-term issue should be filed with a link to the ESP-IDF issue. This allows other folks to find it if searching, and provides us a place to discuss potentially disabling the CPU temp feature on ESP32-S2 for the time being.

anecdata, 2023-05-25:
Is this still an issue? I don't see that the Espressif issue is closed, but for some reason I thought it had been fixed and added back the temperature check in my wifi code some time ago without any problems since.

As a double-check, just ran this for over an hour on
`Adafruit CircuitPython 8.1.0-beta.2 on 2023-04-26; Adafruit QT Py ESP32S2 with ESP32S2`:
```py
import time
import os
import wifi
import microcontroller

def seconds_to_iso_time(s, tz):
    st = time.localtime(s)
    return f"{st[0]:04d}-{st[1]:02d}-{st[2]:02d}T{st[3]:02d}:{st[4]:02d}:{st[5]:02d}{tz:+03}:00"

last = 0
start = time.monotonic()
while True:
    if (time.monotonic() - last) >= 1:
        wifi.radio.connect(os.getenv("CIRCUITPY_WIFI_SSID"), os.getenv("CIRCUITPY_WIFI_PASSWORD"))
        print(f"{seconds_to_iso_time(time.time(), -5)} CPU={(microcontroller.cpu.temperature * 9 / 5 + 32):0.1f}°F [wifi connected={wifi.radio.connected}]")
        last = time.monotonic()
```
results were fine:
```
code.py output:
2023-05-24T19:47:24-05:00 CPU=92.1°F [wifi connected=True]
…
2023-05-24T21:01:36-05:00 CPU=92.1°F [wifi connected=True]
```

dhalbert, 2023-10-28:
Seems like it's fixed.

closed 2023-10-28 by dhalbert