adafruit/circuitpython #6111: memory leak on audiomp3.MP3Decoder Metro M4 Express

phantomsixthplayer, 2022-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Metro M4 Express with samd51j19
Board ID: metro_m4_express
```

### Code/REPL

```python
from adafruit_bitmap_font import bitmap_font
from adafruit_display_text import label
from adafruit_led_animation.animation.solid import Solid
import audioio
import audiomp3
import board
import digitalio
import displayio
import framebufferio
import neopixel
import rgbmatrix
import time

# RGBMatrix
displayio.release_displays()

matrix = rgbmatrix.RGBMatrix(
   width = 64,
   height = 32,
   bit_depth = 4,
   rgb_pins = [board.D6, board.D5, board.D9, board.D11, board.D10, board.D12],
   addr_pins = [board.A5, board.A4, board.A3, board.A2],
   clock_pin = board.D13,
   latch_pin = board.D0,
   output_enable_pin = board.D1
)
display = framebufferio.FramebufferDisplay(matrix)

# display groups
arcade_group = displayio.Group()

# font
font_1 = bitmap_font.load_font("/fonts/font_1.bdf")
font_2 = bitmap_font.load_font("/fonts/font_2.bdf")

# arcade_group graphics
ag_time = label.Label(font_1, text = "TIME", color = 0xB35A00)
ag_time.x = 1
ag_time.y = 4

ag_time_c = label.Label(font_2, text = "60", color = 0x00B300)
ag_time_c.y = 16

ag_score = label.Label(font_1, text = "SCORE", color = 0x0000B3)
ag_score.x = 29
ag_score.y = 4

ag_score_c = label.Label(font_2, text = "0", color = 0xFFFFFF)
ag_score_c.y = 16

ag_hiscore = label.Label(font_2, text = "HISCORE", color = 0x00B3B3)
ag_hiscore.x = 7
ag_hiscore.y = 28

ag_hiscore_c = label.Label(font_2, text = "22", color = 0xB30000)
ag_hiscore_c.x = 52
ag_hiscore_c.y = 28

# add graphics to display groups
arcade_group.append(ag_time)
arcade_group.append(ag_time_c)
arcade_group.append(ag_score)
arcade_group.append(ag_score_c)
arcade_group.append(ag_hiscore)
arcade_group.append(ag_hiscore_c)

# audio files
speaker = audioio.AudioOut(board.A0)
audio_file = {
   "countdown": "/audio/countdown.mp3"
}
mp3stream = audiomp3.MP3Decoder(open(audio_file["countdown"], "rb"))

# NeoPixels
led_pin = board.D25
num_leds = 5
leds = neopixel.NeoPixel(led_pin, num_leds, brightness = 0.20)
leds_green = Solid(leds, color = 0x00B300)

display.show(arcade_group)
score = 0
start_time = time.time()

mp3stream.file = open(audio_file["countdown"], "rb")

while True:
   if time.time() >= start_time + 10:
      leds_green.animate()
      speaker.play(mp3stream)
      start_time = time.time()
      score += 1   
      ag_score_c.text = str(score)
      ag_score_c.color = 0xFFFFFF
```

### Behavior

```
code.py output:
    File "code.py", line 72, in <module>
MemoryError: memory allocation failed, allocating 8708 bytes
```
```
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Metro M4 Express with samd51j19
>>> import gc
>>> gc.mem_free()
117584
```

### Description

Trying on two Metro M4 Express boards I get the memory allocation failed error, however when I try on a Feather M4 Express I do not get any memory allocation failed for the audio. I tried on a longer set of code with about 700 lines the Metro M4 reports memory allocation error but the Feather does not, tried on smaller code as attached and it "sometimes" happens and you hear a whine of the AMP only on the first audio play.

The error happens on this line, but if I comment it out everything works just no audio.
Line 72
```
#mp3stream = audiomp3.MP3Decoder(open(audio_file["countdown"], "rb"))
```

If I move the code of the audio higher to the top I get like 137 memory allocation failed but then the code does not run. I get other module errors if it's before the display appends.

### Additional information

[Metro M4 Express audio memory allocation failed](https://forums.adafruit.com/viewtopic.php?f=8&t=189021)

jepler, 2022-03-04:
The MP3Decoder object requires a lot of _contiguous_ memory. `gc.mem_free()` can only show how much _total_ memory is available. (and, because CircuitPython always clears the memory when you re-start code.py or enter the REPL, the free memory value shown is not actually what was free in your program when you reached line 72! It may be the case that it still won't fit no matter what you do)

Over time, the microcontroller's memory tends to become "fragmented", meaning that it has smaller stretches of free memory interspersed with used memory. I'll include a longer description of this below, but first let me tell you the recommended solution:

 * Only create an MP3Decoder object once, near the beginning of your program, and keep it through the whole program. You can open any valid .mp3 file, not necessarily the first one you'll play. You can literally make the top of your file begin
    ```
    import audiomp3
    mp3stream = audiomp3.MP3Decoder(open(audio_file["countdown"], "rb"))
    ```
    to ensure the most _contiguous_ memory is available
 * If you wish to play multiple different mp3 files, then, each time you're ready to play an MP3 file, update the MP3Decoder's `file` property with the next file to play. Then, play it. Something like:
    ```
    mp3stream.file = open(filename, "rb")
    speaker.play(mp3stream)
    ```
 * To assess memory availability during a program load, you can add a `print(gc.mem_free())` in the middle. Doing it in the REPL after a MemoryError occurs does not show a useful value.
 * Some folks have found that adding `gc.collect()` calls at strategic locations can make things better, but I don't have a specific recommendation about doing this.

All that said, nowhere in the circuitpython.org / readthedocs documentation is any of this actually suggested. Therefore, I'm titling and tagging this issue to reflect that we can improve our documentation for the next person to run into this problem!

Here's a quick overview of how memory fragmentation occurs naturally during running your program. Note that importing modules causes a lot of objects to be allocated and freed (hundreds or thousands of objects are temporarily created!)

Your microcontroller's memory when the program starts is all "free" (`_`):
```
________________
```

Your microcontroller's memory when allocating objects A, B, C, and D:
```
AAAABBBBCCCCDDDD
```

Your microcontroller memory after freeing B and D:
```
AAAA____CCCC____
```
Now, gc.mem_free() would be 8, but you could only create a new object of size 4.

FoamyGuy, 2022-03-04:
There are some details and tips about memory usage in this guide: https://learn.adafruit.com/Memory-saving-tips-for-CircuitPython/reducing-memory-fragmentation

I can't comment much on the internals of MP3 Decoder or many of the other things used by this script, but one thing I do know about that I noticed is that it's using `adafruit_display_text.label`. In my experience you can often use a little bit less RAM and still have the same visual effect by using `adafruit_display_text.bitmap_label` instead. The different ways the two are implemented internally lead to this, there is a little bit of detail in the display_text guide page here: https://learn.adafruit.com/circuitpython-display_text-library/types-of-labels

The two different classes are made to have the same API so that swapping between them is easy. I think you should be able to try switching to bitmap_label by changing the import you have from this:
```
from adafruit_display_text import label
```
to this:
```
from adafruit_display_text import bitmap_label as label
```
The rest of your code should be able to remain unchanged. 

If you are only overdrawing the available RAM by a small amount this change might knock you back under the threshold of running completely out. But if RAM situation is more dire the small amount saved by this might not be enough to allow the script to run fully.

phantomsixthplayer, 2022-03-05:
thanks @tannewt @FoamyGuy I'll give it ago.

Done a bit more testing and it just feels odd that if the rgbmatrix bit_depth is anything other than 1 and using .play(mp3stream) I get memory allocation failed, and this only happens on a Metro M4 Express, and from what I read the Feather M4 Express has the same specs but does not have this issue when I tried.

phantomsixthplayer, 2022-04-28:
Documentation regarding memory has been created here (https://learn.adafruit.com/Memory-saving-tips-for-CircuitPython?view=all)

Maybe some tips on audio.

PR #6329 (2022-05-01): Add note about memory usage recommendation to MP3Decoder
Resolves #6111 by adding a note about memory optimization and reducing memory fragmentation!
closed 2022-05-01 by ladyada