adafruit/circuitpython #6113: displayio crashes into HardFault_Handler with SH1107_addressing

deshipu, 2022-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-21-g6af4c7797 on 2022-03-04; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
import board
import displayio
import busio

_INIT = (
    b'\x2f\x00' # power on
    b'\x8f\x00' # contrast
    b'\xa6\x00' # invert off
    b'\xa4\x00' # display normal
    b'\xaf\x00' # display on
    b'\x40\x00' # start line
)

displayio.release_displays()
#spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
spi = board.SPI()
bus = displayio.FourWire(spi, chip_select=board.TX, reset=board.RX, command=board.A0)
display = displayio.Display(bus, _INIT, width=96, height=68, SH1107_addressing=True)
```

### Behavior

Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

### Description

I tried with main. 7.2.x, 7.2.0, 7.1.1 — they all crash. 6.3.0 doesn't crash.

### Additional information

Removing the `SH1107_addressing=True` parameter stops the crashing.

deshipu, 2022-03-04:
I narrowed it down to commit https://github.com/adafruit/circuitpython/commit/8f1c25c8ae

deshipu, 2022-03-04:
It  looks like the 
```C
subrectangles = rows_per_buffer/8;  // vertical (column mode) write each separately (height times)
rows_per_buffer = 8;
```
is problematic when ``color_depth`` is not 1. There should probably  be a check for this?

tannewt, 2022-03-04:
I'm not sure how color_depth factors into the code you posted. I'd trust your fix. :-)

jepler, 2022-05-03:
@deshipu what hardware does this affect, is it anything adafruit sells?

deshipu, 2022-05-03:
No, I stumbled upon this by working on a new display, and I simply forgot to set it to monochrome. Normally you would never use `SH1107_addressing` with color or grayscale displays, because it doesn't really make sense for them. So this really isn't a problem as such, other than maybe it should be an exception being raised rather than a crash. On the other hand, adding a check and an error message is going to waste additional space in the flash, so I'm not sure it even makes sense to fix it. I initially reported it, because I didn't understand what was happening, and a crash looked serious.

PR #6338 (2022-05-14): Fix display invalid sh1107
Closes: #6113
closed 2022-05-14 by dhalbert