adafruit/circuitpython #6115: Crashing into HardFault_Handler with an empty array in `argsort`

silverman, 2022-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
from ulab import numpy as np
np.argsort(np.array([]), axis=0)
```

### Behavior

The micro crashes into the HardFault_Handler:

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
```

### Description

`argsort` works fine with a non-empty array. With an empty array, I suggest it'd be better to raise `ValueError`.

jepler, 2022-03-04:
This problem doesn't seem to occur with micropython + ulab @ 4.0.0-119-g4407bec, so it may be that an update of ulab in circuitpython is needed in order to resolve this issue.

jepler, 2022-03-04:
On further investigation, I found that there is a problem in current ulab. For whatever reason, this doesn't result in a crash in our testing environment but the problem can be seen with a debugging tool known as valgrind. I've submitted a change with ulab that can fix it; we'll incorporate it in CircuitPython in a future release. See https://github.com/v923z/micropython-ulab/pull/514

```
$ valgrind ./micropython/ports/unix/micropython-2 
==2358956== Memcheck, a memory error detector
==2358956== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2358956== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==2358956== Command: ./micropython/ports/unix/micropython-2
==2358956== 
MicroPython v1.18-141-gff9c70850 on 2022-03-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> from ulab import numpy as np; np.argsort(np.array([]), axis=0)
==2358956== Use of uninitialised value of size 8
==2358956==    at 0x173AD7: numerical_argsort (numerical.c:801)
==2358956==    by 0x140665: fun_builtin_var_call (objfun.c:126)
==2358956==    by 0x139669: mp_call_function_n_kw (runtime.c:664)
==2358956==    by 0x1397EA: mp_call_method_n_kw (runtime.c:680)
==2358956==    by 0x14F887: mp_execute_bytecode (vm.c:1008)
==2358956==    by 0x14078E: fun_bc_call (objfun.c:289)
==2358956==    by 0x139669: mp_call_function_n_kw (runtime.c:664)
==2358956==    by 0x1396AF: mp_call_function_0 (runtime.c:638)
==2358956==    by 0x158FDD: execute_from_lexer (main.c:145)
==2358956==    by 0x159193: do_repl (main.c:255)
==2358956==    by 0x15982F: main_ (main.c:675)
==2358956==    by 0x15989E: main (main.c:436)
==2358956== 
==2358956== Conditional jump or move depends on uninitialised value(s)
==2358956==    at 0x173B0F: numerical_argsort (numerical.c:801)
==2358956==    by 0x140665: fun_builtin_var_call (objfun.c:126)
==2358956==    by 0x139669: mp_call_function_n_kw (runtime.c:664)
==2358956==    by 0x1397EA: mp_call_method_n_kw (runtime.c:680)
==2358956==    by 0x14F887: mp_execute_bytecode (vm.c:1008)
==2358956==    by 0x14078E: fun_bc_call (objfun.c:289)
==2358956==    by 0x139669: mp_call_function_n_kw (runtime.c:664)
==2358956==    by 0x1396AF: mp_call_function_0 (runtime.c:638)
==2358956==    by 0x158FDD: execute_from_lexer (main.c:145)
==2358956==    by 0x159193: do_repl (main.c:255)
==2358956==    by 0x15982F: main_ (main.c:675)
==2358956==    by 0x15989E: main (main.c:436)
==2358956== 
>>> 
```

dhalbert, 2022-03-10:
@jepler is this already fixed in `main`, or are you awaiting something from ulab upstream?

v923z, 2022-03-10:
@dhalbert The corresponding PR https://github.com/v923z/micropython-ulab/pull/514 crashes with the `circuitpython` documentation: https://github.com/v923z/micropython-ulab/runs/5425878718?check_suite_focus=true

jepler, 2022-04-11:
Moving this to the 8.0.0 milestone so we can simply take the latest release from ulab

v923z, 2022-04-18:
https://github.com/v923z/micropython-ulab/pull/519 implements the fix. @jepler thanks for taking care of the documentation generation issue!

PR #6567 (2022-07-11): update ulab to 5.0.9
Closes: #6115