adafruit/circuitpython #6120: seeeduino_xiao port has incorrect/insufficient LED pin definitions

ricksorensen, 2022-03-06:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Seeeduino XIAO with samd21g18
```

### Code/REPL

```python
import board
import digitalio
l1 = digitalio.DigitalInOut(board.LED)
l1.direction = digitalio.Direction.OUTPUT
# in Xiao 1/True is off, 0/False is on for LEDs
l1.value = True     
l1.value = False
l1.value = True
# first error ... same port for BLUE_LED
#                            and D13
l2 = digitalio.DigitalInOut(board.BLUE_LED)
###
l1.deinit()
l2 = digitalio.DigitalInOut(board.BLUE_LED)
 
l2 = digitalio.DigitalInOut(board.BLUE_LED)
i1.direction = digitalio.Direction.OUTPUT
i2.direction = digitalio.Direction.OUTPUT
i1.value = False
i2.value = False
l2.direction = digitalio.Direction.OUTPUT
l2.value = True      # amber LED off    
l2.value = False     # amber LED on
l2.value = True      # amber LED off
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: D13 in use
#### after l1.deinit()
#### amber LED is toggled.  Not BLUE

### Description

in circuitpython/ports/atmel-samd/boards/seeeduion_xiao/pins.c LED pins are defined:
```
    // LED pins
    { MP_ROM_QSTR(MP_QSTR_LED),   MP_ROM_PTR(&pin_PA17) }, // status
    { MP_ROM_QSTR(MP_QSTR_BLUE_LED),   MP_ROM_PTR(&pin_PA17) },
```
LED is correct wired to PA17- to an amber/yellow/orange(?) LED.  Same as D13
BLUE_LED points to the same port- PA17 

The Xiao has two different blue LEDs, wired to the UART RX and TX pins and available for a user.  These are wired to PA18 (RX) and PA19 (TX) on the chip.  A suggested modification is:
```
    // LED pins
    { MP_ROM_QSTR(MP_QSTR_LED),   MP_ROM_PTR(&pin_PA17) }, // status
    { MP_ROM_QSTR(MP_QSTR_RX_LED),   MP_ROM_PTR(&pin_PA18) },
    { MP_ROM_QSTR(MP_QSTR_TX_LED),   MP_ROM_PTR(&pin_PA19) },
```

### Additional information

I rebuilt the xiao firmware with the modifications and the LEDs work as desired I think.  Still a double definition between D13 and LED at PA17

prplz, 2022-03-06:
The suggested changes work on my end. Maybe YELLOW_LED should also be added?
D13 should be removed because it doesn't appear on silkscreen or pinout diagrams and does not have a pin.

dhalbert, 2022-03-06:
https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html
[image]

Could some one make a PR? That would be great!

ricksorensen, 2022-03-06:
I would be willing to learn that process ... I presume PR means "pull request"? (My background is "problem report")
I have little knowledge of git - in the large community development.  
I tend to do all my work locally, and have a modified build from master as of 3/5/2022 with the changes .

- is YELLOW_LED desired for  PA17?   instead of LED?
- I believe D13 is a legacy from Arduino/AVR world where 'pin' 13 is the blinking LED port.  
- are TX__LED and RX_LED okay names for the two blue LEDs?  I would prefer BLUE1 and BLUE2 for general use but schematic uses TX_ and RX_

dhalbert, 2022-03-06:
Glad you are interested! Here is a guide about making and submitting pull requests.

> - is YELLOW_LED desired for PA17? instead of LED?

We always want one LED to be available as `board.LED`, so examples that need an LED have  pin for that. It's fine to have several aliases for the same pin. You can look at other LED names on other boards to see what is standard, but `YELLOW_LED` is fine.

> - I believe D13 is a legacy from Arduino/AVR world where 'pin' 13 is the blinking LED port.

We can omit it in that case.

> - are TX__LED and RX_LED okay names for the two blue LEDs? I would prefer BLUE1 and BLUE2 for general use but schematic uses TX_ and RX_.

In general, we at least want names that correspond to the silkscreen on the board. So if they are labeled TX and RX, those are required. Then you could add the `BLUE` names; that's fine.

Put the primary name first in each set of aliases.

prplz, 2022-03-06:
Here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github/add-author-and-license-information

Neradoc, 2022-03-06:
The Feather NRF52 Express has 2 LEDS, here is how it does it:
```C
    { MP_ROM_QSTR(MP_QSTR_LED), MP_ROM_PTR(&pin_P1_15) },
    { MP_ROM_QSTR(MP_QSTR_RED_LED), MP_ROM_PTR(&pin_P1_15) },
    { MP_ROM_QSTR(MP_QSTR_D3), MP_ROM_PTR(&pin_P1_15) },

    { MP_ROM_QSTR(MP_QSTR_BLUE_LED), MP_ROM_PTR(&pin_P1_10) },
```

RX_LED and TX_LED are labeled "R" and "T" on the silkscreen (I would still use RX_LED and TX_LED then), while the yellow one is labeled "L" (perfect for board.LED). I think it's useful to have the color names too.

PR #6121 (2022-03-07): Xiao led pins
 Fixes #6120.

Modify and correct LED pin mappings and names for Seeduino SAMD21 Xiao
    Added YELLOW_LED=LED=D13,
          BLUE1_LED=RX_LED,
          BLUE2_LED=TX_LED
     Added clarifying comments about duplicate names for pins, and reverse wiring for LEDs on XIAO
closed 2022-03-07 by dhalbert