adafruit/circuitpython #6127: Switch to GCC 11.2

tannewt, 2022-03-08:
Switch to GCC 11.2. Arm now releases it on a different page: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads

The builds are generated in a new way and the libraries have debug info. The compressed archive is 400+ mb vs <200 mb for the current version we use.

Stripping the libraries makes them smaller but breaks linking for some reason.

It looks like it also increases build size. So, we make way to make our own libraries compiled with -Os (ARM compiles with -O2) or use the builds from Arch Linux that already have them as -Os. https://archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/

I started this work here: https://github.com/tannewt/circuitpython/tree/gcc11

tannewt, 2022-03-08:
The arm gcc doesn't work directly because it uses glibc 2.32+

dhalbert, 2023-10-28:
As of 9.0.0 alpha.2 we are using GCC 12

closed 2023-10-28 by dhalbert