adafruit/circuitpython #6133: RP2040 mp3 playback is static

tannewt, 2022-03-09:
Reported on the forum and confirmed by Neradoc and cater on Discord:

https://forums.adafruit.com/viewtopic.php?f=60&t=189077&p=915505#p915505

prplz, 2022-03-16:
I was able to reproduce this, but if I convert the mp3 to mono first, it goes away. Reducing the bitrate further didn't fix the issue.

prplz, 2022-03-16:
Lowering the sample rate of the mp3 down from 44100 to 16000 makes the issue go away. I think the noise i just stuttering because the processor isn't keeping up.

Alizarosebud3, 2022-03-17:
> Lowering the sample rate of the mp3 down from 44100 to 16000 makes the issue go away. I think the noise is just stuttering because the processor isn't keeping up.

How do you go about lowering the sample rate of the mp3?

gamblor21, 2022-03-17:
> How do you go about lowering the sample rate of the mp3?

I use [Audacity](https://www.audacityteam.org/). Lots of tutorials online on how to lower the sample rate and always worked well for me.

Alizarosebud3, 2022-03-17:
lowered to 16kHz and its producing the same static noise on mine. 
@prplz are you still using the following code? or a different one?

`import board
import audiomp3
import audiopwmio

audio = audiopwmio.PWMAudioOut(board.GP0)

decoder = audiomp3.MP3Decoder(open("slow.mp3", "rb"))

print("Begin playing!")

audio.play(decoder)
while audio.playing:
    pass

print("Done playing!")`

prplz, 2022-03-17:
Can you upload `slow.mp3` somewhere so I can try it?

prplz, 2022-03-17:
Here are the ones I'm using: [mp3s.zip](https://github.com/adafruit/circuitpython/files/8292800/mp3s.zip)
mono.mp3 is 44100Hz with one channel and stereo.mp3 is 16000Hz with two channels

Alizarosebud3, 2022-03-17:
yes
[slow.zip](https://github.com/adafruit/circuitpython/files/8292835/slow.zip)

I will try your sounds when I get a chance. Again I am using the STEMMA speaker from adafruit

prplz, 2022-03-17:
Your mp3 files play fine for me. I'm just plugging the PWM pin straight into the line in on my computer (I know, probably not the best idea).

Could this maybe be an issue that the speaker has with PWM? I will see if anyone can do some more testing with other boards and PWM.

Alizarosebud3, 2022-03-17:
hmm your stereo.mp3 works for me, but not my slow16kHz.mp3. 

The stereo.mp3 plays very quietly and the sound goes away when I try changing the volume on the STEMMA speaker itself.

how do you connect it straight into the computer line is that pin 0 on the raspberry pi pico?

prplz, 2022-03-17:
@Neradoc reports that slow.mp3 plays ok if you adjust the volume pot on the speaker, can you try that?

Alizarosebud3, 2022-03-18:
Got it! Thank you for assisting me !

dhalbert, 2022-03-22:
Let's document in the shared-bindings docs that MP3 problems could be due to the bit rate being too high.

closed 2022-04-06 by dhalbert

ghstwhl, 2023-01-13:
While I know this issue was closed several months ago, I am having this exact same problem.  

In my case, I am using a Feather RP2040 running CircuitPython 7.3.3, and using a Prop-Maker feather as my amplifier.  I have downloaded the stereo.mp3 file that folks in this thread say works for them, and it has a lot of static for me.  My code is pretty generic:

`import board
import audiomp3
import audiopwmio
import digitalio

audio = audiopwmio.PWMAudioOut(board.A0)

enable = digitalio.DigitalInOut(board.D10)
enable.direction = digitalio.Direction.OUTPUT
enable.value = True

decoder = audiomp3.MP3Decoder(open("stereo.mp3", "rb"))

audio.play(decoder)
while audio.playing:
    pass

print("Done playing!")`

PR #6245 (2022-04-06): Document TileGrid limitation to tile values 0..255 & improve mp3 doc
Closes #6149. Closes #6133.