adafruit/circuitpython #6136: ESP-C3-32S Error creating custom BLE Service

rlangoy, 2022-03-10:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; AITHinker ESP32-C3S_Kit with ESP32-C3FN4
```

### Code/REPL

```python
from adafruit_ble.services import Service
from adafruit_ble.uuid import StandardUUID

class MyService(Service):
    uuid = StandardUUID(0xA000) # (0xA000 - my/Custom Service ID)
       
mySevice= MyService()
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "adafruit_ble/services/__init__.py", line 36, in __init__
NotImplementedError: 

### Description

Looks like StandardUUID is not working.
Code works on the nRF8240-Dongle

### Additional information

Boot info:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x1374
load:0x403ce000,len:0x8f0
load:0x403d0000,len:0x2c4c
entry 0x403ce000
Adafruit CircuitPython 7.2.0 on 2022-02-24; AITHinker ESP32-C3S_Kit with ESP32-C3FN4
Board ID:ai_thinker_esp32-c3s

ladyada, 2022-03-10:
not surprising because we haven't added native BLE support for the ESP32 series :)
https://github.com/adafruit/circuitpython/issues/5926
closing as a duplicate, please subscribe to the above for updates!

closed 2022-03-10 by ladyada

rlangoy, 2022-03-10:
Sorry for reporting a known bug. But thanks for replying :)

tannewt, 2022-03-10:
BLE support is not complete. It supports GATT client but not GATT server.

casainho, 2023-01-17:
> BLE support is not complete. It supports GATT client but not GATT server.

Is anyone working on this, to implement the GATT server to ESP3-S3 or C3? if so, I can help doing some tests.

dhalbert, 2023-01-17:
See https://github.com/adafruit/circuitpython/issues/5926#issuecomment-1074414184