adafruit/circuitpython #6146: Pi Pico Unable to setup SPI

charles-d-burton, 2022-03-12:
### CircuitPython version

```python
CircuitPython 7.2.0 on Raspberry Pi Pico RP2040
```

### Code/REPL

```python
Pull directly from the RFM69HCW tutorial on your website.  Code adapted to use the Pico pins

# SPDX-FileCopyrightText: 2018 Tony DiCola for Adafruit Industries
# SPDX-License-Identifier: MIT

# Simple example to send a message and then wait indefinitely for messages
# to be received.  This uses the default RadioHead compatible GFSK_Rb250_Fd250
# modulation and packet format for the radio.
import board
import busio
import digitalio

import adafruit_rfm69

# Define radio parameters.
RADIO_FREQ_MHZ = 915.0  # Frequency of the radio in Mhz. Must match your
# module! Can be a value like 915.0, 433.0, etc.

# Define pins connected to the chip, use these if wiring up the breakout according to the guide:
CS = digitalio.DigitalInOut(board.GP17)
RESET = digitalio.DigitalInOut(board.GP20)
# Or uncomment and instead use these if using a Feather M0 RFM69 board
# and the appropriate CircuitPython build:
# CS = digitalio.DigitalInOut(board.RFM69_CS)
# RESET = digitalio.DigitalInOut(board.RFM69_RST)

# Define the onboard LED
LED = digitalio.DigitalInOut(board.LED)
LED.direction = digitalio.Direction.OUTPUT

# Initialize SPI bus.
spi = busio.SPI(board.GP18, board.GP16, board.GP19)

# Initialze RFM radio
rfm69 = adafruit_rfm69.RFM69(spi, CS, RESET, RADIO_FREQ_MHZ)
```

### Behavior

Complains about invalid pins
Traceback (most recent call last):
  File "code.py", line 31, in <module>
ValueError: Invalid pins

### Description

It's pretty self-explanatory. I'm trying to initialize the SPI bus on my Pico and cannot get it to work.

Neradoc, 2022-03-13:
It looks like you swapped MOSI and MISO. GP19 is SPI0 TX (MOSI) and GP16 is SPI0 RX (MISO). So you get "invalid pins".
The signature is `busio.SPI(clock, MOSI, MISO)`

charles-d-burton, 2022-03-13:
That helped for sure, ran into another error but it was soldering flux that didn't quite get washed off.
```Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Temperature: 22.0C
Frequency: 914.999mhz
Bit rate: 250.0kbit/s
Frequency deviation: 250000.0hz
Sent hello world message!
Waiting for packets...```

closed 2022-03-13 by charles-d-burton