adafruit/circuitpython #6147: Asyncio error with custom built fw

muranyia, 2022-03-13:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-38-g862210b3f-dirty on 2022-03-13; Adafruit Trinket M0 with samd21e18
```

### Code/REPL

```python
import asyncio
async def foo(bar):
  print("foobar")
```

### Behavior

``` python
Traceback (most recent call last):
  File "code.py", line 2
SyntaxError: invalid syntax
```

### Description

I managed to squeeze asyncio into a fw build for the trinket_m0 using `FROZEN_MPY_DIRS += $(TOP)/../Adafruit_CircuitPython_Bundle/libraries/helpers/asyncio`:
```
make -j4 BOARD=trinket_m0
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build-trinket_m0/genhdr
FREEZE ../../../Adafruit_CircuitPython_Bundle/libraries/drivers/24lc32 ../../../Adafruit_CircuitPython_Bundle/libraries/helpers/asyncio
MKMANIFEST ../../../Adafruit_CircuitPython_Bundle/libraries/drivers/24lc32 ../../../Adafruit_CircuitPython_Bundle/libraries/helpers/asyncio
(cd build-trinket_m0/frozen_mpy && find * -name \*.py -exec printf 'freeze_as_mpy("frozen_mpy", "%s")\n' {} \; )> build-trinket_m0/manifest.py.tmp && mv -f build-trinket_m0/manifest.py.tmp build-trinket_m0/manifest.py
GEN build-trinket_m0/genhdr/moduledefs.h
QSTR updated
MPY adafruit_24lc32.py
MPY asyncio/task.py
MPY asyncio/lock.py
MPY asyncio/event.py
MPY asyncio/core.py
MPY asyncio/manifest.py
MPY asyncio/stream.py
MPY asyncio/__init__.py
MPY asyncio/funcs.py
GEN build-trinket_m0/frozen_content.c

187304 bytes used, 856 bytes free in flash firmware space out of 188160 bytes (183.75kB).
9872 bytes used, 22896 bytes free in ram for stack and heap out of 32768 bytes (32.0kB).

Converted to uf2, output size: 374784, start address: 0x2000
```
but the `async` keyword triggers an error.

jepler, 2022-03-13:
You need to enable https://github.com/adafruit/circuitpython/blob/f13d2187498974a6f5ad4042417a9be91862a8eb/py/circuitpy_mpconfig.mk#L35

muranyia, 2022-03-13:
Thanks. Tried adding `MICROPY_PY_ASYNC_AWAIT = 1` to `mpconfigboard.h` which didn't work, so I edited `circuitpy_mpconfig.mk`.
 The build was successful but at runtime `asyncio` failed on missing module `adafruit_ticks`. I added it and then it was missing `select` and later `traceback` which I enabled all in `mpconfigboard.h`. It still complains about missing `traceback` though.

dhalbert, 2022-03-13:
> Thanks. Tried adding `MICROPY_PY_ASYNC_AWAIT = 1` to `mpconfigboard.h` which didn't work, so I edited `circuitpy_mpconfig.mk`.

Add it to `mpconfigboard.mk`, but it sounds like we need to add some dependencies as well in `circuitpy_mpconfig.mk` or `.h`

jepler, 2022-03-14:
Yes, in general the build system does not account for changes to '.mk' files so you should `make BOARD=??? clean` after changing an mk file.

deshipu, 2022-03-14:
Btw, if you want to make a clean build in one command, you can do `make BOARD=??? clean all` — "all" is the command it does by default when no command is specified, and you can specify multiple commands to be done, so this will do clean followed by a build.

jepler, 2022-03-14:
> Btw, if you want to make a clean build in one command, you can do `make BOARD=??? clean all` — "all" is the command it does by default when no command is specified, and you can specify multiple commands to be done, so this will do clean followed by a build.

Does this work reliably when using "-j" to perform a parallel build? If so ... how ?

muranyia, 2022-03-17:
Thanks, I `make BOARD=??? clean` after each build. 
I restarted from scratch and managed to build the firmware with `asyncio` working. I had to tinker with `circuitpy_mpconfig.mk` which is not optimal (as opposed to a purely `mpconfigboard.mk` solution) but I understand you cannot easily serve all kinds of custom builds and the stock build types at the same time.

BTW `make -j4 BOARD=trinket_m0 clean all` doesn't work...

closed 2022-03-17 by muranyia

tannewt, 2022-03-17:
Generally, I only clean if I get a QSTR error or change a makefile. The rest of the time, you should be able to do incremental builds.