adafruit/circuitpython #6148: MagTag battery voltage incorrect in 7.2.0

kaptajnen, 2022-03-13:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit MagTag with ESP32S2
```

### Code/REPL

```python
import analogio
import board

battery_value = (analogio.AnalogIn(board.BATTERY).value / 65535.0) * 3.3 * 2
```

### Behavior

It seems the battery voltage calculated from the code is longer correct in CircuitPython 7.2.0 (code based on https://github.com/adafruit/Adafruit_CircuitPython_MagTag/blob/main/adafruit_magtag/peripherals.py).

Using 7.0.0 it returns around 3.7 but with 7.2.0 it returns 5.4 which is considerably more than the capacity of the battery (3.7v 420mAh)

DavePutz, 2022-03-13:
It looks like the ESP IDF function esp_adc_cal_raw_to_voltage() may be returning a different value. Running on CP 7.1 (using USB power, which also shows the reported issue) I see a voltage value returned of 2114. On CP 7.2, I see a value returned of 2613. After all the adjusting calculations; 2114 converts to 4.22 volts and 2613 converts to 5.22 volts.

DavePutz, 2022-03-15:
Further investigation shows that the ESP IDF function get_calibration_offset() is returning a different value than it previously did. This is used to calculate the returned voltage value, and is the cause of the change in CP7.2, which now uses a newer IDF. 
It looks like there was a significant change in the IDF when it comes to calculating the calibration offset. 
@tannewt, do we need to change our calculations in CircuitPython to adjust for this?

tannewt, 2022-03-15:
@DavePutz I have no idea. You are the expert at this point. :-)

DavePutz, 2022-03-30:
Well, according to the Espressif ESP IDF reference manual (at https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/system/esp_err.html#_CPPv49esp_err_t) the ADC on an ESP32S2 is only going to be accurate up to 2500mV. My tests seem to show that past that point the IDF returns a max value. In my tests, applying a voltage of 4.0 to a GPIO pin attached to ADC1, I saw a computed voltage of 5.17395 even going back as far as CircuitPython 6.3.0. So, I'm not sure that the value returned from a battery or USB power will ever be able to be very accurate.

ladyada, 2022-03-30:
the magtag divides the battery voltage by 2 - so it'll be 2.1V maximum into the ADC pin

DavePutz, 2022-03-31:
ESP IDF Version 4.4 has replaced the ADC calibration algorithm (commit 2f874db on Feb 6) , and running a native binary on the MagTag this seems to have resolved the issue. I see a reported voltage of 4.09 on a battery the multimeter shows at 4.1. I am not sure what the plan is for moving CircuitPython to ESP-IDF 4.4 (currently using 4.3, I believe)

dhalbert, 2022-03-31:
> I am not sure what the plan is for moving CircuitPython to ESP-IDF 4.4 (currently using 4.3, I believe)

`main` is on 4.4, so maybe this will just fix itself. I'll be making a 7.3.0-beta.0 release soon.

DavePutz, 2022-04-01:
Tested 7.3.0-beta.0 and the incorrect voltages still exist. Will investigate further.

DavePutz, 2022-04-02:
@dhalbert - I built the example single_read adc application in the ESP IDF from the current CP main tree. It always returns a reading of 8191, which is the max adc reading, and which is what CP is getting to calculate the voltage incorrectly. Building from the main tree of the current espressif/esp-idf on Github yields correct readings. So, is it worth binary searching the 400-some commits between our esp-idf and the current IDF master to figure this one out? Any easier way come to mind?

dhalbert, 2022-04-03:
Does it work for the espressif 4.4 branch, or only their `master`?

DavePutz, 2022-04-03:
using the branch release/v4.4 fails as well. I'll try narrowing down their fix.

closed 2022-04-06 by dhalbert

kaptajnen, 2022-04-14:
Verified it works with latest beta. Thanks for fixing this!

PR #6246 (2022-04-06): Don't reset GPIO4 on the MagTag (used for voltage monitoring)
Fixes #6148. Since the re-doing of Espressif GPIO resets done as part of PR #5892 the voltage monitoring done on the MagTag by pin GPIO4 has been always returning high values. This fix avoids resetting that pin on the MagTag and allows correct readings of the voltage.