adafruit/circuitpython #6149: displayio.TileGrid does not document limitation to 255 frames

lowasser, 2022-03-13:
[TileGrid.__setitem__](https://docs.circuitpython.org/en/latest/shared-bindings/displayio/index.html#displayio.TileGrid.__setitem__) enforces, but does not document, a limitation to 255 tile elements.

This limitation is not great for my needs, but the failure to document what's going on, or provide a more informative error message, led to a merry hour of debugging.  TileGrid also doesn't allow introspection in any way of how many tiles are available to be addressed, so I was extremely confused until I finally just tried "keep setting tiles up to the highest index valid until we crash," at which point I discovered the crash at 256, which immediately suggested a problem with TileGrid, not the design of the bitmap I was indexing into.

lowasser, 2022-03-13:
The enforcement takes place here: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/displayio/TileGrid.c#L434

tannewt, 2022-03-14:
When you say crash do you mean safe mode or a Python exception? If exception, what is the current exception that we raise?

That shared bindings file is where the docs come from so it'd be awesome to add a note to it there.

lowasser, 2022-03-14:
A Python exception -- was the link I provided to the code unclear?  It raises the ValueError specified on line 435 there.

lowasser, 2022-03-14:
(Sorry, I don't mean to sound accusatory; I just meant that I linked that code to demonstrate that that was where the tile number got rejected, and to point out where I was getting the relevant error.)

PR #6245 (2022-04-06): Document TileGrid limitation to tile values 0..255 & improve mp3 doc
Closes #6149. Closes #6133.
closed 2022-04-06 by dhalbert