adafruit/circuitpython #6158: mpy-cross fails to build due to "unused-but-set-variable" warnings on clang 13

nicolasff, 2022-03-15:
Compiling `mpy-cross` with clang 13 currently produces a few warnings, which are incompatible with the `-Werror` flag configured in its `.mk` file.

**TL;DR:** Clang 13 adds a warning for "variable is set but its value is not used", which causes `mpy-cross` to fail to build. Adafruit's GitHub Actions use Clang 12, which does not include this warning.

This is reproducible on `main`, and `mpy-cross` in tags at least as far as 7.0.0. I see that there _are_ successful builds in GitHub Actions for recent commits which include an `mpy-cross-mac` step that does succeed, e.g. [this one](https://github.com/adafruit/circuitpython/runs/5547495073?check_suite_focus=true) from a few hours ago, but they are using clang 12.0.0.

With this setup:
- macOS 11.6 (20G165)
- Homebrew clang version 13.0.1, target: x86_64-apple-darwin20.6.0 (this is really the only version that matters here)
- cmake version 3.21.3

I just pulled a fresh checkout of CircuitPython (`HEAD` commit f13d2187498974a6f5ad4042417a9be91862a8eb), and ran:

```sh
make fetch-submodules
cd mpy-cross
make V=2
```

The build fails with a few warnings (see [full output here](https://gist.github.com/nicolasff/c13d371c09e2db25b1a7fad2885146a2)):

```none
clang -I. -I.. -Ibuild -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_VFS_LFS1=0 -DMICROPY_VFS_LFS2=0 -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -c -MD -o build/py/objgenerator.o ../py/objgenerator.c
../py/objgenerator.c:83:54: error: variable 'n_pos_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                     ^
../py/objgenerator.c:83:66: error: variable 'n_kwonly_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                 ^
../py/objgenerator.c:83:81: error: variable 'n_def_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                                ^
../py/objgenerator.c:83:21: error: variable 'n_exc_stack_unused' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                    ^
../py/objgenerator.c:83:41: error: variable 'scope_flags' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                        ^
5 errors generated.
make: *** [build/py/objgenerator.o] Error 1
```

All of these warnings are due to `-Wunused-but-set-variable` – a variable is set to a value, but later unused; so setting the value is flagged as wasteful computation by this compiler warning.

Looking at `../py/objgenerator.c` line 83, all of the "unused but set" variables are being passed to the `MP_BC_PRELUDE_SIG_DECODE_INTO` macro, which I can see being expanded if I re-run the failing `clang` command with `-E` to run the pre-processor stage. It results in:
```c
size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
do { uint8_t z = *(ip)++; n_state = (z >> 3) & 0xf; n_exc_stack_unused = (z >> 2) & 0x1; scope_flags = 0; n_pos_args = z & 0x3; n_kwonly_args = 0; n_def_args = 0; for (unsigned n = 0; z & 0x80; ++n) { z = *(ip)++; n_state |= (z & 0x30) << (2 * n); n_exc_stack_unused |
       │ = (z & 0x02) << n; scope_flags |= ((z & 0x40) >> 6) << n; n_pos_args |= (z & 0x4) << n; n_kwonly_args |= ((z & 0x08) >> 3) << n; n_def_args |= (z & 0x1) << n; } n_state += 1; } while (0);
size_t n_exc_stack = 0;
```

These variables \*are\* indeed "set but not used". If you follow what happens to any of them, say `n_pos_args`, the macro does:
```c
n_pos_args = z & 0x3;
n_pos_args |= (z & 0x4) << n; // in the for loop
```
but then the value that was computed is not used anywhere else.

`mpy-cross.mk` defines its `CFLAGS` to fail on all warnings: https://github.com/adafruit/circuitpython/blob/f13d2187498974a6f5ad4042417a9be91862a8eb/mpy-cross/mpy-cross.mk#L29-L30

**Why this fails on clang 13.0.0:**

This warning for "variable is set but unused" seems to have been added relatively recently in clang, I found [this thread](https://reviews.llvm.org/D100581) proposing it in April 2021. The version of clang I am using was released on [Feb 2, 2022](https://github.com/llvm/llvm-project/releases/tag/llvmorg-13.0.1) while the successful GitHub Action jobs I found were using "Apple clang version 12.0.0 (clang-1200.0.32.29)" ([logged here](https://github.com/adafruit/circuitpython/runs/5547495073?check_suite_focus=true#step:8:10)), which was released on [April 14, 2021](https://github.com/llvm/llvm-project/releases/tag/llvmorg-12.0.0).

cc @tannewt since I see this macro was brought in on April 23, 2021 with b35fa44c8adeae93000ca9bc413ea6e64456fd94 in your "[Merge MicroPython 1.12 into CircuitPython](https://github.com/adafruit/circuitpython/commit/b35fa44c8adeae93000ca9bc413ea6e64456fd94#diff-0275aedbdf234efaa8c0331d6784ec24a771695afba2b955cf1e55990f5fc7a2R84)".

The simplest fix for now might be just to add `-Wno-error=unused-but-set-variable` to `CFLAGS` in the CMake file. With this change, the warnings are logged but the build continues:

```diff
 # compiler settings
-CWARN = -Wall -Werror
+CWARN = -Wall -Werror -Wno-error=unused-but-set-variable
```

dhalbert, 2022-03-15:
Thanks for the report. We try to avoid blanket disabling of warnings as errors, so that we can catch new unintentional errors. So in this case we could add `void (n_pos_args);` etc to the code at that particular place.

nicolasff, 2022-03-15:
Good point.
Yes, adding this to all 5 variables would work; it might make sense to do it in the macro itself in case it's reused.

tannewt, 2022-03-15:
@nicolasff Are you able to do a PR for this? You are in the best position to test and fix. Thanks!

nicolasff, 2022-03-16:
@tannewt opened! I noticed that the same issue had actually been addressed with other macros already, e.g. in cfd08448a1cba0040ad3c62bce02b5f8bd76dd1d.

I hit more warnings somewhere else while building `mpy-cross` (negative array bounds with `sp[-MP_OBJ_ITER_BUF_NSLOTS + 1]` in `py/vm.c`), I'll investigate and will open a separate issue once I understand exactly what's going on. I looks like clang being too strict and not recognizing a previous pointer increment.

commit 65739813 (2022-03-16): Explicitly ignore unused variables (fixes #6158)

A macro in py/bc.h declares five variables that are used to hold data
temporarily, without their values being used after the assignments. This
causes "unused-but-set-variable" warnings in clang 13. We mark these
variables as explicitly ignored to avoid this new warning.
commit 73aeadb3 (2022-03-16): Merge pull request #6163 from nicolasff/clang-13-mpy-cross

Explicitly ignore unused variables (fixes #6158)
closed 2022-03-16 by tannewt