adafruit/circuitpython #6159: 7.2.2 Hangs After Thonny Connects On Adafruit RP2040 Feather

eightycc, 2022-03-15:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.2 on 2022-03-15; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
No code or input needed to reproduce the problem.
```

### Behavior

```
Connection lost (read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?))

Use Stop/Restart to reconnect.
```
This message occurs shortly (<30 seconds) after connecting ```thonny 3.2.7``` to an ```Adafruit RP2040 Feather```, and then the  device hangs. 

### Description

Reverting to ```Adafruit CircuitPython 7.2.1 on 2022-03-14; Adafruit Feather RP2040 with rp2040``` corrects the problem.

dhalbert, 2022-03-15:
And what is the host computer OS and version of Thonny you are using?

tannewt, 2022-03-15:
@aivarannamaa FYI. We switched our autoreload a bit and may have broken Thonny.

eightycc, 2022-03-15:
Seeing the problem on 7.2.2, not seeing it on 7.2.1, Thonny version 3.2.7, Ubuntu 20.04.4 LTS: ```Linux ub2004 5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:18:32 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux```.

aivarannamaa, 2022-03-16:
Unfortunately I'm not able to reproduce this with the exact same setup: Thonny version 3.2.7, Ubuntu 20.04.4 LTS, RP2040 Feather with CP 7.2.2. I also tried Thonny 3.3.13 and RPi Pico with the same result.

I do get this error when I disconnect the USB cable, but this is expected.

dhalbert, 2022-03-16:
I see this, noted in #6159:

With 7.2.2, ReloadException is printed when a file is written when code.py is not running or does not exist. Reported in #6165, and probably subsumes this.

```py
>>> 
ReloadException: 
>>> 
ReloadException: 
>>> 
ReloadException: 
>>> 
ReloadException: 
>>> 
ReloadException: 
```

PR #6168 (2022-03-16): fix ReloadException when in REPL 
Do not raise ReloadException if in the REPL.
Also undo autoreload suspend before triggering autoreload. This is not imperative, but if `autoreload_trigger()` were to check the suspend state (it does not currently), it would have the wrong value. This was originally necessary when I was testing a different fix.

- Fixes #6159
- Fixes #6165 

Please test! I tested on macOS and Linux with a Metro M4 and a Feather RP2040,
modifying a running code.py, copying a 40kB file while running, and also editing code.py and copying a file when in the REPL.
closed 2022-03-16 by eightycc