adafruit/circuitpython #6161: External clock input support

TravisJoe, 2022-03-15:
In the past I have done a lot of custom boards using Micropython, lately I have been moving to SAME51 chips and one client is wanting to use Circuitpython. 

So far the main limitation I have found is support setting external clock inputs, and allowing them to be just a clock input (bypass mode) so I can use an oscillator instead of a crystal. There are two main clock inputs so having that selectable would be good also. 

Ideally this would be just another value passed in the board config. I worked with Damien at Micropython to add this with the STM32 chips, which seemed fairly painless. It looks slightly more complicated with the SAM chips, but the framework for managing the clocks appears there, just not easily configurable per board. 

I do not really have the time or knowledge to implement this quickly and well, but more than willing to support the efforts even if it means paying for labor and attention. 

It looks like much of what is needed is in the SAM peripheral repo. I am still trying to piece together how this is implemented and how I would modify it for a prototype. 
https://github.com/adafruit/samd-peripherals/blob/main/samd/sam_d5x_e5x/clocks.c

tannewt, 2022-03-16:
You found the right repo. We're happy to review and merge pull requests.

Us Adafruit-funded folks don't have the time to add this either though. Others can speak up if they'd like to help. The Adafruit jobs board is another place to get the offer out there.

It'd be awesome to have you available as a contractor for CircuitPython. You aren't the first person to want to hire someone to make additions.

TravisJoe, 2022-03-16:
I am definitely still getting up to speed with circuitpython and developing on the build system. But I do work with others that are. I think with minimal pointers on review I can help get this developed. 

I feel like github search in repo is less powerful than it used to be, as I am trying to search for references to function calls in the clocks section so I can continue to piece together how to modify this, both to test it works and to suggest implementation changes for a pull request. 

Where are the places I need to be most concerned about in regards to checking how the clock is implemented in the build system?

dhalbert, 2022-03-16:
> I feel like github search in repo is less powerful than it used to be, as I am trying to search for references to function calls in the clocks section so I can continue to piece together how to modify this, both to test it works and to suggest implementation changes for a pull request.

https://cs.github.com is an invite-only codesearch beta that is very nice. You might want to get on the list.

@tannewt and I use `ag` or similar `grep -r` tools to search through the source tree.

TravisJoe, 2022-03-16:
For reference, this comes from the MPLAB configuration, which shows that the basic config of setting enabling the XOSC0 or XOSC1 is enabled for both crystal and oscillator, just the mode is set to either external clock or crystal.  

[image]

tannewt, 2022-03-17:
I'd suggest generating the code from MPLAB for the setting you want and comparing to what we have in samd-peripherals. Generally changing clocks means you need to ensure the dependent clocks (especially USB) are the correct value.

TravisJoe, 2022-03-17:
I agree, and that is what I started. One good thing about the external clock vs crystal is basically all the clock settings are the same except if you need to drive the crystal or not. I am hoping this could be a single flag.

One thing that is odd to me is CAN at speeds greater than ~125K wants a much higher accuracy clock, but the Feather M4 CAN only as the 32kHz crystal on it. So I am not sure if it actually fits into the expected tolerance for CAN at 250K and above.

tannewt, 2022-03-18:
> I agree, and that is what I started. One good thing about the external clock vs crystal is basically all the clock settings are the same except if you need to drive the crystal or not. I am hoping this could be a single flag.

Great! Simple is good.

> One thing that is odd to me is CAN at speeds greater than ~125K wants a much higher accuracy clock, but the Feather M4 CAN only as the 32kHz crystal on it. So I am not sure if it actually fits into the expected tolerance for CAN at 250K and above.

/shrug I've never done CAN myself. Happy to see things improved in general. Nothing is ever perfect.

TravisJoe, 2022-03-30:
Okay, finally I am able to put some more time on this. 

When using MPLAB X this is the initialize clock settings for when a crystal is expected. 

```c
static void OSCCTRL_Initialize(void)
{

    /****************** XOSC1 Initialization   ********************************/

    /* Configure External Oscillator */
    OSCCTRL_REGS->OSCCTRL_XOSCCTRL[1] = OSCCTRL_XOSCCTRL_STARTUP(0) | OSCCTRL_XOSCCTRL_IMULT(3) | OSCCTRL_XOSCCTRL_IPTAT(2) | OSCCTRL_XOSCCTRL_XTALEN_Msk | OSCCTRL_XOSCCTRL_ENABLE_Msk;
    while((OSCCTRL_REGS->OSCCTRL_STATUS & OSCCTRL_STATUS_XOSCRDY1_Msk) != OSCCTRL_STATUS_XOSCRDY1_Msk)
    {
        /* Waiting for the XOSC Ready state */
    }
}
```

And when I change it back to using a external clock instead of driving a crystal it is this. 

```c
static void OSCCTRL_Initialize(void)
{

    /****************** XOSC1 Initialization   ********************************/

    /* Configure External Oscillator */
    OSCCTRL_REGS->OSCCTRL_XOSCCTRL[1] = OSCCTRL_XOSCCTRL_STARTUP(0) | OSCCTRL_XOSCCTRL_IMULT(3) | OSCCTRL_XOSCCTRL_IPTAT(2) | OSCCTRL_XOSCCTRL_ENABLE_Msk;
    while((OSCCTRL_REGS->OSCCTRL_STATUS & OSCCTRL_STATUS_XOSCRDY1_Msk) != OSCCTRL_STATUS_XOSCRDY1_Msk)
    {
        /* Waiting for the XOSC Ready state */
    }
}
```

So the big difference is are 
`SCCTRL_XOSCCTRL_STARTUP(0) | OSCCTRL_XOSCCTRL_IMULT(3) | OSCCTRL_XOSCCTRL_IPTAT(2) | OSCCTRL_XOSCCTRL_ENABLE_Msk;`
vs
`OSCCTRL_XOSCCTRL_STARTUP(0) | OSCCTRL_XOSCCTRL_IMULT(3) | OSCCTRL_XOSCCTRL_IPTAT(2) | OSCCTRL_XOSCCTRL_XTALEN_Msk | OSCCTRL_XOSCCTRL_ENABLE_Msk;`

Or more specifically just the OSCCTRL_XOSCCTRL_XTALEN_Msk config masked into "OSCCTRL_REGS->OSCCTRL_XOSCCTRL[1]"

In my examples above I am using clock input 1 instead of 0.

TravisJoe, 2022-03-30:
So good thing so far is it does look like it is really only one mask bit change.

The hard part is I am not familiar with how this repo is setup, and what pulls from where... so I am doing a lot of digging and trying to piece this together. 

Looks like this is where some of the initial clock configs for XOSC0 and XOSC1 are done. It is in the asf4 sub module. 
https://github.com/adafruit/asf4/blob/84f56af13292d8f32c40acbd949bde698ddd4507/same51/hpl/oscctrl/hpl_oscctrl.c

What feels very odd is there is a lot exposed for the 32k clock in SAMD-Peripheral library, but it seems to ignore for the most part abstracting XOSC0 and XOSC1 which I feel are critical tools on anything outside of a development board. 
https://github.com/adafruit/samd-peripherals/blob/main/samd/sam_d5x_e5x/clocks.c

The other issue I see is that there is a board definition setting for "has_crystal" which is extremely vague and limiting. This should more likely be broken down as follow

"has_xosc32k" or "has_external_osc32k"

Plus then you can more easily do
"has_xosc0" or "has_external_xosc0"
and
"has_xosc1" or "has_external_xosc1"

Then you would have the 3 options for if it is an external clock source or crystal(likely default to crystals because most people use crystals)

"external_xosc0_is_crystal" 
"external_xosc1_is_crystal" 
"external_xosc32k_is_crystal" 

This should not cause any new clock configuration otherwise, just the source. Plus it is a lot easier to force somebody on a frequency, 12MHz for example, than it is to say you can or cannot use an external clock.

tannewt, 2022-05-09:
/me is back from paternity leave

I don't think we use much from ASF4 and rely on samd-peripherals instead.

The functionality you are suggesting makes sense. We don't have it currently because no one has needed it.

TravisJoe, 2022-05-10:
I did get initial functionality working with external oscillator (bypass crystal mode) by modifying one line in this file
hpl_osc32kctrl_config.h

One issue is this description I do not feel is correct. This is for enabling the external clock source, instead assuming crystal. MPLABs also has ad descriptions but when comparing changes between external clock and external crystal this is all that changes. 

```c
// <q> Crystal connected to XIN32/XOUT32 Enable
// <i> Indicates whether the connections between the I/O pads and the external clock or crystal oscillator is enabled or not
// <id> xosc32k_arch_xtalen
#ifndef CONF_XOSC32K_XTALEN
#define CONF_XOSC32K_XTALEN 1  // MODIFIED
#endif
```
 
 So next step is adding a flag for the 32k being being a external clock source, or crystal. The normal clock input does not seem to have the framework setup in the ecosystem to work similar with such minor changes... but right now I will take what I can get.

tannewt, 2022-05-10:
Great! You should be able to define `CONF_XOSC32K_XTALEN` earlier then to override this version.

dhalbert, 2023-10-28:
Implemented by #8183.

closed 2023-10-28 by dhalbert