adafruit/circuitpython #6169: CPX with 7.2.2 has less available RAM than expected on boot

ladyada, 2022-03-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.2 on 2022-03-15; Adafruit CircuitPlayground Express with samd21g18
```

### Code/REPL

```python
import time
import gc
print(gc.mem_free())
import digitalio
import touchio
import busio
import board
import usb_midi
import neopixel

import adafruit_lis3dh
print(gc.mem_free())
from adafruit_midi import MIDI

from adafruit_midi.note_on          import NoteOn
from adafruit_midi.control_change   import ControlChange
from adafruit_midi.pitch_bend       import PitchBend
gc.collect()
print(gc.mem_free())
```

### Behavior

code.py output:
14144
11840
3472

### Description

scott thinks there should be more SRAM available on the CPX during boot...and yet there is not! a mystery :)

tannewt, 2022-03-17:
Here is what 4.1.2 has available in the REPL:
```
Adafruit CircuitPython 4.1.2 on 2019-12-18; Adafruit CircuitPlayground Express with samd21g18
>>> import gc
>>> gc.mem_free()
21168
```

And 7.2.2:
```
Adafruit CircuitPython 7.2.2 on 2022-03-15; Adafruit CircuitPlayground Express with samd21g18
>>> import gc
>>> gc.mem_free()
17808
```

I glanced at the map file and think this is a combination of TinyUSB updates and the switch to split stack (c and pystacks).

The difference with code.py is the ram used to hold the code.py code itself.

dhalbert, 2023-10-28:
Closing as explained.

closed 2023-10-28 by dhalbert