adafruit/circuitpython #6185: Implement slice object and string slicing

phantomsixthplayer, 2022-03-22:
When using [label_direction="RTL"](https://docs.circuitpython.org/projects/display_text/en/latest/api.html) the label is inverted, would it be possible to implement the blow?

`time.text = str(gatime - int(time() - gatimer))[::-1]
time.text = str(score)[::-1]`

`NotImplementedError: only slices with step=1 (aka none) are supported`

tannewt, 2022-03-23:
The error comes from here: https://github.com/adafruit/circuitpython/blob/4465adfe01c0321adb27cde83afde24de8c4a210/py/objstr.c#L455

It doesn't look like there is a richer implementation that is turned off so this will take more work than just build tweaks.

deshipu, 2022-03-23:
I think you could always do `reversed(text)`.