adafruit/circuitpython #6186: mDNS safe mode - find

anecdata, 2022-03-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-52-g4465adfe0 on 2022-03-21; FeatherS2 with ESP32S2
(this is the version that corresponds to the mDNS merge)
```

### Code/REPL

```python
import wifi
import mdns
from secrets import secrets

print(" Connecting", end="\r")
wifi.radio.connect(secrets['ssid'], secrets['password'])
print(wifi.radio.ipv4_address)

print("Starting mDNS server...")
server = mdns.Server(wifi.radio)
service_types = ("_http", )

protocols = ("_tcp", )
for service_type in service_types:
    print(service_type, end=" ")
    for protocol in protocols:
        print(protocol)
        remote_service_tuple = server.find(service_type=service_type, protocol=protocol, timeout=5)
        for remote_service in remote_service_tuple:
            print(remote_service.hostname, remote_service.instance_name, remote_service.service_type, remote_service.protocol, remote_service.port)
```

### Behavior

```
code.py output:
Connected.    
Starting mDNS server...
_upnp _udp

[tio 21:28:16] Disconnected
[tio 21:28:20] Connected
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Pattern seems to be Safe Mode when trying to `find` a service type and protocol that do not exist on the network.

tannewt, 2022-03-23:
What exact code are you running? The prints don't match the code. (udp vs tcp).

I'm able to run a modified version ok:

```
code.py output:
192.168.1.94
Starting mDNS server...
_http _tcp
BRW7429AF6E7F02 Brother HL-L2340D series _http _tcp 80
_udp
_upnp _tcp
_udp
```

My version:
```python
import wifi
import mdns

print(" Connecting", end="\r")
wifi.radio.connect("ssid", "password")
print(wifi.radio.ipv4_address)

print("Starting mDNS server...")
server = mdns.Server(wifi.radio)
service_types = ("_http", "_upnp" )

protocols = ("_tcp", "_udp")
for service_type in service_types:
    print(service_type, end=" ")
    for protocol in protocols:
        print(protocol)
        remote_service_tuple = server.find(service_type=service_type, protocol=protocol, timeout=5)
        for remote_service in remote_service_tuple:
            print(remote_service.hostname, remote_service.instance_name, remote_service.service_type, remote_service.protocol, remote_service.port)
```

anecdata, 2022-03-23:
That's the right code (well, I changed it to print the IP address when connected). If I run that code above as-is, it works, but, for example, _http + _udp crashes.

tannewt, 2022-03-23:
I wonder if it is an S2 vs S3 thing. I tested on S3s. I also tested with a later version off main.

tannewt, 2022-03-23:
I've replicated this on an S2.

anecdata, 2022-03-24:
S3 is quirky too, but I think PR fix will fix S3 also. Thanks!

I often put a `while True:` loop after test code (with a `pass` or a sleep or a print) because I don't want to immediately drop into the REPL, partially since sometimes memory effects can take a moment to manifest. On S3, the non-existent checks would not crash immediately, but would either reset the serial connection or safe mode after entering the `while` loop.

tannewt, 2022-03-24:
Makes sense. I'm surprised the S3 doesn't crash...

PR #6193 (2022-03-26): Fix MDNS crash on S2
Fixes #6186
commit 8642dc4a (2022-03-26): Fix MDNS crash on S2

Fixes #6186
closed 2022-03-26 by jepler