adafruit/circuitpython #6195: Creating an audio sample array corrupts flash memory

guighub, 2022-03-24:
### CircuitPython version

```python
adafruit-circuitpython-raspberry_pi_pico-en_US-7.2.3
```

### Code/REPL

```python
def PlayTone(inst, length, pitch, mixer, volume, effect):

    tone_frequency = int(440 * (pitch ** 1.059463094359))
    tone_length = int(length // tone_frequency)

    sine_wave = array.array("h", [0] * tone_length)
    for i in range(tone_length):
        sine_wave[i] = int((1 + math.sin(math.pi * 2 * i / tone_length)) * (volume / 7) * (2 ** 15 - 1))

    sine_wave_sample = RawSample(sine_wave, sample_rate=22050)

    while True:
        mixer.voice[0].play(sine_wave_sample, loop=True)
        time.sleep(length)
        mixer.stop_voice()
```

### Behavior

REPL disconnects and flash memory corrupts.

### Description

Every time I save my code trying to play an audio sample from an array I get corrupted sounds coming out of the DAC and am unable to save or copy anything from the board. Eventually it just disconnects completely.

### Additional information

Using flask_nuke.uf2 and reinstalling CircuitPython fixes the issue until caused again.

tannewt, 2022-03-24:
Does it happen if you don't play the sound but do everything else? I wonder if there is an interrupt routine that we're doing but should be in RAM so it doesn't cause flash reads.

rsbohn, 2022-03-25:
Can you post the code that sets up the mixer? I haven't been able to replicate the corruption on the Feather RP2040.

jepler, 2022-04-05:
@guighub can you please post complete code to reproduce the problem?

dhalbert, 2022-11-16:
@guighub We reviewed this issue today. Did you find the full code?

jepler, 2022-11-16:
We think this may be a duplicate of https://github.com/adafruit/circuitpython/issues/6005 which does have a full program.

closed 2022-11-16 by jepler

guighub, 2022-11-17:
@dhalbert No sorry, I don't think I have it anymore. If I do it's probably saved somewhere along with a bunch of other code, so I'd have to go through every one and test it and I don't know if I have all the hardware I did back then.

Although like jepler said, it might be a duplicate, and looking through the code on issue 6005 it looks familiar to what I was doing at the time.