adafruit/circuitpython #6205: MatrixPortal hangs and disconnects from USB after reading URL multiple times

jldeon, 2022-03-26:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit Matrix Portal M4 with samd51j19
Board ID:matrixportal_m4
```

### Code/REPL

```python
import time
import board
import traceback

from adafruit_matrixportal.matrixportal import MatrixPortal
from secrets import secrets

matrix_portal = MatrixPortal(status_neopixel=board.NEOPIXEL, debug=True)
network = matrix_portal.network
display = matrix_portal.graphics.display

url = "http://example.com"

while True:
    try:
        print(f"Reading url: {url}")
        matrix_portal.set_background(0x400000)
        resp = network.fetch(url)
        matrix_portal.set_background(0x000000)
        print(resp.content.decode('utf-8')[:64])
    except (ValueError, RuntimeError) as e:
        print("Likely data read error:")
        traceback.print_exception(None, e, e.__traceback__)
    except OSError as e:
        print("Likely wifi connection error:")
        traceback.print_exception(None, e, e.__traceback__)
    except Exception as e:
        print("Unknown error:")
        traceback.print_exception(None, e, e.__traceback__)
    time.sleep(1)
```

### Behavior

After some time passes (random, but typically less than an hour), the device stops responding.  

The last thing printed is "Retrieving data..." (which is a print that can't be turned off in the CircuitPython libraries)

After some time passes, USB disconnects.  The board has to be hard-reset to recover.

There is no Python backtrace or any other error message printed via the serial port.

### Additional information

I tested this initially with a text file hosted on an Apache server on the local LAN.  I then tested with example.com just to rule out anything weird on my Apache instance.  (So you ought to be able to repro this with various URLs, basically)

jldeon, 2022-03-26:
Slightly more minimal example:

```
import time
import board

from adafruit_matrixportal.matrixportal import MatrixPortal

matrix_portal = MatrixPortal(status_neopixel=board.NEOPIXEL, debug=True)
network = matrix_portal.network

url = "http://example.com"

while True:
    print(f"Fetch start")
    resp = network.fetch(url)
    print(f"Fetch end")
    time.sleep(1)

```

ajs256, 2022-03-27:
Can confirm this.

mattikro, 2022-03-31:
Same issue, this code will not run properly as it needs access to my Spotify api.
[MAR30.zip](https://github.com/adafruit/circuitpython/files/8385026/MAR30.zip)

mpicker90, 2022-04-03:
same issue. I tried resetting the Network connection every couple minutes as well, but the issue still persists. Running on the latest 7.2.4 uf2

jepler, 2022-04-04:
It's not a fix, but an option you have is to try the hardware watchdog, which would reset the whole microcontroller (the one where CircuitPython resides) if your program stops running normally.

Before doing this, be comfortable with entering safe mode on your board, because an errant watchdog reset can make it difficult to modify your code.py.

You'd initialize the watchdog with a timeout:
```python
from microcontroller import watchdog as w
from watchdog import WatchDogMode
w.timeout=2.5 # timeout in seconds
w.mode = WatchDogMode.RESET
w.feed()
```
and within your top-level loop you'd keep the watchdog happy with `w.feed()`. If it's ever NOT called for 2.5 seconds, the whole microcontroller is reset, similar to pressing the reset button.

Wintaru, 2022-04-04:
I battled this as well and worked with staff to figure out the issue. We tried replacing hardware too which helped failure rates but it still fails. Trying the watchdog thing now and it seems to be stable so far, I just started a rather long print so we'll see if it's still running in the morning.

dhalbert, 2022-04-04:
For those of you who had this problem, what version of the NINA-FW firmware are you using on the ESP32 that's on the MtarixPortal? That is printed out by the simple Internet test given here: https://learn.adafruit.com/adafruit-matrixportal-m4/internet-connect. The key value is `esp.firmware_version`.

Wintaru, 2022-04-04:
Firmware vers. bytearray(b'1.2.2\x00')

mpicker90, 2022-04-04:
Firmware vers. bytearray(b'1.2.2\x00')

Wintaru, 2022-04-04:
> I tried using the watchdog time out as suggested above. Just had it crash and reset automatically. Thanks for the suggestion @jepler

I had to increase the timeout to 15, and make sure you have the feed call both there and in your loop otherwise it's not fed soon enough to enter the loop.

anecdata, 2022-04-04:
It's worth testing whether NINA v1.7.4 improves things, though I couldn't definitively tie any NINA fixes to this problem.

It seems odd to me that this is only reported on MatrixPortal and not other M4+Airlift configs like PyPortal.

jepler, 2022-04-04:
On this board the 'effective' watchdog timeout values are slightly above 1, 2, 4, 8, and 16 seconds; values are rounded up.

dhalbert, 2022-04-06:
OK, I have managed to get a crash, running NINA-FW 7.1.1, after 1766 seconds doing simple fetches from a local webserver, using the example above: https://github.com/adafruit/circuitpython/issues/6205#issuecomment-1079777657.
```
763 Fetch start Retrieving data...Fetch end
1764 Fetch start Retrieving data...Fetch end
1765 Fetch start Retrieving data...Fetch end
1766 Fetch start Retrieving data...Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "c6205.py", line 14, in <module>
  File "adafruit_portalbase/network.py", line 480, in fetch
  File "adafruit_requests.py", line 864, in get
  File "adafruit_requests.py", line 734, in request
OutOfRetries: Repeated socket failures
```
There were [upstream fixes in NINA-FW 1.7.2](https://github.com/adafruit/nina-fw/releases/tag/1.7.2), so I will upgrade to 1.7.4 and see if I can reproduce.

jldeon, 2022-04-07:
> OK, I have managed to get an exception on MatrixPortal, using CIrcuitPython 7.2.5 and running NINA-FW 7.1.1, after 1766 seconds doing simple fetches from a local webserver, using the example above: [#6205 (comment)](https://github.com/adafruit/circuitpython/issues/6205#issuecomment-1079777657).
> 
> ```
> 763 Fetch start Retrieving data...Fetch end
> 1764 Fetch start Retrieving data...Fetch end
> 1765 Fetch start Retrieving data...Fetch end
> 1766 Fetch start Retrieving data...Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "c6205.py", line 14, in <module>
>   File "adafruit_portalbase/network.py", line 480, in fetch
>   File "adafruit_requests.py", line 864, in get
>   File "adafruit_requests.py", line 734, in request
> OutOfRetries: Repeated socket failures
> ```
> 
> There were [upstream fixes in NINA-FW 1.7.2](https://github.com/adafruit/nina-fw/releases/tag/1.7.2), so I will upgrade to 1.7.4 and see if I can reproduce.

I don't think this is the same issue.  I've seen various exceptions from the lower-level libraries, including this one.  

The issue I'm reporting is that it hangs - there's no exception raised.

dhalbert, 2022-04-07:
>The issue I'm reporting is that it hangs - there's no exception raised.

I upgraded to 1.74, and I am also now seeing hangs, sometimes after a couple of minutes sometimes much longer. No panel is attached to the MatrixPortal, so that's not a factor. So it's disappointing the upgrade is not helping, but I have some leads.

Wintaru, 2022-04-07:
> > OK, I have managed to get an exception on MatrixPortal, using CIrcuitPython 7.2.5 and running NINA-FW 1.7.1, after 1766 seconds doing simple fetches from a local webserver, using the example above: [#6205 (comment)](https://github.com/adafruit/circuitpython/issues/6205#issuecomment-1079777657).
> > ```
> > 763 Fetch start Retrieving data...Fetch end
> > 1764 Fetch start Retrieving data...Fetch end
> > 1765 Fetch start Retrieving data...Fetch end
> > 1766 Fetch start Retrieving data...Traceback (most recent call last):
> >   File "<stdin>", line 1, in <module>
> >   File "c6205.py", line 14, in <module>
> >   File "adafruit_portalbase/network.py", line 480, in fetch
> >   File "adafruit_requests.py", line 864, in get
> >   File "adafruit_requests.py", line 734, in request
> > OutOfRetries: Repeated socket failures
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > There were [upstream fixes in NINA-FW 1.7.2](https://github.com/adafruit/nina-fw/releases/tag/1.7.2), so I will upgrade to 1.7.4 and see if I can reproduce.
> 
> I don't think this is the same issue. I've seen various exceptions from the lower-level libraries, including this one.
> 
> The issue I'm reporting is that it hangs - there's no exception raised.

 Same for me, it hangs and, importantly, the USB device is unmounted.

anecdata, 2022-04-08:
I haven't cleanly reproduced this yet, but using CP 7.2.5 and NINA 1.6.1, I notice that occasionally it takes over a minute for "Retrieving data..." to complete (usually more like 1.5 seconds), and if I try to re-save `code.py` during that time, the device will disconnect from serial and eject CIRCUITPY. Not sure if that's a related issue or separate.

dhalbert, 2022-04-09:
I have been testing a simpler web fetcher that does not use the MaxtrixPortal/PortalBase library. It's been running for hours without crashing. It is like https://github.com/adafruit/circuitpython/issues/6205#issuecomment-1079777657 above, but without the library. For both I have been fetching from a simple local webserver: `python -m http.server`. The lack of a crash doesn't mean there's not a core bug, but it means something about the library might be provoking it.
```py
import board
import busio
from digitalio import DigitalInOut
import adafruit_requests as requests
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from adafruit_esp32spi import adafruit_esp32spi
import time

try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

requests.set_socket(socket, esp)

if esp.status == adafruit_esp32spi.WL_IDLE_STATUS:
    print("ESP32 found and in idle mode")
print("Firmware vers.", esp.firmware_version)
print("MAC addr:", [hex(i) for i in esp.MAC_address])

print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(secrets["ssid"], secrets["password"])
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)
print("My IP address is", esp.pretty_ip(esp.ip_address))

URL = "http://192.168.1.222:8000"

secs = 0
while True:
    print(secs, "Fetch start ", end = "")
    requests.get(URL)
    print("Fetch end")
    time.sleep(1)
    secs += 1
```

anecdata, 2022-04-09:
fwiw, ESP32SPI has been very stable for me though not using any of the higher-level networking or portal-type libraries, only esp32spi and requests, no crashes in a long, long time.

But... I restarted the Matrix Portal and let the code above (minor tweaks to print more info) run again until it hung (but again it didn't eject CIRCUITPY or disconnect from serial). After a few hours I came back and it hadn't changed, but macOS Finder, once selected, was spinning beach ball then eventually timed out, _then_ CIRCUITPY ejected and serial disconnected.

jepler, 2022-04-09:
I assume MatrixPortal initializes the rgbmatrix display. One suspicion I have is that there's still a low-probability "crash during soft reset" bug in that code.

Maybe a program which
 * sets up the display
 * sets auto-restart on exception
 * triggers an exception

would give interesting results. It could help confirm/disconfirm my suspicion.

dsohrabian, 2022-05-06:
Has anyone found any clues for resolving this/ work arounds for now? or news of firmware update in development that may fix it?

dhalbert, 2022-05-06:
As noted above, when not using the Matrix Portal library, I don't see these crashes. We're not sure why the library is causing a problem, but you could try rewriting your program to use `adafruit_requests` directly and see whether you see the problem.

dsohrabian, 2022-05-06:
I'm not using Matrix Portal library; direct adafruit_requests post methods with similar network setup via Internet connect guide, leading to the same symptoms mentioned here (hang, USB disconnect). My flash just wiped itself clean as I press reset button. So unless there's a manual wipe button sequence I just pressed, it seems like other things are wrong with my board. Not sure what to do at this point. I'll reset firmware and try to repeat symptoms I guess.

I might as well ask.: Should a Matrix Portal M4 be able to run 3 small post requests every 30 seconds with 5kb responses? Does this raise any red flags for hardware limits of MP M4? Should that be something it can manage? I suppose I should have studied that before pursuing this project. I've tried several ways to do basic requests, from MatrixPortal library to wifi manager to direct requests. I can't get this to consistently do anything with the internet longer than 3-4 hours max without hanging. And I'm left wondering if this is simply beyond the capability of the device?

dhalbert, 2022-05-06:
@dsohrabian We would be quite interested in the simplest example program you have that seems to cause trouble, including your descriptions of what is returned from the requests. 5kB reponses might eventually lead to memory fragmentation, depending on how you hang on to the results. But there also might be bugs in the ESP32 firmware.

dsohrabian, 2022-05-18:
> @dsohrabian We would be quite interested in the simplest example program you have that seems to cause trouble, including your descriptions of what is returned from the requests. 5kB reponses might [...]

After testing, I believe that the hanging issue happens when I import adafruit_matrixportal Matrix object. I have commented out the whole board object in my main loop, and the POST requests come in smoothly with no hanging all day.

Here is the module below (train_board.py) that is used to run my LED board that causes the hanging/freezing. The get_new_data function sends POST requests and returns a list without any bugs that's used to update the Train objects.  This module creates the display and updates the pieces of the LED board to show transit updates. Something in here is the culprit.

```import displayio
import time
from adafruit_display_shapes.rect import Rect
from adafruit_display_text.label import Label
from adafruit_matrixportal.matrix import Matrix

from config import config

class TrainBoard:
	def __init__(self, get_new_data):
		self.get_new_data = get_new_data

		self.display = Matrix().display

		# create parent group for entire display relative positioning
		self.parent_group = displayio.Group()

		# setup header row showing LINE DESTINATION MINUTES
		self.heading_label = Label(x=0, y=3, font=config['font'], anchor_point=(0,0))
		self.heading_label.color = config['heading_color']
		self.heading_label.text=config['heading_text']
		self.parent_group.append(self.heading_label)

		# add small clock at bottom
		self.time_update = Label(anchored_position=(32, 27),
		                         font=config['time_font'],
		                         anchor_point=(0.5, 0)
		                         )

		self.time_update.text = 'RIDE GCRTA'
		self.parent_group.append(self.time_update)

		# instantiate configured number of rows (trains) on board and add to parent group
		self.trains = []

		for i in range(config['num_routes']):
			self.trains.append(Train(self.parent_group, index= i))

		# display parent displayio group on board
		self.display.show(self.parent_group)

	def refresh(self) -> bool:
		# prep time stamp for clock update
		cur_hour = time.localtime().tm_hour
		cur_min = time.localtime().tm_min
		tstamp = f'{cur_hour}:{cur_min:0>2}'
		print(f'{tstamp} Refreshing train information...')

		# call function fetch_predictions from main code.py, get parsed list
		train_data = self.get_new_data()

		if train_data is not None:
			for i in range(config['num_routes']):
				if i < len(train_data):
					train = train_data[i]
					if not train['arrival']:
						self._hide_train(i)
					else:
						self._update_train(i, train['line_color'], train['destination'], train['arrival'])
				else:
					self._hide_train(i)
			self.time_update.text = tstamp # update clock
		else:
			print('No data received. Clearing display.')
			for i in range(config['num_routes']):
				self._hide_train(i)

	def _hide_train(self, index: int):
		self.trains[index].hide()

	def _update_train(self, index: int, line_color: int, destination: str, minutes: str):
		self.trains[index].update(line_color, destination, minutes)

	def blink(self):
		pass

class Train:
	def __init__(self, parent_group, index):

        # position for aligning row correctly
		y = (index+1) * 7

		self.line_rect = Rect(x=0, y=0, width=config['train_line_width'], height=config['train_line_height'], fill=config['loading_line_color'])

		self.destination_label = Label(x=1 + config['text_padding'], y=3, font=config['font'], )
		self.destination_label.color = config['text_color']
		self.destination_label.text = config['loading_destination_text'][:config['destination_max_characters']]

		self.min_label = Label(x= config['matrix_width']- (config['min_label_characters']*config['character_width']),
        y=3, font=config['font'])
		self.min_label.color = config['text_color']
		self.min_label.text = (' ' * (config['min_label_characters'] -len(config['loading_min_text'])) +
		 config['loading_min_text'])

		self.group = displayio.Group(x=0, y=y)
		self.group.append(self.line_rect)
		self.group.append(self.destination_label)
		self.group.append(self.min_label)

		parent_group.append(self.group)

	# show and hide a row on board
	def show(self):
		self.group.hidden = False

	def hide(self):
		self.group.hidden = True

	# lower level update functions
	def set_line_color(self, line_color: int):
		self.line_rect.fill = line_color

	def set_destination(self, destination: str):
		self.destination_label.text = destination[:config['destination_max_characters']]

	def set_arrival_time(self, minutes: str):
		# Ensuring we have a string
		minutes = str(minutes)
		minutes_len = len(minutes)

		# Left-padding the minutes label
		minutes = ' ' * (config['min_label_characters'] - minutes_len) + minutes

		self.min_label.text = minutes

	# high level update using above
	def update(self, line_color: int, destination: str, minutes: str):
		self.show()
		self.set_line_color(line_color)
		self.set_destination(destination)
		self.set_arrival_time(minutes)

dsohrabian, 2022-05-18:
@dhalbert As @anecdata @jepler mention, I think the hang/freeze is coming from MatrixPortal library.

This import from my code above seems to be the only common thread from prior discussion here and similar symptom of hanging. 
`from adafruit_matrixportal.matrix import Matrix`

 I don't know if the repeating post/get requests that are happening in main loop on code.py are coincidental or not.

dsohrabian, 2022-05-21:
@dhalbert can you offer any tips on how to debug the Matrix object in MatrixPortal library? Do I keep using lower and lower level objects until the freeze doesn't occur?

dhalbert, 2022-05-22:
I am looking at some issues with chunked responses, but if you have an example, that would be great. If it's small enough, paste it inline in a comment. Otherwise you can upload it as a .txt file or a .zip file. Thanks!

dsohrabian, 2022-05-22:
OK thank you for taking a look.
[transit_board_20220521.zip](https://github.com/adafruit/circuitpython/files/8748739/transit_board_20220521.zip)
 
This is a .zip of my project. It has a secrets.py that needs wifi and aio values. 

I successfully ran this on my MatrixPortal M4 for 24 hours when I don't use my display module and only call requests. So I suspect the requests are not the primary cause, and it could be a joint or separate issue with the display libraries.

In a nutshell my display manager module (train_board) manages displayio Groups, Rect and Labels on a matrixportal.Matrix based on the results of the api functions. Hope this is useful enough to debug the hang/disconnect within an hour.

dsohrabian, 2022-05-28:
[code.txt](https://github.com/adafruit/circuitpython/files/8791793/code.txt)

@dhalbert Here is a simplified single file version that should be easier to parse through. At its core, it's doing the same thing without object-oriented methods.  I got a USB disconnect and hang at about 90 minutes on this one.

dhalbert, 2022-05-28:
I have proposed one fix to the ESP32SPI library to solve a different problem. I'm not sure it has anything to do with this, but worth trying: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/166

dhalbert, 2022-05-29:
@dsohrabian if you add ` gc.collect()` at the end before or after the `time.sleep(30)`, does it make any difference? I'm wondering if it's fragmentation or not.

dsohrabian, 2022-05-29:
@dhalbert No luck, board freezes (LED keeps lit with last update) and USB disconnect.
I am printing `gc.mem_free()` every round and it is staying stable.

Can you reproduce the error with your own Matrix Portal M4 and LED 64x32?  It would be good to know that this isn't a hardware or quality control issue with one of the pieces. This seems like it would be a major and prevalent bug for a product that is advertised to specifically fetch internet data and put it on an LED board. But I don't see any others having the problem so I have to wonder.

dhalbert, 2022-05-29:
`gc.mem_free()` being stable means that gc is working. But the gc is not compacting, so there can be enough free memory, but it could be fragmented, and not have enough contiguous space for a requested large buffer. If one does `gc.collect()` before the fragmentation has a chance to happen, then it cn be avoided.

I think it may have to do with what is being fetched. If the HTTP response is of constant length, or nearly so, then the same memory block can get used over and over. If it grows and shrinks, or straddles a boundary in allocated-block size, then it's easier for fragmentation to develop. E.g., if allocated block sizes are, say, 4kB and 6kB, and sometimes the size needed is < 4k, and sometimes greater, than fragmentation could develop if one doesn't gc until forced to, when heap has run out. (I am not saying that's the actual block size -- it's just an illustrative hypothetical).

> Can you reproduce the error with your own Matrix Portal M4 and LED 64x32?

Since it may depend on the site and what it returns, I can try exactly your transit datasite, or we could get sample data from it, and then make up our own test server and we can return the same data, but fetch much more quickly. I don't want to hammer the transit site: it may rate-limit or cut a fetcher off completely if it looks like abuse. That is why I suggested you try adding the `gc.collect()` since you are running this anyway.

dsohrabian, 2022-05-29:
Thank you, that is interesting. I have a lot to learn about memory use. 
I have `gc.collect()` sprinkled throughout now. And will see how this goes.

As far the transit site: it definitely tolerates requests every 30 seconds. The site is even hard-coded to auto-refresh on that interval if you leave it open on browser. It is meant for public wear and tear and probably doesn't get much traffic.  I push it and fetch 4 times on 30-second interval (which is the ultimate need of this project, but 1 per 30 seconds freezes the MP4 too). So far, the transit server doesn't seem to mind. Have only gotten 200 status codes on responses so far.

This version of my `code.py` only fetches one route per 30 seconds.
[code.txt](https://github.com/adafruit/circuitpython/files/8794144/code.txt)

dsohrabian, 2022-05-30:
@dhalbert One thing I've reproduced a couple times: when I comment out all the `Matrix` objects and don't engage with the LED matrix in any way, the board hums along and requests for 24 hours+ with no issues. The second I bring the LED display (via adafruit_matrixportal Matrix) into the picture, alongside fetching data online, the system will inevitably hang.

dhalbert, 2022-05-30:
> The second I bring the LED display (via adafruit_matrixportal Matrix) into the picture, alongside fetching data online, the system will inevitably hang within couple hours.

Thanks, that is _very_ helpful.

dhalbert, 2022-06-04:
@dsohrabian I tried your 30-second-interval code.txt above. I stopped it after round 740, so it ran for hours without difficulty.

The fix I mentioned above for `adafruit_esp32spi` is now in its version 4.2.2 or later, which you can get from the bundle, circup or the repo. I have been using that fix. It generates less garbage as it's running and fixes a problem that showed up with certain servers that produced chunked output. Could you try that version for a few hours?

dsohrabian, 2022-06-04:
@dhalbert Thank you so much for testing out the code. I will give your esp32spi library version a shot.

By the way, the code.py I shared only runs 1 request per 30 seconds. The project needs 4 requests per 30 seconds, but that crashes even sooner when feeding the data to the LED display. And I was trying to keep it as simple as possible for reproducing so cut the other requests. I was hoping this wouldn't need to be manually reset every 1-6 hours, and was shooting for more like every couple weeks or months at worst.

Without engaging with the `Matrix` class and the display, the MP can run the 4 post requests per 30 seconds indefinitely, so I know the ESP is capable and implies to me that the requests aren't the issue. That's where I am lost.

Were you able to debug the crash on your end at all or did it disconnect without any clues?

dhalbert, 2022-06-04:
> Were you able to debug the crash on your end at all or did it disconnect without any clues?

It did not crash or disconnect at all: I just stopped it because it had run for about 6 hours, far longer than the 90 minutes you mentioned. So definitely try the new library version. I was also using CircuitPython 7.3.0 Update all the libraries: the Portal libraries have also been revised, though not as recently.

I am running the latest (1.7.4) NINA-FW firmware as well. To determine the version, you can use the test program here, which will print it out: https://learn.adafruit.com/adafruit-matrixportal-m4/internet-connect#connect-to-wifi-3034771-4

dhalbert, 2022-06-04:
I changed your code to have it do four fetches at a time (removed `[:1]`), and  it stopped after 47 cycles. So I have a good place to start looking. It did the fetches and then something happened before it could print "Finished round". I am printing out the fetches to collect the data so I could simulate it locally at higher speed.

dsohrabian, 2022-06-04:
> I changed your code to have it do four fetches at a time (removed `[:1]`), and it stopped after 47 cycles. So I have a good place to start looking. It did the fetches and then something happened [...]

That sounds like a great lead! Please let me know if I can help in any way.

I will be running tests to confirm again that the freeze doesn't happen when all display setup and text update line are commented out.

I am running 1.7.4 for the ESP too.

dsohrabian, 2022-06-05:
@dhalbert A correction for you: performing the import via `from adafruit_matrixportal.matrix import Matrix` is enough to cause the freeze. Creating a Matrix object is not in fact necessary.

mpicker90, 2022-06-05:
Is there a memory leak in 7.3.0 I just tried updating from a 7.2.x version and i had roughly 30,000 print out when calling gc.mem_free(),  and only printing 7000 when calling gc.mem_free() and i keep getting this in the logs "memory allocation failed, allocating 232 bytes"

dhalbert, 2022-06-05:
@mpicker90 Please open a new issue or a thread in the forums with the details, including your code. Make sure your libraries are all the latest version.

dsohrabian, 2022-06-06:
@dhalbert

Now confirmed with this code.py that the 4 requests / 30 seconds don't cause an issue with no `adafruit_matrixportal.matrix` library imports and no LED display. Ran it for 48 hours and no hiccups.
[code.txt](https://github.com/adafruit/circuitpython/files/8841718/code.txt)

dhalbert, 2022-06-06:
Well, now I had the problem and your did not :) . I'll do some more testing!

dhalbert, 2022-06-06:
Oops, misinterpreted your latest post. So some interaction is causing problems, not sure what. I will see what is happening when the example with the MatriPortal library hangs.

dhalbert, 2022-06-15:
Closing in favor of #6489, which is meant to describe the root cause. Thanks to everyone who tested and contributed to this issue, and feel free to continue discussions in #6489.

dsohrabian, 2022-06-15:
> Closing in favor of #6489, which is meant to describe the root cause. Thanks to everyone who tested and contributed to this issue, and feel free to continue discussions in #6489.

Dan, I really appreciate the efforts to figure this out. Will be eagerly following the issue. While this is helping me learn some lower-level concepts, this is above my understanding. But I would be glad to help how I can.

dhalbert, 2022-06-15:
@dsohrabian -- so you are not blocked, here is a build to try that might run for longer, and not hang up. It does not use DMA at all for SPI: 
[matrixportal-no-dma-spi.uf2.zip](https://github.com/adafruit/circuitpython/files/8905313/matrixportal-no-dma-spi.uf2.zip). It says 8.0.0-alpha1-something, but you should be able to use it with the 7.x libraries. I think it may seem slower, but I'm not sure if you'll notice or not.

It might still sporadically have the other problem I mentioned, which is a communications issue with the ESP32 Airlift. You might see `Error response to command`. That I don't understand yet.

dhalbert, 2022-06-15:
Closing, but can continue ancillary discussion here if not directly relevant to #6489.

closed 2022-06-15 by dhalbert

dsohrabian, 2022-06-15:
Thank you, I will try the no DMA option and report back if it made a difference.

> It might still sporadically have the other problem I mentioned, which is a communications issue with the ESP32 Airlift. You might see `Error response to command`. That I don't understand yet.
 I've noticed this bug too and early on it mislead me as the possible root cause. Glad you are aware of it too. It didn't happen often. And luckily it prompts an error exception which seems it can be handled with `ESP_SPIcontrol.reset()` or similar

dsohrabian, 2022-06-21:
@dhalbert  The project now works with this non-DMA version 🙏 Thank you so much for making the bootloader for me. And the `Error response to command` issue is managed by resetting ESP and reconnecting to wifi on exceptions. 

I will hang on to this no-DMA version. It looks like you fixed it for the next release, so I'll plan on upgrading.

dhalbert, 2022-06-21:
I was just going to write to you. I fixed it in a different way, which I think will work too. See https://github.com/adafruit/circuitpython/pull/6498 and https://github.com/adafruit/samd-peripherals/pull/42. Try the build artifacts: https://github.com/adafruit/circuitpython/actions/runs/2516315728. Scroll down to see the artifacts. Unzip the file for your board and get the .uf2 you need.

gadjodilo83, 2022-06-28:
> I was just going to write to you. I fixed it in a different way, which I think will work too. It is not the same fix as what I gave you. See #6498 and [adafruit/samd-peripherals#42](https://github.com/adafruit/samd-peripherals/pull/42). Try the build artifacts: https://github.com/adafruit/circuitpython/actions/runs/2516315728. Scroll down to see the artifacts. Unzip the file for your board and get the .uf2 you need.

I used the latest version of the uf2 file and the newest library and still got the following error message:

Getting time for timezone Europe/Zurich
Zurückverfolgung (jüngste Aufforderung zuletzt):
  Datei "code.py", Zeile 139, in <module>
  Datei "adafruit_portalbase/network.py", Zeile 247, in get_local_time
  Datei "adafruit_portalbase/network.py", Zeile 216, in get_strftime
  Datei "adafruit_requests.py", Zeile 818, in get
  Datei "adafruit_requests.py", Zeile 674, in request
  Datei "adafruit_esp32spi/adafruit_esp32spi_socket.py", Zeile 138, in recv
  Datei "adafruit_esp32spi/adafruit_esp32spi_socket.py", Zeile 217, in available
  Datei "adafruit_esp32spi/adafruit_esp32spi.py", Zeile 776, in socket_available
  Datei "adafruit_esp32spi/adafruit_esp32spi.py", Zeile 332, in _send_command_get_response
  Datei "adafruit_esp32spi/adafruit_esp32spi.py", Zeile 299, in _wait_response_cmd
  Datei "adafruit_esp32spi/adafruit_esp32spi.py", Zeile 274, in _wait_spi_char
RuntimeError: Error response to command

thanks
marc

dsohrabian, 2022-06-28:
Hi Marc,

I get that too still. It's a separate issue that I think needs to be
explored. My workaround is to write a try except block for that error and
use esp.reset() and esp.connect to reconnect to internet in the except
block. That has been working for me well.

dhalbert, 2022-06-28:
Yes, the `RuntimeError: Error response to command` is caused by something else, and we still have to track that down. At least it is recoverable via a try-except.