adafruit/circuitpython #6206: ESP-BOX (ESP32-S3) crash to HardFault_Handler on wifi.radio.connect()

fivesixzero, 2022-03-27:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-64-gdd7318244-dirty on 2022-03-27; ESP32-S3-Box-2.5 with ESP32S3
Board ID:espressif_esp32s3_box
```

### Code/REPL

```python
import wifi
from secrets import secrets

for network in wifi.radio.start_scanning_networks():
    print("network [{}], ssid [{}], channel [{}]".format(network, network.ssid, network.channel))
wifi.radio.stop_scanning_networks()

print("WiFi Init: ssid [{}]".format(secrets["ssid"]))
wifi.radio.connect(ssid=secrets["ssid"], password=secrets["password"])
print("WiFi Connected: bssid [{}], ip [{}]".format(wifi.radio.ap_info.bssid, wifi.radio.ipv4_address))
```

### Behavior

Network scan functions as expected but it looks like the hard fault is happening within `wifi.radio.connect()`.

```python
WiFi Init: ssid [jita]
```

Hard reset into safe mode

```python
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Fresh `7.3.0-alpha0` build using `main` branch.

Tried doing a full `esptool.py erase_flash` followed by a fresh `make flash`, same results.

The network in question works fine with a bunch of ESP32-S2 and other boards running CircuitPython.

fivesixzero, 2022-03-27:
I tested the same code with a different `ESP32-S3` board, a `ESP32-S3-DevKitC-1 N8R2`.

On this board `wifi.radio.connect()` connects to the network just fine without a hard fault/reset.

Board:

```
Adafruit CircuitPython 7.3.0-alpha.0-64-gdd7318244-dirty on 2022-03-27; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
Board ID:espressif_esp32s3_devkitc_1_n8r2
```

anecdata, 2022-03-27:
Is the debug console accessible on that device? If you do a `DEBUG=1` build, you can then use https://github.com/adafruit/circuitpython/blob/main/ports/espressif/tools/decode_backtrace.py to narrow down where it fails.

fivesixzero, 2022-03-27:
> Is the debug console accessible on that device? If you do a `DEBUG=1` build, you can then use https://github.com/adafruit/circuitpython/blob/main/ports/espressif/tools/decode_backtrace.py to narrow down where it fails.

That's a good question. I haven't done any debugging on the ESP32 series before so I'm not sure what to look for. I do see that the `PMOD` headers include a pair of UART pins though, which map to `GPIO44` (rx)/`GPIO43` (tx) according to the [schematic](https://github.com/espressif/esp-box/tree/master/hardware/esp32_s3_box_v2.5/schematic).

It does look like those are set up for `DEBUG` in the `board.c`, so I'll give that a shot and see if there's a console as soon as I can dig up a USB->UART board.

fivesixzero, 2022-03-27:
Alright, found a USB UART and managed to get the debug console output.

```
ESP-ROM:esp32s3-20210327                                           
Build:Mar 27 2021                                                               
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)                                
SPIWP:0xee                                                                      
mode:DIO, clock div:2                                                           
load:0x3fcd0108,len:0x1654                                                      
load:0x403b6000,len:0xb7c                                                       
load:0x403ba000,len:0x2f78                                                      
entry 0x403b6248                                                                
I (29) boot: ESP-IDF 121ddb87e-dirty 2nd stage bootloader                       
I (30) boot: compile time 13:24:27                                              
I (30) boot: chip revision: 0
I (32) boot.esp32s3: Boot SPI Speed : 40MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (85) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (92) boot:  4 uf2              factory app      00 00 00410000 00040000
I (100) boot:  5 user_fs          Unknown data     01 81 00450000 00bb0000
I (107) boot: End of partition table
I (112) boot: Defaulting to factory image
E (116) esp_image: image at 0x410000 has invalid magic byte (nothing flashed here?)
E (125) boot: Factory app partition is not bootable
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=49c68h (302184) map
I (214) esp_image: segment 1: paddr=00059c90 vaddr=3fc99d10 size=055d0h ( 21968) load
I (220) esp_image: segment 2: paddr=0005f268 vaddr=40374000 size=00db0h (  3504) load
I (222) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=113274h (1127028) map
I (508) esp_image: segment 4: paddr=0017329c vaddr=40374db0 size=14f5ch ( 85852) load
I (538) esp_image: segment 5: paddr=00188200 vaddr=50000000 size=01010h (  4112) load
I (540) esp_image: segment 6: paddr=00189218 vaddr=600fe000 size=00028h (    40) load
I (554) boot: Loaded app from partition at offset 0x10000
I (554) boot: Disabling RNG early entropy source...
I (567) opi psram: vendor id : 0x0d (AP)
I (567) opi psram: dev id    : 0x02 (generation 3)
I (568) opi psram: density   : 0x03 (64 Mbit)
I (571) opi psram: good-die  : 0x01 (Pass)
I (576) opi psram: Latency   : 0x01 (Fixed)
I (581) opi psram: VCC       : 0x01 (3V)
I (585) opi psram: SRF       : 0x01 (Fast Refresh)
I (591) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (596) opi psram: BurstLen  : 0x01 (32 Byte)
I (601) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (607) opi psram: DriveStrength: 0x00 (1/1)
W (612) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future .
I (623) spiram: Found 64MBit SPI RAM device
I (627) spiram: SPI RAM mode: sram 80m
I (631) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (639) cpu_start: Pro cpu up.
I (642) cpu_start: Starting app cpu, entry point is 0x40378438
I (0) cpu_start: App cpu up.
I (1075) spiram: SPI SRAM memory test OK
I (1085) cpu_start: Pro cpu start user code
I (1085) cpu_start: cpu freq: 240000000
I (1085) cpu_start: Application information:
I (1088) cpu_start: Project name:     circuitpython
I (1093) cpu_start: App version:      7.3.0-alpha.0-247-gdd7318244-di
I (1100) cpu_start: Compile time:     Mar 27 2022 13:24:18
I (1107) cpu_start: ELF file SHA256:  18c89cf36f23cacc...
I (1113) cpu_start: ESP-IDF:          121ddb87e-dirty
I (1119) heap_init: Initializing. RAM available for dynamic allocation:
I (1126) heap_init: At 3FCAA988 len 00035678 (213 KiB): D/IRAM
I (1132) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
[... 98 lines cut ...]
W (8329) wifi: scan
W (8499) wifi: scan
W (8629) wifi: scan
W (8829) wifi: scan
W (8949) wifi: scan
W (9069) wifi: scan
W (9199) wifi: scan
W (9369) wifi: scan
I (11419) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40386813  PS      : 0x00060330  A0      : 0x80387071  A1      : 0x3fcae980  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x00000040  A5      : 0xff000fff  
A6      : 0x600c0024  A7      : 0x00000000  A8      : 0x00000009  A9      : 0x00000000  
A10     : 0x00000010  A11     : 0x00000024  A12     : 0x00000005  A13     : 0x3fca8b28  
A14     : 0x3fc9d52c  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace:0x40386810:0x3fcae9800x4038706e:0x3fcae9a0 0x403799c5:0x3fcae9c0 0x403799d9:0x3fcD

ELF file SHA256: 18c89cf36f23cacc

CPU halted.
```

fivesixzero, 2022-03-27:
I just double-checked with `DEBUG=1` builds on my other two S3 boards - a `DevKitM-1` with a `mini` module and a `DevKitC-1-N8R2` with a very similar `WROOM-1` module - and both connect to the network without any odd behavior in the debug output.

The only obvious differences I can see between the `esp-box` and the `DevKitC` hardware are size differences in flash and PSRAM (8 vs 16 MB and 2 vs 8 MB respectively).

On a possibly related note, the debug output for the `esp-box` seems to indiciate a different init process for its PSRAM, complete with a warning not to use it for mass production.

`esp-box` PSRAM init:

```
E (125) boot: Factory app partition is not bootable
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=49c68h (302184) map
I (214) esp_image: segment 1: paddr=00059c90 vaddr=3fc99d10 size=055d0h ( 21968) load
I (220) esp_image: segment 2: paddr=0005f268 vaddr=40374000 size=00db0h (  3504) load
I (222) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=113274h (1127028) map
I (508) esp_image: segment 4: paddr=0017329c vaddr=40374db0 size=14f5ch ( 85852) load
I (538) esp_image: segment 5: paddr=00188200 vaddr=50000000 size=01010h (  4112) load
I (540) esp_image: segment 6: paddr=00189218 vaddr=600fe000 size=00028h (    40) load
I (554) boot: Loaded app from partition at offset 0x10000
I (554) boot: Disabling RNG early entropy source...
I (567) opi psram: vendor id : 0x0d (AP)
I (567) opi psram: dev id    : 0x02 (generation 3)
I (568) opi psram: density   : 0x03 (64 Mbit)
I (571) opi psram: good-die  : 0x01 (Pass)
I (576) opi psram: Latency   : 0x01 (Fixed)
I (581) opi psram: VCC       : 0x01 (3V)
I (585) opi psram: SRF       : 0x01 (Fast Refresh)
I (591) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (596) opi psram: BurstLen  : 0x01 (32 Byte)
I (601) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (607) opi psram: DriveStrength: 0x00 (1/1)
W (612) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future .
I (623) spiram: Found 64MBit SPI RAM device
I (627) spiram: SPI RAM mode: sram 80m
I (631) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (639) cpu_start: Pro cpu up.
I (642) cpu_start: Starting app cpu, entry point is 0x40378438
```

`DevKitC-1-N8R2` PSRAM init:

```
E (125) boot: Factory app partition is not bootable
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=49bd0h (302032) map
I (214) esp_image: segment 1: paddr=00059bf8 vaddr=3fc994b0 size=05328h ( 21288) load
I (225) esp_image: segment 2: paddr=0005ef28 vaddr=40374000 size=010f0h (  4336) load
I (227) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=112fa8h (1126312) map
I (511) esp_image: segment 4: paddr=00172fd0 vaddr=403750f0 size=143b8h ( 82872) load
I (540) esp_image: segment 5: paddr=00187390 vaddr=50000000 size=01010h (  4112) load
I (541) esp_image: segment 6: paddr=001883a8 vaddr=600fe000 size=00028h (    40) load
I (555) boot: Loaded app from partition at offset 0x10000
I (555) boot: Disabling RNG early entropy source...
I (569) spiram: Found 16MBit SPI RAM device
I (569) spiram: SPI RAM mode: sram 80m
I (569) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (574) cpu_start: Pro cpu up.
I (577) cpu_start: Starting app cpu, entry point is 0x40378358
```

Not sure if this is related but it seemed interesting enough to share.

anecdata, 2022-03-27:
I wouldn't think flash would be the issue unless the board file is misconfigured or it's some slightly-different flash variant. The PSRAM looks promising as the enabling variable. You should be able to get some module names and line numbers from `Backtrace:0x40386810:0x3fcae9800x4038706e:0x3fcae9a0 0x403799c5:0x3fcae9c0 0x403799d9:0x3fcD` using the `[decode_backtrace.py](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/tools/decode_backtrace.py)` utility.

fivesixzero, 2022-03-29:
I haven't used `decode_backtrace` before, and in my environment it wouldn't run - couldn't find the `addr2line` utility at first. So I did some digging and read up on the `idf.py monitor` function. It looks like `monitor` will do the same decode on the fly.

After setting the dev environment target to `esp32s3` then copying the `firmware.elf` output to `build/circuitpython.elf` I ran the monitor utility (`idf.py -p /dev/ttyUSB0 monitor`) and got this idf-parsed output from the debug console:

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x1654
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f78
entry 0x403b6248
I (29) boot: ESP-IDF 121ddb87e-dirty 2nd stage bootloader
I (30) boot: compile time 15:13:53
I (30) boot: chip revision: 0
I (32) boot.esp32s3: Boot SPI Speed : 40MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (85) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (92) boot:  4 uf2              factory app      00 00 00410000 00040000
I (100) boot:  5 user_fs          Unknown data     01 81 00450000 00bb0000
I (107) boot: End of partition table
I (112) boot: Defaulting to factory image
E (116) esp_image: image at 0x410000 has invalid magic byte (nothing flashed here?)
E (125) boot: Factory app partition is not bootable
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=49cb8h (302264) map
I (214) esp_image: segment 1: paddr=00059ce0 vaddr=3fc99d10 size=055d0h ( 21968) load
I (225) esp_image: segment 2: paddr=0005f2b8 vaddr=40374000 size=00d60h (  3424) load
I (227) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=1132ach (1127084) map
I (511) esp_image: segment 4: paddr=001732d4 vaddr=40374d60 size=14fb0h ( 85936) load
I (536) esp_image: segment 5: paddr=0018828c vaddr=50000000 size=01010h (  4112) load
I (538) esp_image: segment 6: paddr=001892a4 vaddr=600fe000 size=00028h (    40) load
I (552) boot: Loaded app from partition at offset 0x10000
I (552) boot: Disabling RNG early entropy source...
I (565) opi psram: vendor id : 0x0d (AP)
I (565) opi psram: dev id    : 0x02 (generation 3)
I (570) opi psram: density   : 0x03 (64 Mbit)
I (571) opi psram: good-die  : 0x01 (Pass)
I (574) opi psram: Latency   : 0x01 (Fixed)
I (579) opi psram: VCC       : 0x01 (3V)
I (584) opi psram: SRF       : 0x01 (Fast Refresh)
I (589) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (595) opi psram: BurstLen  : 0x01 (32 Byte)
I (600) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (606) opi psram: DriveStrength: 0x00 (1/1)
W (611) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (621) spiram: Found 64MBit SPI RAM device
I (625) spiram: SPI RAM mode: sram 80m
I (630) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (637) cpu_start: Pro cpu up.
I (641) cpu_start: Starting app cpu, entry point is 0x4037843c
0x4037843c: call_start_cpu1 at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/esp_system/port/cpu_start.c:161

I (0) cpu_start: App cpu up.
I (1073) spiram: SPI SRAM memory test OK
I (1083) cpu_start: Pro cpu start user code
I (1083) cpu_start: cpu freq: 240000000
I (1083) cpu_start: Application information:
I (1086) cpu_start: Project name:     circuitpython
I (1092) cpu_start: App version:      7.3.0-alpha.0-247-gdd7318244-di
[... 179 lines cut ...]

0x420c7b35: ieee80211_timer_do_process at ??:?

abort() was called at PC 0x40379ccf on core 0
0x40379ccf: lock_acquire_generic at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/newlib/locks.c:141

Backtrace:0x40378412:0x3fcae5800x40380a89:0x3fcae5a0 0x4038785a:0x3fcae5c0 0x40379ccf:0x3fcae630 0x40379de5:0x3fcae660 0x40379eb4:0x3fcae680 0x40050965:0x3fcae6b0  |<-CORRUPTED
0x40378412: panic_abort at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/esp_system/panic.c:402

0x40380a89: _frxt_setup_switch at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/freertos/port/xtensa/portasm.S:70

0x4038785a: abort at ??:?

0x40379ccf: lock_acquire_generic at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/newlib/locks.c:141

0x40379de5: _lock_acquire_recursive at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/newlib/locks.c:168

0x40379eb4: __retarget_lock_acquire_recursive at /home/me/code/circuitpython/ports/espressif/build-espressif_esp32s3_box/esp-idf/../../esp-idf/components/newlib/locks.c:324 (discriminator 3)

ELF file SHA256: 11489ef08d80ed0f

CPU halted.
```

I've got some time today so I'm going to keep digging but, frankly, this low-level ESP stuff is all new territory for me. I'm eager to learn but cautious about getting lost in minutia. :joy: May have some questions but I'll ask on Discord if I do.

fivesixzero, 2022-03-29:
Digging into the backtrace didn't point to a smoking gun. Brought back memories of dealing with memory issues in past lives...

So I decided to disable PSRAM altogether in `sdkconfig`. After doing that the build succeeded and the error was gone, everything worked properly.

That seems to narrow this down to either a `sdkconfig` issue or an issue with `esp-idf` relating to PSRAM allocation, maybe? Beside having 8 MB (vs 2 MB) of PSRAM this board also has a display that gets initialized on boot, so that might be causing some interactions as well.

This is new territory for me so any advice on where to look next would be appreciated.

dhalbert, 2022-03-29:
The `esp_lcd` code in the ESP-IDF allocates a buffer from the heap:
https://github.com/espressif/esp-idf/blob/c29343eb94d2f2ca17b3a5b38c82452e556147f2/components/esp_lcd/src/esp_lcd_rgb_panel.c#L134
(thanks @kmatch)
@anecdata @kmatch is this the cause of this problem?

@fivesixzero is this duplicative of the discussion I was having with @kmatch in #circuitpython-dev just recently?

fivesixzero, 2022-03-29:
> @fivesixzero is this duplicative of the discussion I was having with @kmatch in #circuitpython-dev just recently?

At the very least it looks like it's related. I'm still pretty green when it comes to low-level esp stuff though so I didn't want to cross the streams without a bit more understanding. :)

makermelissa, 2022-07-11:
@tannewt this is probably related the issue I noticed with the Web Workflow not working on the ESP Box and ESP Box Lite.

fivesixzero, 2022-07-15:
@tannewt - Looks like there's still an issue with WiFi on this port. Just tried a fresh build on a freshly reinstalled dev environment on my ESP-BOX (original flavor, not lite) and its still crashing when attempting `wifi.connect()`.

The behavior is a bit different now though. No stack dump at all this time, just a hard reboot that results in a WDT error on subqeuent boots, causing CircuitPython to load in safe mode on all subsequent boots.

```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 8.0.0-alpha.1-54-g145cce93d on 2022-07-15; ESP32-S3-Box-2.5 with ESP32S3
>>>
```

Here's the boot output from the debug UART header while running the above code which just scans and connects to WiFi. The only thing I changed was the addition of a `time.sleep(20)` before the WiFi scan and connect, to give me some extra time to make sure I've got the debug and normal consoles capturing before it rolls into the actual working code.

<details>
  <summary><i>Click to expand debug UART output...</i></summary>

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f90
entry 0x403b6254
I (29) boot: ESP-IDF 171e5af9 2nd stage bootloader
I (30) boot: compile time 10:52:08
I (30) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 40MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (69) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (77) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (84) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (92) boot:  4 uf2              factory app      00 00 00410000 00040000
I (99) boot:  5 user_fs          Unknown data     01 81 00450000 00bb0000
I (107) boot: End of partition table
I (111) boot: Defaulting to factory image
E (116) esp_image: image at 0x410000 has invalid magic byte (nothing flashed here?)
E (124) boot: Factory app partition is not bootable
I (130) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=45804h (284676) map
I (209) esp_image: segment 1: paddr=0005582c vaddr=3fc9a440 size=056bch ( 22204) load
I (220) esp_image: segment 2: paddr=0005aef0 vaddr=40374000 size=05128h ( 20776) load
I (227) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=111938h (1120568) map
I (505) esp_image: segment 4: paddr=00171960 vaddr=40379128 size=11314h ( 70420) load
I (531) esp_image: segment 5: paddr=00182c7c vaddr=50000000 size=01010h (  4112) load
I (532) esp_image: segment 6: paddr=00183c94 vaddr=600fe000 size=00028h (    40) load
I (547) boot: Loaded app from partition at offset 0x10000
I (547) boot: Disabling RNG early entropy source...
I (560) opi psram: vendor id : 0x0d (AP)
I (560) opi psram: dev id    : 0x02 (generation 3)
I (560) opi psram: density   : 0x03 (64 Mbit)
I (564) opi psram: good-die  : 0x01 (Pass)
I (568) opi psram: Latency   : 0x01 (Fixed)
I (573) opi psram: VCC       : 0x01 (3V)
I (578) opi psram: SRF       : 0x01 (Fast Refresh)
I (583) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (589) opi psram: BurstLen  : 0x01 (32 Byte)
I (594) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (600) opi psram: DriveStrength: 0x00 (1/1)
[... 147 lines cut ...]
I (82) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (97) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (105) boot:  4 uf2              factory app      00 00 00410000 00040000
I (112) boot:  5 user_fs          Unknown data     01 81 00450000 00bb0000
I (120) boot: End of partition table
I (124) boot: Defaulting to factory image
E (129) esp_image: image at 0x410000 has invalid magic byte (nothing flashed here?)
E (137) boot: Factory app partition is not bootable
I (143) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=45804h (284676) map
I (222) esp_image: segment 1: paddr=0005582c vaddr=3fc9a440 size=056bch ( 22204) load
I (228) esp_image: segment 2: paddr=0005aef0 vaddr=40374000 size=05128h ( 20776) load
I (235) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=111938h (1120568) map
I (515) esp_image: segment 4: paddr=00171960 vaddr=40379128 size=11314h ( 70420) load
I (535) esp_image: segment 5: paddr=00182c7c vaddr=50000000 size=01010h (  4112) load
I (537) esp_image: segment 6: paddr=00183c94 vaddr=600fe000 size=00028h (    40) load
I (551) boot: Loaded app from partition at offset 0x10000
...[snipped, since rest is like the previous boot]...
```

</details>

Like prior tests, I just did a clean build and ran `make BOARD=espressif_esp32s3_box DEBUG=1 flash PORT=/dev/ttyACM0`, which did the build/flash just fine. Other CircuitPython things seem to work fine except for WiFi connect causing this crash.

One thing I noticed is that this board doesn't have a [`tinyuf2` build](https://github.com/adafruit/tinyuf2/tree/master/ports/espressif/boards) for it (yet?), so it relies on a direct flash using the `make` scripts. The [board page on cpy.org](https://circuitpython.org/board/espressif_esp32s3_box/) also doesn't have any bootloader flashing details. Don't know if that would cause any issues by itself but I figured its worth mentioning.

tannewt, 2022-07-15:
@fivesixzero Thanks for the retest! I wish it was fixed. :-)

Last time I had a TG1 issue, I had to put in prints to find where it was hanging.

jepler, 2022-11-16:
We believe this was closed by #7140.

closed 2022-11-16 by jepler