adafruit/circuitpython #6215: mpy-cross build failure on MacOS

jerryneedell, 2022-03-30:
On my M1 Mac Mini running MacOS 12.3

I did a fresh clone of CircuitPython and attempted to build mpy-cross but it fails with this:

```
jerryneedell@Mac-mini circuitpython % make -C mpy-cross      
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/moduledefs.h
QSTR updated
../py/vm.c:742:25: error: array index -3 refers past the last possible element for an array in 64-bit address space containing 64-bit (8-byte) elements (max possible 2305843009213693952 elements) [-Werror,-Warray-bounds]
                        sp[-MP_OBJ_ITER_BUF_NSLOTS + 1] = MP_OBJ_NULL;
                        ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~
../py/vm.c:269:13: note: array 'sp' declared here
            mp_obj_t *sp = code_state->sp;
            ^
../py/vm.c:743:25: error: array index -2 refers past the last possible element for an array in 64-bit address space containing 64-bit (8-byte) elements (max possible 2305843009213693952 elements) [-Werror,-Warray-bounds]
                        sp[-MP_OBJ_ITER_BUF_NSLOTS + 2] = obj;
                        ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~
../py/vm.c:269:13: note: array 'sp' declared here
            mp_obj_t *sp = code_state->sp;
            ^
../py/vm.c:754:25: error: array index -3 refers past the last possible element for an array in 64-bit address space containing 64-bit (8-byte) elements (max possible 2305843009213693952 elements) [-Werror,-Warray-bounds]
                    if (sp[-MP_OBJ_ITER_BUF_NSLOTS + 1] == MP_OBJ_NULL) {
                        ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~
../py/vm.c:269:13: note: array 'sp' declared here
            mp_obj_t *sp = code_state->sp;
            ^
../py/vm.c:755:31: error: array index -2 refers past the last possible element for an array in 64-bit address space containing 64-bit (8-byte) elements (max possible 2305843009213693952 elements) [-Werror,-Warray-bounds]
                        obj = sp[-MP_OBJ_ITER_BUF_NSLOTS + 2];
                              ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~
../py/vm.c:269:13: note: array 'sp' declared here
            mp_obj_t *sp = code_state->sp;
            ^
4 errors generated.
make: *** [build/py/vm.o] Error 1

```
After a suggestion by @jepler  I tried the following patch to the mpy-cross.mk
```
jerryneedell@Mac-mini espressif % git diff
diff --git a/mpy-cross/mpy-cross.mk b/mpy-cross/mpy-cross.mk
index 642fe9b5a..c7bd91ad1 100644
--- a/mpy-cross/mpy-cross.mk
+++ b/mpy-cross/mpy-cross.mk
@@ -29,6 +29,8 @@ INC += -I$(BUILD)
 # compiler settings
 CWARN = -Wall -Werror
 CWARN += -Wpointer-arith -Wuninitialized
+# Disable errors for array-bounds warnings on "sp[-MP_OBJ_ITER_BUF_NSLOTS + 2]" access
+CWARN += -Wno-array-bounds
 CFLAGS = $(INC) $(CWARN) -std=gnu99 $(CFLAGS_MOD) $(COPT) $(CFLAGS_EXTRA)
 CFLAGS += -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables
 
```

This appears to resolve the issue.

Before creating a PR, I would like to understand what changed and see if anyone else can reproduce this issue. 

My Mac has the following gcc compiler installed:
```
jerryneedell@Mac-mini espressif % gcc -v
Apple clang version 13.1.6 (clang-1316.0.21.2)
Target: arm64-apple-darwin21.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jepler, 2022-04-04:
micropython fixed it in some other way: https://github.com/micropython/micropython/issues/8467 -> https://github.com/micropython/micropython/commit/bb70874111dbb246624a68c013e8f1c3245ca0d8

we'll naturally get this in a future merge of micropython, thought it may not occur before 8.0. So if it's vexing you, we should consider cherry-picking the change now instead of waiting.

jerryneedell, 2022-04-04:
> 

I'm fine waiting for it to be fixed by the upstream merge whenever it happens. Thanks.

jepler, 2022-04-11:
We think we fixed this, please re-open if needed.

closed 2022-04-11 by jepler

askpatrickw, 2022-04-11:
FYI: I did some builds off of lastest main on MACos the past two days which have worked.

PR #6231 (2022-04-05): py/vm: Prevent array bound warning when using -MP_OBJ_ITER_BUF_NSLOTS.
This warning can happen on clang 13.0.1 building mpy-cross:
```
../py/vm.c:748:25: error: array index -3 refers past the last possible
  element for an array in 64-bit address space containing 64-bit (8-byte)
  elements (max possible 2305843009213693952 elements)
  [-Werror,-Warray-bounds]
                        sp[-MP_OBJ_ITER_BUF_NSLOTS + 1] = MP_OBJ_NULL;
                        ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Using pointer access instead of array access works around this warning.

This is a cherry-pick of https://gi [... cut]