adafruit/circuitpython #6220: QTPY ESP32C3 REPL unstable

jerryneedell, 2022-04-01:
The REPL on the QTPY ESP32C# appears to be dropping characters.
I use rshell with my ESP32C3 DevKit (I have modified it to use binascii) and it works well.
On the QPYU ESP32C3, rshell is unable to establish a connection to the board.

If I use screen I can get to the REPL, but cutting and pasting code often misses characteres.
Here is an example of executing help('modules') and you can see the erratic formatting of the output. The columns should; be aligned.

```
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; Adafruit QT P
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; Adafruit QT Py ESP32C3 with ESP32-C3FN4
>>> help('modules')
__future__        collections       neopixel_write    terminalio
__main__         digitalio         onewireio         time
_asyncio         displayio         os               touchio
_bleio            errno             pulseio          traceback
adafruit_bus_device                 espidf           pwmio             ulab
adafruit_bus_device.i2c_device     fontio            rainbowio         ulab
adafruit_bus_device.spi_device     framebufferio     random            ulab.fft
adafruit_pixelbuf gc               re                ulab.linalg
analogio          getpass           rgbmatrix         ulab.numpy
array            gifio             rtc              ulab.scipy
atexit            i2cperipheral     sdcardio         ulab.scipy.linalg
audiocore         io               select            ulab.scipy.optimize
audiomixer        ipaddress        sharpdisplay      ulab.scipy.signal
binascii          json              socketpool        ulab.scipy.special
bitbangio         keypad           ssl               ulab.utils
bitmaptools      math              storage           uselect
board            mdns              struct           vectorio
builtins          microcontroller   supervisor        
busio             micropython       synthio          wifi
canio             msgpack           sys
Plus any modules on the filesystem
>>> 
```

prplz, 2022-05-11:
Test on some other boards, the issue is only present on the USB CDC. All is good on TX/RX UART pins.

RetiredWizard, 2022-05-22:
Confirming that after building with DEBUG_UART, the REPL is fine over TX/RX. 

I was also surprised that the Circuit Python build uses more memory than the MicroPython usb build but then I know it doesn't make much sense to give this board much attention until the BLE REPL is available.

dhalbert, 2022-07-06:
Check buffer size required by ESP-IDF for this.

askpatrickw, 2022-07-26:
This is totally anecdotal, but in playing with the C3 boards lately, I have also seen this. Ex: If run `dir(boards)` several times, I will sometimes be missing entries, or missing the # for a pin (D1, D, D3,..).

I assumed this was something with my setup using tio on MacOS in iTerm.

evildave666, 2022-07-26:
Have seen this on the Lolin C3 Mini as well. Webserial is fine.

PR #6660 (2022-07-28): Improve USB to Serial/JTAG TX
It had a tight 5ms timeout before that caused some characters to
drop. Now the wait is longer and reset after a successful transmit.
This follows what MicroPython does.

Fixes #6220
commit bea95522 (2022-07-28): Improve USB to Serial/JTAG TX

It had a tight 5ms timeout before that caused some characters to
drop. Now the wait is longer and reset after a successful transmit.
This follows what MicroPython does.

Fixes #6220
closed 2022-07-28 by dhalbert