adafruit/circuitpython #6227: PyPortal: board.DISPLAY is None returns wrong value

bablokb, 2022-04-02:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.4 on 2022-03-31; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
Adafruit CircuitPython 7.2.4 on 2022-03-31; Adafruit PyPortal with samd51j20
>>> import board
>>> import displayio
>>> board.DISPLAY
<Display>
>>> displayio.release_displays()
>>> board.DISPLAY
None
>>> type(board.DISPLAY)
<class 'NoneType'>
>>> board.DISPLAY is None
False
>>> foo=None
>>> foo
>>> type(foo)
<class 'NoneType'>
>>> foo is None
True
```

### Behavior

After displayio.release_displays() the attribute board.DISPLAY is None but "is None" returns False.

Note that  I know I should not call this method for boards with integrated displays, but errors happen. But I  expect "isNone" to always return True for variables of type 'NoneType'.

jepler, 2022-07-06:
The way we make the board.DISPLAY object "be None" is not quite correct, because it doesn't make it obey an "is" test. We need to define a proper way to check for the display being deinitialized that doesn't have this problem. For now, it is possible to use `type(board.DISPLAY) is type(None)`, but we recognize it's not ideal.

deshipu, 2022-07-06:
How about `if not board.DISPLAY`?

jepler, 2022-07-07:
> How about `if not board.DISPLAY`?

Unfortunately this fake-None is also treated as a true value in truthy contexts :-(

deshipu, 2022-07-07:
We don't even need to call it None, as long as it's false-y.

tannewt, 2022-07-07:
> > How about `if not board.DISPLAY`?
> 
> Unfortunately this fake-None is also treated as a true value in truthy contexts :-(

We could change this by having a custom unary op for NoneType.

> Maybe we could make it an int zero instead of a pointer?

Make what an int zero? We can't modify the pointer because it is in a ROM table. That's why it changes the object type now.

deshipu, 2022-07-07:
> We could change this by having a custom unary op for NoneType.
Wouldn't it be just as simple as having a `NoDisplay` type with a `__nonzero__()` method that returns `False`?

dhalbert, 2022-07-07:
Not thinking well. But we could make it an empty list or tuple or bytearray, all of which are False-y?

deshipu, 2022-07-07:
None should be False-y too? Not sure why this doesn't work?

dhalbert, 2022-07-07:
Because `None` is a singleton, and it does an address comparison only.

deshipu, 2022-07-07:
Ideally, it should be `board.DISPLAYS[0]` — perhaps we could change to that eventually? Have both `board.DISPLAY` and `board.DISPLAYS` for a time...

jepler, 2022-07-07:
Using a list (or function) board.DISPLAYS does help with the problem, because the list would reside in RAM and thus the object pointer could actually be updated. It would also help with the rare boards that have more than one display.

There's still the problem of what if you write `d = board.DISPLAYS[0]; displayio.release_displays()` and later refer to `d`. It needs a way to be turned into a deinitialized display object.

I have an incomplete and non-working branch that introduces the Deinitialized object, which might be of interest. It tries to solve the problem generally for other types of objects as well: https://github.com/adafruit/circuitpython/compare/main...jepler:circuitpython:deinited-type?expand=1

This needs to be elaborated on so that it builds and works. The object would always act False in a truthy context (so you could say 'if not board.DISPLAY`, unlike currently).