adafruit/circuitpython #6228: Uart on Teensy 4/4.1 act differently between 7.24 and 7.30B0

KurtE, 2022-04-02:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.4 on 2022-03-31; Teensy 4.0 with IMXRT1062DVJ6A
```

### Code/REPL

```python
import board
import digitalio
import time
import busio
import supervisor

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
loop_count = 0
uart1 = busio.UART(board.D1, board.D0, baudrate=1000000)
uart2 = busio.UART(board.D8, board.D7, baudrate=1000000)

while True:
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("*** Serial input ***", text)
        b = bytearray()
        b.extend(text)
        uart1.write(b)
    waiting_count = uart1.in_waiting
    if waiting_count:
        print("*** Uart1 input(", waiting_count, ") ***")
        data = uart1.read(waiting_count)
        uart2.write(data)
        print("\t len:",len(data), "data:", data.decode())
    waiting_count = uart2.in_waiting
    if waiting_count:
        print("*** Uart2 input(", waiting_count, ") ***")
        data = uart2.read(waiting_count)
        print("\t len:",len(data), "data:", data.decode())
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.8)
    loop_count += 1
    if (loop_count & 0x1F) == 0:
        print(loop_count)
```

### Behavior

When I run int on T4 or T4.1 with 7.2.4, and I have jumpers between pins 0 and 1 and 7 and 8, when I type something in
I see it goes through first to logical Serial1 to Serial2 back to Serial...
```
code.py output:
abcd
*** Serial input *** abcd
*** Uart1 input( 4 ) ***
	 len: 4 data: abcd
*** Uart2 input( 4 ) ***
	 len: 4 data: abcd
efgh
*** Serial input *** efgh
*** Uart1 input( 4 ) ***
	 len: 4 data: efgh
*** Uart2 input( 4 ) ***
	 len: 4 data: efgh
xxxx
*** Serial input *** xxxx
*** Uart1 input( 4 ) ***
	 len: 4 data: xxxx
*** Uart2 input( 4 ) ***
	 len: 4 data: xxxx
32
64

```

Now if I run it with Beta 1 (or current sources synced this morning)
Adafruit CircuitPython 7.3.0-beta.0-2-g87e59a444 on 2022-04-02; Teensy 4.1 with IMXRT1062DVJ6A
It stops working properly as you can see in this output:
```
code.py output:
abcd
*** Serial input *** abcd
*** Uart1 input( 4 ) ***
	 len: 4 data: abcd
*** Uart2 input( 63 ) ***
	 len: 63 data: ( 4 ) ***
abcd	 len: 4 data: abcd
*** Uart2 input( 63 ) ***

*** Uart2 input( 63 ) ***
	 len: 63 data:  abcd
*** Uart2 input( 63 ) ***

*** Uart2 input( 63 ) ***

*** Uart2 input( 63 ) ***
	 len: 63 data: ***

*** Uart2 input( 63 ) ***

*** Uart2 input( 63 ) ***

*** Uart2 input( 63 ) ***
	 len: 63 data: ***

*** Uart2 input( 63 ) ***
```
Note: I broke out of it by ctrl-c

### Additional information

Not sure if best to create an issue for this or something on forum, or Teensy forum? 

Thanks 
Kurt

dhalbert, 2022-04-02:
This is the right place to report this, thanks.

jepler, 2022-04-04:
```
f5d90fc84 Switch to port_serial_* hooks
45f9522a6 Fix EVK status led to be inverted
96f5eec2e Add Teensy 4.1 power pin and fix SWD for DEBUG=1
83593a155 Start of USB host API
```
These are the commits between 7.2.4 and 7.3.0-beta.0 that touch the mimxrt1011 port. Both 83593a155 and f5d90fc84 touch the uart. If you take a look at those commits, it may help you isolate what's changed and help you prepare a pull request.

KurtE, 2022-04-04:
@jepler  - I am taking a look at it. will look at the two changes you mentioned.

Current sketch, I setup the way I have my arduino sketches...
```python
import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

led = DigitalInOut(board.LED)
led.direction = Direction.OUTPUT

check_pin = DigitalInOut(board.D2)
check_pin.direction = Direction.INPUT
check_pin.pull = Pull.DOWN
uart1 = None

last_check_value = check_pin.value

def CheckAndEchoUart(index, uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        print("*** Uart", index, " input(", waiting_count, ") ***")
        data = uart.read(waiting_count)
       
        if (index == 1):
            print("Echoed back len:",len(data), "data:", data.decode())
        else: 
            print(index,">>")
            uart.write(data)

while True:
    if check_pin.value:
        if uart1 == None:
            # we have not yet tried to iniatialize the uarts
            print("Initialize Uarts")
            uart1 = busio.UART(board.TX, board.RX, baudrate=1000000)
            uart2 = busio.UART(board.D8, board.D7, baudrate=1000000)
            uart3 = busio.UART(board.D14, board.D15, baudrate=1000000)
            uart4 = busio.UART(board.D17, board.D16, baudrate=1000000)
            uart5 = busio.UART(board.D20, board.D21, baudrate=1000000)
            uart6 = busio.UART(board.D24, board.D25, baudrate=1000000)
            uart7 = busio.UART(board.D29, board.D28, baudrate=1000000)
            uart8 = busio.UART(board.D35, board.D34, baudrate=1000000)
           
        if supervisor.runtime.serial_bytes_available:
            text = input()
            print("*** Serial input ***", text)
            b = bytearray()
            b.extend(text)
            uart1.write(b)
           
        CheckAndEchoUart(1, uart1)
        CheckAndEchoUart(2, uart2)
        CheckAndEchoUart(3, uart3)
        CheckAndEchoUart(4, uart4)
        CheckAndEchoUart(5, uart5)
        CheckAndEchoUart(6, uart6)
        CheckAndEchoUart(7, uart7)
        CheckAndEchoUart(8, uart8)
    # cycle colors
    if led.value:
        led.value = False
    else:
        led.value = True
    time.sleep(0.5)
```
The fun things with sketch like this, is I can try different combinations.  
For example from 1->5->1  
Appears to work. (that is the numbers are the uart number...
```
abdd
*** Serial input *** abdd
*** Uart 5  input( 4 ) ***
5 >>
*** Uart 1  input( 4 ) ***
Echoed back len: 4 data: abdd

But 1-2-1 does not, so could be configuration issue, investigating.

```

KurtE, 2022-04-04:
update: With the above sketch, as I mentioned it did not work from 1->2->1... But did 1->3->1 and 1->5->1...
Have not tried all of them.

So first thought is one or more configuration issues with the pins or uart.. But nothing changed in those areas as far as I can tell, and it looked like it should be setting the pins in to the correct MUX setting as well as the proper Input select value... 

So: I wonder what might be not working RX or TX... so connected up Logic analyzer to pins 0, 1, 7, 8 (Serial1-LPUART6) and Serial2 (LPUART4) and removed the jumpers.  Do text I typed would simply come out Pin 1... 
[image]

But when I ran the sketch and typed text, the Logic Analyzer showed output on pin 1 as expected, but a lot more on Pin 8.  Which matches the text begin output by print().

My guess is there is some new debug code somewhere that someone is outputting over serial port...

KurtE, 2022-04-04:
Looking in the Circuitpython\ports\mimxrt10xx\supervisor\print.c I see:
```
// TODO: Switch this to using DEBUG_UART.

// static LPUART_Type *uart_instance = LPUART1; // evk
static LPUART_Type *uart_instance = LPUART4; // feather 1011
```
The only change I see in blame view mode is the TODO line, but now to see who might be calling
port_serial_write_substring(...)

KurtE, 2022-04-04:
@tannewt  

I believe the issue has to do with the change:
https://github.com/adafruit/circuitpython/commit/f5d90fc84f001cfd9fa46bbbdcd0af9e8dffe43d#diff-f03765ae6bb99c05c83b89eab7e1043f8083b00003434e5980c8d38f8906dd71R291

With the changes to supervisor\shared\serial.c which now the serial_write_substring is calling:
    port_serial_write_substring(text, length);

Which the version in that file is weak and as such the one in the port is being used...  And for the T4.x is UART4.

KurtE, 2022-04-05:
@tannewt @jepler @mjs513

I was able to make the code work correctly, by not having the Serial port stubs be created.  
That is in the file: circuitpython\ports\mimxrt10xx\supervisor\serial.c

Right below the comment I added an if:
```
// TODO: Switch this to using DEBUG_UART.
#if defined(USE_DEBUG_PORT_CODE)
// static LPUART_Type *uart_instance = LPUART1; // evk
static LPUART_Type *uart_instance = LPUART4; // feather 1011
// static LPUART_Type *uart_instance = LPUART2; // feather 1062
```
and added end if at the end: #endif //USE_DEBUG_PORT_CODE

So none of the code compiled and as such the higher level security code used the default (weak) stubs which did not do anything,  After this, the code worked pretty well:

```python

import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

led = DigitalInOut(board.LED)
led.direction = Direction.OUTPUT

check_pin = DigitalInOut(board.D2)
check_pin.direction = Direction.INPUT
check_pin.pull = Pull.DOWN
uart1 = None

last_check_value = check_pin.value

def CheckAndEchoUart(index, uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        print("*** Uart", index, " input(", waiting_count, ") ***")
        data = uart.read(waiting_count)
       
        if (index == 1):
            print("Echoed back len:",len(data), "data:", data.decode())
        else: 
            print(index,">>", data.decode())
            uart.write(data)

while True:
    if check_pin.value:
        if uart1 == None:
            # we have not yet tried to iniatialize the uarts
            print("Initialize Uarts")
            uart1 = busio.UART(board.TX, board.RX, baudrate=500000)
            uart2 = busio.UART(board.D8, board.D7, baudrate=500000)
            uart3 = busio.UART(board.D14, board.D15, baudrate=500000)
            uart4 = busio.UART(board.D17, board.D16, baudrate=500000)
            uart5 = busio.UART(board.D20, board.D21, baudrate=500000)
            uart6 = busio.UART(board.D24, board.D25, baudrate=500000)
            uart7 = busio.UART(board.D29, board.D28, baudrate=500000)
            uart8 = busio.UART(board.D35, board.D34, baudrate=500000)
           
        if supervisor.runtime.serial_bytes_available:
            text = input()
            print("*** Serial input ***", text)
            b = bytearray()
            b.extend(text)
            uart1.write(b)
           
        CheckAndEchoUart(1, uart1)
        CheckAndEchoUart(2, uart2)
        CheckAndEchoUart(3, uart3)
        CheckAndEchoUart(4, uart4)
        CheckAndEchoUart(5, uart5)
        CheckAndEchoUart(6, uart6)
        CheckAndEchoUart(7, uart7)
        CheckAndEchoUart(8, uart8)
    # cycle colors
    if led.value:
        led.value = False
    else:
        led.value = True
    time.sleep(0.5)
```
[image]
I had The Tx of one serial port connected to the RX of the next one... 

```
abcdefghijk
*** Serial input *** abcdefghijk
*** Uart 2  input( 11 ) ***
2 >> abcdefghijk
*** Uart 3  input( 9 ) ***
3 >> abcdefghi
*** Uart 4  input( 8 ) ***
4 >> abcdefgh
*** Uart 5  input( 8 ) ***
5 >> abcdefgh
*** Uart 6  input( 7 ) ***
6 >> abcdefg
*** Uart 7  input( 6 ) ***
7 >> abcdef
*** Uart 8  input( 5 ) ***
8 >> abcde
*** Uart 1  input( 5 ) ***
Echoed back len: 5 data: abcde
*** Uart 3  input( 2 ) ***
3 >> jk
*** Uart 4  input( 1 ) ***
4 >> i
*** Uart 6  input( 1 ) ***
6 >> h
*** Uart 7  input( 1 ) ***
7 >> g
*** Uart 8  input( 2 ) ***
8 >> fg
*** Uart 1  input( 2 ) ***
Echoed back len: 2 data: fg
*** Uart 4  input( 2 ) ***
4 >> jk
*** Uart 5  input( 1 ) ***
5 >> i
*** Uart 7  input( 1 ) ***
7 >> h
*** Uart 5  input( 2 ) ***
5 >> jk
*** Uart 6  input( 1 ) ***
6 >> i
*** Uart 7  input( 1 ) ***
7 >> i
*** Uart 8  input( 1 ) ***
8 >> h
*** Uart 1  input( 1 ) ***
Echoed back len: 1 data: h
*** Uart 6  input( 2 ) ***
6 >> jk
*** Uart 8  input( 1 ) ***
8 >> i
*** Uart 1  input( 1 ) ***
Echoed back len: 1 data: i
*** Uart 7  input( 2 ) ***
7 >> jk
*** Uart 8  input( 2 ) ***
8 >> jk
*** Uart 1  input( 2 ) ***
Echoed back len: 2 data: jk
```

So it now works, but I don't know the proper fix for your setup.  Typically for things like this, there would probably be a define somewhere where is set either in an options file or as part of build or ... 

Let me know if you would like some form of PR and if so, how you would setup the #define and #if... 

Kurt

jepler, 2022-04-06:
Thanks for doing the detective work!  Opening a PR next would give a good basis for further discussion. Tannewt is out for a few weeks, so we'll probably end up making a decision without him. Most likely, disabling this (with an easy way to re-enable it) is the way to go. Marking the UART as "in use" (so that it can't be constructed via busio.UART or board.UART) when the debug UART is enabled would also be good, because at least you'd get an exception instead of just something that doesn't work as expected.

jepler, 2022-04-06:
Closing in favor of #6241 which has better information. Note that it's always possible to edit the title & initial message of an issue to reflect changing understanding of the problem; doing that or creating a fresh issue are both OK.