adafruit/circuitpython #6229: Adafruit Feather RP2040 with rp2040 - Fails to create uart on D24 and D25

KurtE, 2022-04-02:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
import board
import digitalio
import time
import busio
import supervisor

############################
"""CircuitPython Essentials UART possible pin-pair identifying script"""
"""

import board
import busio
from microcontroller import Pin

def is_hardware_uart(tx, rx):
    try:
        p = busio.UART(tx, rx)
        p.deinit()
        return True
    except ValueError:
        return False

def get_unique_pins():
    exclude = ['NEOPIXEL', 'APA102_MOSI', 'APA102_SCK']
    pins = [pin for pin in [
        getattr(board, p) for p in dir(board) if p not in exclude]
            if isinstance(pin, Pin)]
    unique = []
    for p in pins:
        if p not in unique:
            unique.append(p)
    return unique

for tx_pin in get_unique_pins():
    print("start tx: ", tx_pin)
    for rx_pin in get_unique_pins():
        if rx_pin is tx_pin:
            continue
        if is_hardware_uart(tx_pin, rx_pin):
            print("RX pin:", rx_pin, "\t TX pin:", tx_pin)
            
            
##########################
"""
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
loop_count = 0
uart1 = busio.UART(board.D1, board.D0, baudrate=1000000)
uart2 = busio.UART(board.D25, board.D24, baudrate=1000000)

while True:
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("*** Serial input ***", text)
        b = bytearray()
        b.extend(text)
        uart1.write(b)
    waiting_count = uart1.in_waiting
    if waiting_count:
        print("*** Uart1 input(", waiting_count, ") ***")
        data = uart1.read(waiting_count)
        uart2.write(data)
        print("\t len:",len(data), "data:", data.decode())
    waiting_count = uart2.in_waiting
    if waiting_count:
        print("*** Uart2 input(", waiting_count, ") ***")
        data = uart2.read(waiting_count)
        print("\t len:",len(data), "data:", data.decode())
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.8)
    loop_count += 1
    if (loop_count & 0x1F) == 0:
        print(loop_count)
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 52, in <module>
ValueError: Invalid pins

Code done running.

### Description

I was curious if the issue I found with Teensy 4/4.1 would replicate with the RP2040, so 
looking at the product page, I believe it has two UARTS  and looking at the pin out diagram:
https://learn.adafruit.com/adafruit-feather-rp2040-pico/pinouts

It looked like D24 and D25 were valid for RX1/TX1,  but when I tried it failed on the line...

Note: I then curious if any pins would work so went to the uart page:
https://learn.adafruit.com/circuitpython-essentials/circuitpython-uart-serial?gclid=Cj0KCQjw6J-SBhCrARIsAH0yMZjK2fy-7NfvWWWtwnJ0AteCGJzXWiwsAJxAXq2IplZnNILAEqKG7uoaArHCEALw_wcB

And copied in the code to detect valid rx/tx pins.  
which is commented out in the above.  but if I then remove it from comments and comment out the main code. 

And ran it and received no output.  So I put output line on the outer loop showing the tx pin being tried:
Output:

```
code.py output:
start tx:  board.A0
start tx:  board.A1
start tx:  board.A2
start tx:  board.A3
start tx:  board.D0
start tx:  board.D1
start tx:  board.D10
start tx:  board.D11
start tx:  board.D12
start tx:  board.D13
start tx:  board.D24
start tx:  board.D25
start tx:  board.D4
start tx:  board.D5
start tx:  board.D6
start tx:  board.D9
start tx:  board.MISO
start tx:  board.MOSI
start tx:  board.SCK
start tx:  board.SCL
start tx:  board.SDA
```

### Additional information

Sorry if this is a known issue..

anecdata, 2022-04-02:
You can have two UARTs on RP2040, but "you can only do two at a time, and only on separate peripherals, 0 and 1"
https://learn.adafruit.com/assets/107203
Both pairs in the code above use peripheral 1.

KurtE, 2022-04-02:
Thanks @anecdata,
giving it a try:
```
import board
import digitalio
import time
import busio
import supervisor

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
loop_count = 0
uart1 = board.UART()
uart1.baudrate = 1000000
uart2 = busio.UART(board.D25, board.D24, baudrate=1000000)

while True:
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("*** Serial input ***", text)
        b = bytearray()
        b.extend(text)
        uart1.write(b)
    waiting_count = uart1.in_waiting
    if waiting_count:
        print("*** Uart1 input(", waiting_count, ") ***")
        data = uart1.read(waiting_count)
        uart2.write(data)
        print("\t len:",len(data), "data:", data.decode())
    waiting_count = uart2.in_waiting
    if waiting_count:
        print("*** Uart2 input(", waiting_count, ") ***")
        data = uart2.read(waiting_count)
        print("\t len:",len(data), "data:", data.decode())
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.2)
    led.value = True
    led.value = True
    time.sleep(0.2)
    led.value = False
    time.sleep(0.8)
    loop_count += 1
    if (loop_count & 0x1F) == 0:
        print(loop_count)
```
And the board faulted.  I see in Serial portion of window:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Attempted heap allocation when VM not running.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

Wondering if it is the 1mbs I requested so trying to do 38400 instead.

KurtE, 2022-04-03:
Quick update, I believe my crashing the board, has to do with my attempts to use the board.UART 
and maybe specifically with me attempting to figure out how to set the baud rate on it. 

That is I tried also starting from simple Neiopixel example (docs should mention you need to download the library...)
And then tried to build out, to use uarts...
```python
import time
import board
import busio
import supervisor

import neopixel

led = neopixel.NeoPixel(board.NEOPIXEL, 1)

led.brightness = 0.3

#uart1 = board.UART()
#uart1.baudrate(38400)
#print(uart1.baudrate)

while True:
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("Serial input: ", text)
    led[0] = (255, 0, 0)
    time.sleep(0.5)
    led[0] = (0, 255, 0)
    time.sleep(0.5)
    led[0] = (0, 0, 255)
    time.sleep(0.5)
```

Was unclear how to set the baudrate as well it did not appear like this is following the docs, so  I looked at the
directory
```
>>> uart1 = board.UART()
>>> dir(uart1)
['__class__', '__enter__', '__exit__', '__next__', 'read', 'readinto', 'readline', 'write', '__del__', 'Parity', 'baudrate', 'deinit', 'in_waiting', 'reset_input_buffer', 'timeout']
>>> 
```
and saw: baudrate
So I tried accessing it and it faults.  Not sure of correct usage of it.
I tried assigning it, calling it, or printing it out, and board reboots.

Actually just having the uar1 = board.UART() line, 
hitting ctrl-c rebooted the board. 

Probably done, playing with this board for now, only tried it as was handy to try out the multiple uart code in other issue.

anecdata, 2022-04-04:
I think the pins issue is sorted out, but there should never be a crash into Safe Mode. @KurtE can we close this issue, and would you like to open a new issue for the safe mode?

dhalbert, 2022-04-04:
Hi, this is a dupe of #6213 and #6219. I have a fix, but there are still some peculiarities when connecting pins to unpowered pins on another board (#6219).

KurtE, 2022-04-04:
Sorry feel free to close if appropriate.  Especially if it is a duplicate. 

And wait to see what happens when I pick up the changes. 

Example  just now: 
Had simple Noepixel sketch running:
```
import time
import board
import busio
import supervisor

import neopixel

led = neopixel.NeoPixel(board.NEOPIXEL, 1)

led.brightness = 0.3

#uart1 = board.UART()
#uart1.baudrate(38400)
#print(uart1.baudrate)

while True:
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("Serial input: ", text)
    led[0] = (255, 0, 0)
    time.sleep(0.5)
    led[0] = (0, 255, 0)
    time.sleep(0.5)
    led[0] = (0, 0, 255)
    time.sleep(0.5)
```
I then ctrl+c into the serial monitor.
typed in: 
import board, busio
uart1 = board.UART)
<ctrl-d>
And board crashed, window went away and MU complained:

[image]

But will wait until I pull in updates and try from sources

KurtE, 2022-04-04:
@anecdata @dhalbert - Will go ahead and close this issue. 

The crashes do appear to be duplicates of the issues mentioned.  As I said in previous post will sync up and try with latest sources. 

however it is still very unclear on what is the proper way to use two hardware uarts of the RPI2040?

From the pinout page, I believe I have options on which pins to use for each:
[image]

So suppose I wish to use the ones I circled, and I would both to run at 1000000, I would think I should be able to do:
```
uart1 = busio.UART(board.D12, board.D13, baudrate=1000000)
uart2 = busio.UART(board.D6, board.9, baudrate=1000000)
```
or likewise the pins I showed in the original
```
uart1 = busio.UART(board.D1, board.D0, baudrate=1000000)
uart2 = busio.UART(board.D25, board.D24, baudrate=1000000)
```
And who knows, maybe I wish to set the parity and buffersize... 

What is the proper way to do this on this board?   I am not sure if this is a documentation issue and/or code issue?

With using the board.UART() object like:
```
>>> import board, busio
>>> uart1 = board.UART()
>>> dir(uart1)
['__class__', '__enter__', '__exit__', '__next__', 'read', 'readinto', 'readline', 'write', '__del__', 'Parity', 'baudrate', 'deinit', 'in_waiting', 'reset_input_buffer', 'timeout']
>>> 
```
So my impression is maybe I can set the baudrate?   Sorry again Python is not my native programming language, 
so unclear from the dir, if this is method, or asignment, but from Mu it looks like maybe baudrate() returns the current
baud rate... So how do I set it? ... 

But for now probably best to close this out and wait and see.

Thanks
Kurt

closed 2022-04-04 by KurtE

anecdata, 2022-04-04:
This seems to work:
```py
Adafruit CircuitPython 7.2.0-alpha.1-224-gac7a80753 on 2022-01-26; Adafruit Feather RP2040 with rp2040
>>> import board, busio
>>> uart0 = board.UART()
>>> uart0.deinit()
>>> 
>>> uart1 = busio.UART(board.D24, board.D25)
>>> 
>>> uart0 = busio.UART(board.D12, board.D13)
```
maybe there's a cleaner way to do the deinit?

KurtE, 2022-04-04:
Thanks, this appears to work on RP2040.  Here is an updated sketch I am running, where I also put in a fail safe in that if D5 is not connected to 3.3v it won't run the uart code...   I have it still playing with the neopixel, although a little differently...
```
import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

import neopixel

led_colors = [
    (255, 0, 0),
    (0, 255, 0),
    (0, 0, 255),
    (255, 255, 0),
    (255, 0, 255),
    (0, 255, 255),
    (255, 255, 255),
    (0, 0, 0),
]
color_index = 0
check_pin = DigitalInOut(board.D5)
check_pin.direction = Direction.INPUT
check_pin.pull = Pull.DOWN
uart1 = None
uart2 = None

led = neopixel.NeoPixel(board.NEOPIXEL, 1)

led.brightness = 0.3

last_check_value = check_pin.value

def CheckAndEchoUart(index, uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        print("*** Uart", index, " input(", waiting_count, ") ***")
        data = uart.read(waiting_count)
        
        if (index == 1):
            print("\t len:",len(data), "data:", data.decode())
        else:    
            uart2.write(data)

while True:
    if check_pin.value:
        if uart1 == None:
            # we have not yet tried to iniatialize the uarts
            uart1 = board.UART()
            uart1.deinit()
            uart1 = busio.UART(board.TX, board.RX, baudrate=1000000)
            uart2 = busio.UART(board.D24, board.D25, baudrate=1000000)
            
        if supervisor.runtime.serial_bytes_available:
            text = input()
            print("*** Serial input ***", text)
            b = bytearray()
            b.extend(text)
            uart1.write(b)
            
        CheckAndEchoUart(1, uart1)
        CheckAndEchoUart(2, uart2)
    # cycle colors
    led[0] = led_colors[color_index]
    color_index += 1
    if color_index >= len(led_colors):
        color_index = 0
    time.sleep(0.5)
```
And I changed it more like my normal teensy code. 
Where I have jumper from TX to D25 and another from D24 to RX... will import this version sort of back to Teensy, although no NeoPixel...

Will try for the heck of it on ESP32S3

anecdata, 2022-04-04:
I should have realized / tried before... doing `board.UART().deinit()` directly also works
```py
>>> import board, busio
>>> board.UART().deinit()
>>> uart1 = busio.UART(board.TX, board.RX, baudrate=1000000)
>>> uart2 = busio.UART(board.D24, board.D25, baudrate=1000000)
>>> 
```

KurtE, 2022-04-04:
Thanks, looks cleaner. 

right now waiting for esp32 build... 
Oops looks like I need to do something to make it build... does not like cmake... 
Back to playin