adafruit/circuitpython #6235: SCK in use after soft reboot on Feather m4 Express

muranyia, 2022-04-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-38-g862210b3f-dirty on 2022-03-24; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
import board
import busio

spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
```

### Behavior

After a soft reboot:
```python
Traceback (most recent call last):
  File "code.py", line 4, in <module>
ValueError: SCK in use
```
The only way out at this point is power cycling as I cannot init the SPI bus in  the REPL either.
I can catch the keyboard interrupt and do `spi.unlock()` before exiting but in case I'm rebooting because of a code error I still need to power cycle.

### Additional information

The CircuitPython firmware was custom built 1-2 weeks ago. I'm not attempting a fresh build as git could not pull some modules.

Neradoc, 2022-04-04:
What else are you doing with it ? Are you setting up a display ? If so, you need to use `displayio.release_displays()` first, or the display and SPI bus are retained so it can show the REPL when the board reloads.

muranyia, 2022-04-04:
In my full code I did `displayio.release_displays()` albeit not before setting up SPI but before creating the display.

When I put `displayio.release_displays()` in the minimal code example above, and *before setting up SPI*, it seems to solve the issue.

Interestingly, I did not have this problem while using i2c, only when I switched to SPI.

Thank you!

closed 2022-04-04 by muranyia