adafruit/circuitpython #6243: What about a Teensy MicroMod build?

KurtE, 2022-04-05:
Would be great to have the Teensy Micromod working on this as well.

Will play with it some, 

but not sure what PID/VID to use?

I noticed that the Teensy 4 and 4.1 have an Adafruit VID... 

Let me know what you think

dhalbert, 2022-04-05:
We used to give out PID's, but stopped because either the manufacturer or pid.codes are the best sources for non-Adafruit boards.

dhalbert, 2022-04-05:
Yes, we sell Teensy's :slightly_smiling_face:, but we refer people to PJRC sources for other than simple support.

We use separate PID's for the bootloader, for a running Arduino program, and for CircuitPython, to make it easier to identify what the board is running. Arduino cares about this, and some other programs do too. We used it to help Mu identify boards for a while, but now we identify based on the USB device interface names.

ladyada, 2022-04-05:
we require unique VID/PID for bootloader vs running circuitpy vs other runtimes. we used to not require it but had windows OS lockups.

the manufacturer (SFE) has a VID, you can ask them for a PID - they have approx 65000. or PJRC can provide one as it is a licensed device. that way it will pop up as a SFE or PJRC device, not an Adafruit device (since we do not manufacture or design or license the Teensy)

KurtE, 2022-04-06:
So far it does not look like anyone is interested, so will closet this one out.

ladyada, 2022-04-06:
@KurtE why don't you submit a PR for it since ya think it would be cool to have? do you have the hardware?

KurtE, 2022-04-07:
I went ahead and reopened this as Sparkfun did come through with a VID/PID:
VID: 0x1B4F
PID: 0x002E

@mjs513 and myself now have a at least partially working version of it.  
I just pushed up one of the last pieces need to build.  Still have formatting issues will work on that later 
It looks like the clang... used for previous PRs on other projects is not meet the same coding requirements. 

My Python skills are not that great, but having fun to do pin check sketch that shows you all of the names in board that map to the same physical pin and shows that when they change values:

```python
import board, busio, digitalio, time
from microcontroller import Pin
from digitalio import DigitalInOut, Direction, Pull

#led = digitalio.DigitalInOut(board.LED)
#led.direction = digitalio.Direction.OUTPUT

# lets first get a list of the boards pins
# temporary exclude led
exclude = ['USB_HOST_DM', 'USB_HOST_DP', 'USB_HOST_POWER', 'LED']

class pinNamesRecord(object):
    def __init__(self, pin, pin_name):
        self.pin = pin
        self.pin_names = pin_name
        self.last_check = False
        

print("check for all Pins")    

all_pins = {}
for board_item in dir(board):
    if board_item not in exclude:
        item_attr = getattr(board, board_item)
        if isinstance(item_attr, Pin):
            #print("Check pin: ", board_item, "(",item_attr, ")")
            pin_str = str(item_attr)
            if pin_str in all_pins:
                pnr = all_pins[pin_str]
                pnr.pin_names = pnr.pin_names + ' ' + board_item
                all_pins[pin_str] = pnr
                #print("\tDup: ", pnr.pin_names)
            else:
                # new one
                iopin = DigitalInOut(item_attr)
                iopin.direction = Direction.INPUT
                iopin.pull = Pull.UP
                pnr = pinNamesRecord(iopin, board_item)
                all_pins[pin_str] = pnr
                #print("\tnew: ", pnr.pin_names)
    
print("\nEnum pin list")
for pin_entry in all_pins:
    pny = all_pins[pin_entry]
    print(pin_entry, "\t", pny.last_check, "\t", pny.pin_names)
  
loop_count_no_changes = 0
while True:
    pin_changed = False
    for pin_entry in all_pins:
        pny = all_pins[pin_entry]
        new_pin_value = pny.pin.value
        if new_pin_value != pny.last_check:
            if loop_count_no_changes:
                loop_count_no_changes = 0
                print()
                
            pny.last_check = new_pin_value
            print(pin_entry, "\t", pny.last_check, "\t", pny.pin_names)
            pin_changed = True
    if pin_changed == False:
        loop_count_no_changes += 1
        if loop_count_no_changes < 80:
            print('.', end='')
        else:
            print('.')
            loop_count_no_changes = 0
            
    time.sleep(0.25)
```

This is sort of more interesting on micromod as they give different names to the different breakout boards, so will probably
try to map those as well as the Teensy pin numbers.

Example output:
```
board.D1 	 False 	 D1 TX TX1 UART_TX1
.
board.D1 	 True 	 D1 TX TX1 UART_TX1
....
board.D3 	 False 	 D3 PWM0
.
board.D3 	 True 	 D3 PWM0
.
board.D5 	 False 	 D5 MM_D1
..
board.D5 	 True 	 D5 MM_D1
.
board.D7 	 False 	 AUD_OUT D7 RX4
.
board.D7 	 True 	 AUD_OUT D7 RX4
................................................................................
..
```