adafruit/circuitpython #6249: Usage Example from Adafruit_CircuitPython_EPD States Output: "ValueError: SCK in use"

LTB2022, 2022-04-05:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0
```

### Code/REPL

```python
import digitalio
import busio
import board
from adafruit_epd.epd import Adafruit_EPD
from adafruit_epd.il0373 import Adafruit_IL0373

# create the spi device and pins we will need
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
ecs = digitalio.DigitalInOut(board.D12)
dc = digitalio.DigitalInOut(board.D11)
srcs = digitalio.DigitalInOut(board.D10)    # can be None to use internal memory
rst = digitalio.DigitalInOut(board.D9)    # can be None to not use this pin
busy = digitalio.DigitalInOut(board.D5)   # can be None to not use this pin

# give them all to our driver
print("Creating display")
# Using the 2.9" ThinkInk Greyscale EPD Display (296X128)
display = Adafruit_IL0373(296, 128, spi,          # 2.13" Tri-color display
                          cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
                          rst_pin=rst, busy_pin=busy)

display.rotation = 1

# clear the buffer
print("Clear buffer")
display.fill(Adafruit_EPD.WHITE)
display.pixel(10, 100, Adafruit_EPD.BLACK)

print("Draw Rectangles")
display.fill_rect(5, 5, 10, 10, Adafruit_EPD.RED)
display.rect(0, 0, 20, 30, Adafruit_EPD.BLACK)

print("Draw lines")
display.line(0, 0, display.width-1, display.height-1, Adafruit_EPD.BLACK)
display.line(0, display.height-1, display.width-1, 0, Adafruit_EPD.RED)

print("Draw text")
display.text('hello world', 25, 10, Adafruit_EPD.BLACK)
display.display()
```

### Behavior

Below is a copy of the cody.py output:

code.py output:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
ValueError: SCK in use

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

I have read and understood "SCK in use error on feather m4 express #1088" and "nrf: implement checking for pin in use #1089", which are referenced below. As far as I know, only 1 spi pin is in use. 

I apologize if this isn't the correct avenue for discussion. I understand if I am redirected to a more suitable channel. 

https://github.com/adafruit/circuitpython/issues/1088
https://github.com/adafruit/circuitpython/issues/1089

### Additional information

Hardware:

Adafruit Feather M4 Express with samd51j19
Board ID: feather_m4_express 

2.9" ThinkInk Greyscale EPD Display

ladyada, 2022-04-05:
please try very latest version of cpy, in case something got fixed recently

Neradoc, 2022-04-05:
This is normal if it happens on reload. The SPI is retained by displayio when you setup a display, so that it can be used to show the REPL when the code ends or reloads. You need to execute this before initing the SPI line.
```py
displayio.release_displays()
```
See example here:
https://learn.adafruit.com/adafruit-2-9-eink-display-breakouts-and-featherwings/circuitpython-usage#grayscale-display-usage-3102538-29

(A better place for support issues is [the forums](https://forums.adafruit.com/) or [discord server](https://adafru.it/discord)).

ladyada, 2022-04-05:
@Neradoc i dont think this is the displayio driver, its the framebuf one

Neradoc, 2022-04-05:
Ah my bad I misread that.

LTB2022, 2022-04-06:
> ```python
> import digitalio
> import busio
> import board
> from adafruit_epd.epd import Adafruit_EPD
> from adafruit_epd.il0373 import Adafruit_IL0373
>
> # create the spi device and pins we will need
> spi = [...]

@ladyada Thanks for your reply, I have updated to the most recent CircuitPython, Adafruit CircuitPython 7.2.4. Now the code runs, but there is no output on the screen. I just ran the program. Here is the exact output:

"Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Creating display
Clear buffer
Draw Rectangles
Draw lines
Draw text

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload."

I can display images on this device, so I am sure that it is functional.

ladyada, 2022-04-06:
ok @makermelissa can try to replicate

makermelissa, 2022-04-07:
I'm getting some output when I use the correct pins( see https://learn.adafruit.com/adafruit-2-9-eink-display-breakouts-and-featherwings/pinouts) with this script:
```
import time
import digitalio
import busio
import board
from adafruit_epd.epd import Adafruit_EPD
from adafruit_epd.il0373 import Adafruit_IL0373

# create the spi device and pins we will need
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
ecs = digitalio.DigitalInOut(board.D9)
dc = digitalio.DigitalInOut(board.D10)
srcs = digitalio.DigitalInOut(board.D6)    # can be None to use internal memory
rst = None    # can be None to not use this pin
busy = None   # can be None to not use this pin

# give them all to our driver
print("Creating display")
# Using the 2.9" ThinkInk Greyscale EPD Display (296X128)
display = Adafruit_IL0373(296, 128, spi,          # 2.13" Tri-color display
                          cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
                          rst_pin=rst, busy_pin=busy)

display.rotation = 1

# clear the buffer
print("Clear buffer")
display.fill(Adafruit_EPD.WHITE)
display.pixel(10, 100, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)

print("Draw Rectangles")
display.fill_rect(5, 5, 10, 10, Adafruit_EPD.RED)
display.rect(0, 0, 20, 30, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)

print("Draw lines")
display.line(0, 0, display.width-1, display.height-1, Adafruit_EPD.BLACK)
display.line(0, display.height-1, display.width-1, 0, Adafruit_EPD.RED)
display.display()
time.sleep(15)

print("Draw text")
display.text('hello world', 25, 10, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)
```
However, I don't think we ever updated the library to work with the grayscale display because it was a FeatherWing and we were really only using this library for the Raspberry Pi.

makermelissa, 2022-04-07:
Oof, I don't know why I thought you were using grayscale. Let me try with a tri-color...

makermelissa, 2022-04-07:
Ok, tri-color is sorta working. I'll try messing with it a bit and get back to you.

makermelissa, 2022-04-07:
Ok, the width and height were reversed in your example too. Try this:
```
import time
import digitalio
import board
from adafruit_epd.epd import Adafruit_EPD
from adafruit_epd.il0373 import Adafruit_IL0373

# create the spi device and pins we will need
spi = board.SPI()
ecs = digitalio.DigitalInOut(board.D9)
dc = digitalio.DigitalInOut(board.D10)
srcs = digitalio.DigitalInOut(board.D6)    # can be None to use internal memory
rst = None    # can be None to not use this pin
busy = None   # can be None to not use this pin

# give them all to our driver
print("Creating display")
# Using the 2.9" ThinkInk Greyscale EPD Display (296X128)
display = Adafruit_IL0373(128, 296, spi,          # 2.13" Tri-color display
                          cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
                          rst_pin=rst, busy_pin=busy)

display.rotation = 1

# clear the buffer
print("Clear buffer")
display.fill(Adafruit_EPD.WHITE)
display.pixel(10, 100, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)

print("Draw Rectangles")
display.fill_rect(5, 5, 10, 10, Adafruit_EPD.RED)
display.rect(0, 0, 20, 30, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)

print("Draw lines")
display.line(0, 0, display.width-1, display.height-1, Adafruit_EPD.BLACK)
display.line(0, display.height-1, display.width-1, 0, Adafruit_EPD.RED)
display.display()
time.sleep(15)

print("Draw text")
display.text('hello world', 25, 10, Adafruit_EPD.BLACK)
display.display()
time.sleep(15)
```

makermelissa, 2022-04-07:
Here's a version that draws it all at once and I put back in the board.SCK:
```
import time
import digitalio
import busio
import board
from adafruit_epd.epd import Adafruit_EPD
from adafruit_epd.il0373 import Adafruit_IL0373

# create the spi device and pins we will need
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
ecs = digitalio.DigitalInOut(board.D9)
dc = digitalio.DigitalInOut(board.D10)
srcs = digitalio.DigitalInOut(board.D6)    # can be None to use internal memory
rst = None    # can be None to not use this pin
busy = None   # can be None to not use this pin

# give them all to our driver
print("Creating display")
# Using the 2.9" ThinkInk Greyscale EPD Display (296X128)
display = Adafruit_IL0373(128, 296, spi,          # 2.13" Tri-color display
                          cs_pin=ecs, dc_pin=dc, sramcs_pin=srcs,
                          rst_pin=rst, busy_pin=busy)

display.rotation = 1

# clear the buffer
print("Clear buffer")
display.fill(Adafruit_EPD.WHITE)
display.pixel(10, 100, Adafruit_EPD.BLACK)

print("Draw Rectangles")
display.fill_rect(5, 5, 10, 10, Adafruit_EPD.RED)
display.rect(0, 0, 20, 30, Adafruit_EPD.BLACK)

print("Draw lines")
display.line(0, 0, display.width-1, display.height-1, Adafruit_EPD.BLACK)
display.line(0, display.height-1, display.width-1, 0, Adafruit_EPD.RED)

print("Draw text")
display.text('hello world', 25, 10, Adafruit_EPD.BLACK)
display.display()
```

makermelissa, 2022-04-07:
I used CircuitPython 7.2.5 and the original issue appears to be fixed.

closed 2022-04-07 by makermelissa

LTB2022, 2022-04-08:
> Oof, I don't know why I thought you were using grayscale. Let me try with a tri-color...

You thought right, I was using greyscale! Wasn't sure how to adapt the code to my device.

LTB2022, 2022-04-08:
> Ok, the width and height were reversed in your example too. Try this:
>
> ```
> import time
> import digitalio
> import board
> from adafruit_epd.epd import Adafruit_EPD
> from adafruit_epd.il0373 import [...]

I see, I was using the class as a guide in the il0373.py file, so I probably just didn't understand where this was coming from. Thanks!
[image]

LTB2022, 2022-04-08:
Thanks so much for your help, works great now!