adafruit/circuitpython #6254: ESP32-S3 issues with uarts - data output corrupt

KurtE, 2022-04-06:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
```

### Code/REPL

```python
>>> import board, busio
>>> uart1 = board.UART()
>>> uart1.deinit()
>>> uart1 = busio.UART(board.TX, board.RX, baudrate=38400)
>>> uart2 = busio.UART(board.IO17, board.IO18, baudrate=38400)
>>> uart1.write(b'abcdefghijklmnopqrstuvwxyz')
26
>>> uart2.write(b'abcdefghijklmnopqrstuvwxyz')
26
>>>
```

### Behavior

Wrong data output,  looks like maybe a queue issue:  Here is what was seen on the Logic Analyzer

For uart1:
[image]

For uart2:
[image]

### Additional information

Note: I also tried this with baud of 1000000 and also the data output was not correct.

anecdata, 2022-04-06:
Can confirm quirky behavior on ESP32-S3. I've been running this on two separate N8R8 devices (one transmitting, one receiving) for several weeks:
`uart = busio.UART(board.IO6, board.IO7, baudrate=921600, parity=busio.UART.Parity.ODD, stop=1, receiver_buffer_size=8192)`
Data chunk sizes are variable, but typically several hundred bytes. It will work properly for long stretches, then gets erroneous bytes typically at the beginning of each data chunk (resulting in erroneous CRC). Resetting the sender fixes the problem again temporarily.

KurtE, 2022-04-09:
I was curious about this on an ESP32-S2 as the release notes say that these are stable whereas the S3 are considered Alpha...
So I purchased one from Amazon, which I think is the same as the Adafruit one:
https://www.adafruit.com/product/4693
Would have ordered that one, but .. (wish you would ship to my PMB in town)

I setup a Frankenstein setup:
[image]
Darn thing won't fit in normal breadboard and have row of pins on both sides...

Adafruit CircuitPython 7.2.5 on 2022-04-06; Saola 1 w/Wroom with ESP32S2

Hacked up version of the code,  I was not getting anything back from UART1, so have it output a string to each every cycle and then check for data back.

```python
import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

import neopixel

led_colors = [
    (255, 0, 0),
    (0, 255, 0),
    (0, 0, 255),
    (255, 255, 0),
    (255, 0, 255),
    (0, 255, 255),
    (255, 255, 255),
    (0, 0, 0),
]
color_index = 0
check_pin = DigitalInOut(board.IO42)

check_pin.direction = Direction.INPUT
check_pin.pull = Pull.DOWN
uart1 = None
uart2 = None

led = neopixel.NeoPixel(board.NEOPIXEL, 1)

led.brightness = 0.3

last_check_value = check_pin.value

def CheckAndEchoUart(index, uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        data = uart.read(waiting_count)
        print(">> Uart", index, " input(", waiting_count, "): ", data.decode())
        '''
        if (index == 1):
            print("\t len:",len(data), "data:", data.decode())
        else:
            uart2.write(data)
        '''
        
while True:
    if check_pin.value:
        if uart1 == None:
            print("*** initialize the uarts ***")
            # we have not yet tried to iniatialize the uarts
            #uart1 = board.UART()
            #uart1.deinit()
            uart1 = busio.UART(board.TX, board.RX, baudrate=1000000)
            uart2 = busio.UART(board.IO17, board.IO16, baudrate=1000000)
        uart1.write(b'uart1')
        uart2.write(b'uart2')
        '''
        if supervisor.runtime.serial_bytes_available:
            text = input()
            print("*** Serial input ***", text)
            b = bytearray()
            b.extend(text)
            uart1.write(b)
        '''        
        CheckAndEchoUart(1, uart1)
        CheckAndEchoUart(2, uart2)
   # cycle colors
    led[0] = led_colors[color_index]
    color_index += 1
    if color_index >= len(led_colors):
        color_index = 0
        
    time.sleep(0.5)
```
I have a jumper from RX to TX and one from IO16 to IO17, and: 
I am not getting any data back on RX and similar garbage on IO16 like S3...
```
>> Uart 2  input( 5 ):  art2u
>> Uart 2  input( 5 ):  art2u
>> Uart 2  input( 5 ):  art2u
>> Uart 2  input( 5 ):  art2u
>> Uart 2  input( 5 ):  art2u
```

Also secondary note, it does not appear like the UARTs output concurrently, but instead maybe complete their output before returning?  Also good size time gap between the two outputs:
[image]

dhalbert, 2022-04-09:
`busio.UART()` is synchronous. It waits for the sending to complete before returning.

@KurtE Could you open separate issues when you are encountering different problems? It helps us keep track of them separately, since they may have different fixes.

ThomasAtBBTF, 2022-05-21:
Observing UART problems also with a FeatherS3.
If connecting RX to TX and sending bytes to the UART data is coming with an offset.
As if the queue is not working correctly. Initially, after a reset, the send characters are received 2 or three characters "later"
With every reload of the code this gap increases. 
Also, additional UART's created on other Pins than TX and RX are instantiated with no problems but do not work at all.

futari-media, 2022-05-21:
Also observing the UART offset on a FeatherS3 here.

ThomasAtBBTF, 2022-05-25:
@dhalbert 
What does 8.0.0 mean?
Will there be 7.4 / 7.5 / 7.6... versions to be expected?
And, do you really think that the queue problems observed in CircuitPython test code are rooted in the ESP-IDF?
I have a hard time believing that such a basic feature like UART communication is not working well in the ESP-IDF.

dhalbert, 2022-05-25:
> What does 8.0.0 mean?

We use the "semver" versioning model. 8 is a major version, and means there are incompatible changes in API or (significant) behavior changes.

> Will there be 7.4 / 7.5 / 7.6... versions to be expected?

There will be more 7.3.x versions. The  rightmost `x` indicates bug fixes. If we decide to add significant features to 7.x.x, we would go to 7.4.x, etc. Such a decision might depend on how long we expect it to take to get to 8.0.0 final. Our attitude is that integers are free :) .

> And, do you really think that the queue problems observed in CircuitPython test code are rooted in the ESP-IDF? I have a hard time believing that such a basic feature like UART communication is not working well in the ESP-IDF.

There are definitely ESP-IDF I2C problems. One problem was already fixed, and there is an S3 fix in process. I don't think I said we know of ESP-IDF UART problems; if I did, I apologize. I2C is more complicated. The S2 problem had to do with another FreeRTOS task running, and the still-unifixed problem seems to have to do with handling clock-stretching properly.

KurtE, 2022-05-25:
@dhalbert (@ThomasAtBBTF ) - Sounds like, lots of stuff to keep you busy!

Simple point of reference is that simple UART stuff is working on CircuitPython... 
Some simple testing.
```
MPY: soft reboot
MicroPython v1.18 on 2022-01-17; FeatherS2 with ESP32-S2
Type "help()" for more information.
>>> import machine
>>> uart1 = machine.UART(1, baudrate=9600, tx=43, rx=44)
>>> uart1.write("ABCDEFGHIJKLMNOPQRSTUVWXYZ")
26
>>> print(uart1.read())
b'ABCDEFGHIJKLMNOPQRSTUVWXYZ'
>>>

```

As I mentioned on ESP32-S3... I tried similar for uart2, uart1 acts stranger... 
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.0.0-alpha.0-12-gecd30d233 on 2022-05-24; ESP32-S3-DevKitC-1-N8R8 with ESP32S3
>>> import board, busio
>>> uart2 = busio.UART(board.IO17, board.IO18, baudrate=115200)
>>> uart2.write(b'1234567890')
10
>>> uart2.in_waiting
11
>>> print(uart2.readline())
b'\x00\x00\x00\x00\x00\x00\x00\x00H67'
>>> 
```
And I have jumper between the two pins

I will try to take a quick look and see if there is anything obvious between the uart.write() and where it actually outputs to see if anything looks obvious.

ThomasAtBBTF, 2022-05-25:
@KurtE 
Thank you so much for testing this!
Do you have any idea on which ESP-IDF version the Micropython is working?
And on which version we (CircuitPython) is working?

Also, even though I have very little understanding of the inner workings of the CircuitPython UART code, I am pretty sure, that the bug is located in the "receiving code" inside CircuitPython and not in the "sending code".

Reason: I did tests also with another board connected to the UART-pins and the CircuitPython ESP32-S3 code on the receiving side did show the same bug. When the ESP32-S3 code was sending data to the other board all characters arrived "well" on the "other side".

KurtE, 2022-05-25:
@ThomasAtBBTF 
I am not sure which one they are using right now:

Their instructions for build it show:

```
The ESP-IDF changes quickly and MicroPython only supports certain versions.
Currently MicroPython supports v4.0.2, v4.1.1, v4.2.2, v4.3.2 and v4.4,
although other IDF v4 versions may also work.
```

And yes it appears like TX may be working:
```
>>> import board, busio
>>> uart2 = busio.UART(board.IO17, board.IO18, baudrate=115200)
>>> uart2.write(b'1234567890')
10
>>> 

```![image](https://user-images.githubusercontent.com/1558080/170320619-7a4cf168-bb7f-4add-95fb-8c64e2d44785.png)

dhalbert, 2022-05-25:
If any of you see these UART problems on ESP32-S3, but not ESP-S2, that would be interesting.

ThomasAtBBTF, 2022-05-25:
I read the message from Kurt so that he does not see the problem on **MicroPython** on a S2
But on **CircuitPython** on a S3.
I will test with a ESP32-S2 board tomorrow with CircuitPython 7.3

futari-media, 2022-05-25:
> If any of you see these UART problems on ESP32-S3, but not ESP-S2, that would be interesting.

I see the problem on a QT PY ESP32 S2 as receiver and Feather RP2040 as transmitter. Bytes are shifted/swapped on the S2. Stopping and Restarting the S2 in the repl causes a unique shift. Stopping and starting the Feather RP2040 in the repl has the same shift offset in the QT PY ESP32 S2.
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit QT Py ESP32S2 with ESP32S2

KurtE, 2022-05-26:
> If any of you see these UART problems on ESP32-S3, but not ESP-S2, that would be interesting.

I have seen it on both as I mentioned back earlier in this issue (April 9th) 

Was hooking up again with a newer S2, that I see for sale on Adafruit: 
https://www.adafruit.com/product/4769

And tried for a long time to reinstall CPy...  But I screwed up that I did not realize that this Feather S2 is not the same as the Feather S2.   Only when I unplugged it from a breadboard did I see that it was unexpected maker...

And does not show up in the list: https://circuitpython.org/downloads?q=ESP32-S2
But the Adafruit product page does take you to:  [**https://circuitpython.org/downloads?q=ESP32-S2**](https://circuitpython.org/board/unexpectedmaker_feathers2/)

Did not find that at first so built my own copy and now have it running... Now to try with it...

simple version of sketch:
```
import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

uart1 = board.UART()
uart1.deinit()
uart1 = busio.UART(board.TX, board.RX, baudrate=115200)
uart2 = busio.UART(board.IO17, board.IO18, baudrate=115200)

while True:
    text = input()
    text1_encoded = text.encode();
    ba = bytearray(text1_encoded)
    print("\tBA: ", ba)
    uart1.write(ba);
    text1 = uart2.read();
    uart2.write(text1)
    print("\tt1:", text1)
    text2 = uart1.read();
    print(text2)
```

Yes sort of garbage, I have the two uarts jumpers so  TX->17, 18 -> RX
And some sample output:
'''
code.py output:
123456789abcdefgh
    BA:  bytearray(b'123456789abcdefgh')
    t1: b'\x00\x00123456789abcdefg'
b'\x00\x00\x00\x00\x00123456789abcde'
```
Yes I know that this code is nothing special, but shows how the data gets messed up.  Also the alignment and how much of the data that gets messed up is not consistent.

dhalbert, 2022-05-26:
@KurtE Just to make sure:

```py
uart1 = busio.UART(board.TX, board.RX, baudrate=115200)
uart2 = busio.UART(board.IO17, board.IO18, baudrate=115200)
```

>I have the two uarts jumpers so TX->17, 18 -> RX

That would be TX->TX and RX->RX. Did you swap them around? I'd expect TX->RX and RX->TX.

KurtE, 2022-05-26:
Sorry typo’s yes to to rx…

KurtE, 2022-05-26:
@dhalbert and all - quick update,  Found the simple example I posted, which is now up at:
https://github.com/KurtE/circuitPython_sketches/blob/main/ESP32%20Uarts/simple_version.py
replicates the issue a lot more than my more advanced option which is up there as well:
https://github.com/KurtE/circuitPython_sketches/blob/main/ESP32%20Uarts/code.py

The differences mainly have to do with: In simple case I read as many bytes as possible until timeout:
```
        text = input()
        text1_encoded = text.encode();
        ba = bytearray(text1_encoded)
        print("\tBA: ", ba)
        uart1.write(ba);
        text1 = uart2.read();
        uart2.write(text1)
        print("\tText1: ", text1)
        text2 = uart1.read();
        print("\ttext2: ", text2)
```
where the more advanced version of the code: 
For each uart it does:
```
def CheckAndEchoUart(index, uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        print("*** Uart", index, " input(", waiting_count, ") ***")
        data = uart.read(waiting_count)

        if (index == 0):
            print("Echoed back len:",len(data), "data:", data.decode())
        else:
            print(index,">>", data.decode())
            uart.write(data)
```

So it only tries to read however many bytes that are currently available and then write those out.  And with a bunch of debug code in place appeared to echo the right stuff.

This appears to then point to different code paths in ports\espressif\esp-idf\components\drivers\uart.c
in the function: uart_read_bytes

I added a bunch of debug prints to this like:
```
int uart_read_bytes(uart_port_t uart_num, void *buf, uint32_t length, TickType_t ticks_to_wait)
{
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), (-1), UART_TAG, "uart_num error");
    ESP_RETURN_ON_FALSE((buf), (-1), UART_TAG, "uart data null");
    ESP_RETURN_ON_FALSE((p_uart_obj[uart_num]), (-1), UART_TAG, "uart driver error");
    mp_printf(&mp_plat_print,"!!!!uart_read_bytes(%u %p %u)\n", uart_num, buf, length);

    uint8_t *data = NULL;
    size_t size;
    size_t copy_len = 0;
    int len_tmp;
    if (xSemaphoreTake(p_uart_obj[uart_num]->rx_mux, (portTickType)ticks_to_wait) != pdTRUE) {
        return -1;
    }
    while (length) {
        if (p_uart_obj[uart_num]->rx_cur_remain == 0) {
            data = (uint8_t *) xRingbufferReceive(p_uart_obj[uart_num]->rx_ring_buf, &size, (portTickType) ticks_to_wait);
            mp_printf(&mp_plat_print,"\trx_cur_remain == 0: %u %p %u\n", uart_num, data, size);

            if (data) {
                for (size_t i = 0; i < size; i++) mp_printf(&mp_plat_print," %02x", data[i]);
                mp_printf(&mp_plat_print,"\n");
                p_uart_obj[uart_num]->rx_head_ptr = data;
...
```
Need to see how to get when I change this function for it to update the generated uf2 file...  (side story)... But with the one that only reads the number of bytes that it currently has, a lot of these prints don't output...  But in the simple case it goes into hard loop printing and wipes out the Mu terminal... (Or kitty for that matter)   Now to see if I can reduce this output

KurtE, 2022-05-28:
For now I think I will leave this for someone who actually uses and knows the ESP32 hardware/software to debug and update the code to work. 

I sort of traced the code, to where I think the data comes in on an ISR, and the handler than calls through a few layers of code to read the stuff from the ESP32, into a buffer.  It then returns and then this buffer is copied into a ring buffer.  When the data is requested, the code then calls off again through layers to copy all or part of the data from the ring buffer to another buffer, and then this buffer is copied to the user buffer.  

So far I have not drilled debug code down far enough to see where the data become corrupted.  
It could be any number of things like:

a) requesting more stuff from the hardware RX fifo than there is in it.
b) Maybe same buffer used for two different things
c) Maybe an issue with the objects values not properly protected from updates elsewhere.  or maybe not marked volatile when should be.
d) ...

I thought I might get some hints looking at the MicroPython code base, but this part is very much different. 
Like I believe their ISR for RX is very simple:
```
STATIC void IRAM_ATTR uart_irq_handler(void *arg) {
    volatile uart_dev_t *uart = &UART0;
    #if CONFIG_IDF_TARGET_ESP32S3
    uart->int_clr.rxfifo_full_int_clr = 1;
    uart->int_clr.rxfifo_tout_int_clr = 1;
    #else
    uart->int_clr.rxfifo_full = 1;
    uart->int_clr.rxfifo_tout = 1;
    uart->int_clr.frm_err = 1;
    #endif
    while (uart->status.rxfifo_cnt) {
        #if CONFIG_IDF_TARGET_ESP32
        uint8_t c = uart->fifo.rw_byte;
        #elif CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
        uint8_t c = READ_PERI_REG(UART_FIFO_AHB_REG(0)); // UART0
        #endif
        if (c == mp_interrupt_char) {
            mp_sched_keyboard_interrupt();
        } else {
            // this is an inline function so will be in IRAM
            ringbuf_put(&stdin_ringbuf, c);
        }
    }
}
```

Again sorry I have not so far been able to resolve it.

KurtE, 2022-05-30:
@dhalbert @ThomasAtBBTF @futari-media  - I was playing around instrumenting more of the code like setting/clearing IO pins around the ISR and the like, but every time I would start my sketches I am getting a little garbage on the input of each UART...

So I added a call to reset the input like:
```
uart2 = busio.UART(board.IO17, board.IO18, baudrate=115200)
uart2.timeout = 0.1
uart2.reset_input_buffer()
```
Which cleared that up, and it also so far has the current test sketch working better. 

If you get a chance you might try adding the reset to your sketch and see if it helps

anecdata, 2022-05-30:
fwiw, for over a month I've had three ESP32-S3 UART senders sending to one UART receiver ESP32-S3, with the following (pretty aggressive) config:
```
uart[0] = busio.UART(board.IO6, board.IO7, baudrate=921600, stop=1, timeout=1, receiver_buffer_size=65535)
uart[1] = busio.UART(board.IO16, board.IO17, baudrate=921600, stop=1, timeout=1, receiver_buffer_size=65535)
uart[2] = busio.UART(board.IO10, board.IO11, baudrate=921600, stop=1, timeout=1, receiver_buffer_size=65535)
```
The code does `reset_input_buffer()` before the main loop. Each sender is sending about 20 packets per second, variable packet length typically a couple hundred bytes, and a CRC is sent along with each packet. With three senders, I see about a 2% CRC failure rate, less failure with two or just one sender. It does seem to be the sender that occasionally messes up, so I have the receiver connected to the reset on each sender and trigger a hard reset of the sender if there is a consecutive sequence of bad CRCs or missing packets. It varies, but perhaps every few hours on average, one of the senders needs to be reset due to a string of bad CRCs or a string of missing packets typically due to a Safe Mode.

ThomasAtBBTF, 2022-05-30:
@anecdata 
Thank you for sharing your observations and thank you for mentioning `reset_input_buffer()` !
I did not have this in my test-code and I can say, that calling `reset_input_buffer()` certainly changes the behavior of the UARTs.

This code:
`uart1 = busio.UART(board.TX, board.RX, timeout=0.01)
uart2 = busio.UART(board.D13, board.D12, timeout=0.01)
uart3 = busio.UART(board.D11, board.D10, timeout=0.01)
uart1.reset_input_buffer()
uart2.reset_input_buffer()
uart3.reset_input_buffer()
print("reset_input_buffer done")
count = 0
while True:
    if supervisor.runtime.serial_bytes_available:
        k = sys.stdin.read(1)
        if ord(k) == 27:
            break
        elif k == "1":
            count += 1
            s = f"hi uart1! {count}".encode()
            uart1.write(s)
            print("TX :", s)
        elif k == "2":
            count += 1
            s = f"hi uart2! {count}".encode()
            uart2.write(s)
            print(" "*25 + "TX :", s)
        elif k == "3":
            count += 1
            s = f"hi uart3! {count}".encode()
            uart3.write(s)
            print(" "*50 + "TX :", s)

    r = uart1.read()
    if r:
        print("RX :", r)
    r = uart2.read()
    if r:
        print(" "*25+"RX :", r)
    r = uart3.read()
    if r:
        print(" "*50+"RX :", r)
print("done")
`
works now ok after a reset (not ctrl-c!) for UART1 and UART2 but not for UART3.

After a ctrl-c and ctrl-d in the REPL all is still messed up.

Now it looks as if the problem is on the sender side!

If I connect RX and TX of UART1 to an external terminal the characters sent by the external terminal are always received and printed correctly on the REPL.
But the characters sent by UART1 of the ESP32-S3 are received with an offset by the external terminal.
By "received with an offset" I mean, that for every character send by UART1 of the ESP32-S3 a character is received by the external terminal. But not the character which has been just sent but the character which was sent two or more transmissions before.

ThomasAtBBTF, 2022-05-30:
@KurtE  It seems to me that you should look at the ESP32-S3 transmitter side and not the receiver side.

KurtE, 2022-05-30:
@ThomasAtBBTF and @anecdata  thanks for the inputs.  I am not sure how much farther I will take this.  

I personally don't really use ESP32s.   I was mainly trying them out to see how they behave.  In particular I was debugging some stuff with the Teensy 4.x ports to CircuitPython.  So I tried out some test sketches on a few different processor chips to compare, including the RP2040 and then ESP32-S3. 

I reported the issue on the S3, but it was mentioned that CircuitPython on Teensy and ESP32-S3 were still in the Alpha or Beta state, but the S2 was in a more mature state... So picked up a couple different S2s and found they exhibit the same issue. 

Hopefully someone who more routinely works with ESP32 hardware and software will be able to resolve the issue(s).

I may still play with this off and on, but don't have high hopes.

Kurt

KurtE, 2022-05-30:
> @KurtE It seems to me that you should look at the ESP32-S3 transmitter side and not the receiver side.

You are right that the TX is screwed up... 
Sketch:
```
import time
import board
import busio
import supervisor

uarts = []
board.UART().deinit()
uarts.append( busio.UART(board.IO6, board.IO7, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))
uarts.append(busio.UART(board.IO16, board.IO17, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))
uarts.append(busio.UART(board.IO10, board.IO11, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))

for uart in uarts:
    uart.reset_input_buffer()

while True:
    
    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("*** Serial input ***", text)
        b = bytearray()
        b.extend(text)
        #uart1.write(b)
        for uart in uarts:
            uart.write(b)
    time.sleep(0.5)
```
REPL window
```
code.py output:
abcdefghijklmnopqrstuvwxyz
*** Serial input *** abcdefghijklmnopqrstuvwxyz
```
Logic Analyzer:
[image]

KurtE, 2022-05-31:
@dhalbert @ThomasAtBBTF @futari-media  - From the hints from the last few posts, I think I may have found something that is making it work better,  but not sure what to do with it.   Also would need more testing. 

And then there is what to do to make it work, versus someone to properly do it. 

Theory: the hardware FIFO queues were not properly setup and working right, especially after soft reboot... 

So what I did that appears to resolve the issue with TX being screwed up and guessing RX as well  was to hack on to:
https://github.com/adafruit/esp-idf/blob/circuitpython-v4.4/components/driver/uart.c#L286

And have the setting the baud rate reset both queues like:
```
esp_err_t uart_set_baudrate(uart_port_t uart_num, uint32_t baud_rate)
{
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    uart_hal_set_baudrate(&(uart_context[uart_num].hal), baud_rate);
    
    uart_hal_txfifo_rst(&(uart_context[uart_num].hal));
    uart_hal_rxfifo_rst(&(uart_context[uart_num].hal));

    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    return ESP_OK;
}

```

But this is in the IDF project... So not sure about if/how to update here. 

Other possibility would be to see if I could somehow call these methods from the higher level like:
from the CPY/ports/espressif/common-hal/busio/UART.c
But these take the HAL context... which I am not sure this level should play with.

Probably the proper fix would be to change: void common_hal_busio_uart_construct(busio_uart_obj_t *self,
to not call all of the individual setup functions within the idf, but instead call:

```
esp_err_t uart_param_config(uart_port_t uart_num, const uart_config_t *uart_config)
{
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
    ESP_RETURN_ON_FALSE((uart_config), ESP_FAIL, UART_TAG, "param null");
    ESP_RETURN_ON_FALSE((uart_config->rx_flow_ctrl_thresh < SOC_UART_FIFO_LEN), ESP_FAIL, UART_TAG, "rx flow thresh error");
    ESP_RETURN_ON_FALSE((uart_config->flow_ctrl < UART_HW_FLOWCTRL_MAX), ESP_FAIL, UART_TAG, "hw_flowctrl mode error");
    ESP_RETURN_ON_FALSE((uart_config->data_bits < UART_DATA_BITS_MAX), ESP_FAIL, UART_TAG, "data bit error");
    uart_module_enable(uart_num);
#if SOC_UART_SUPPORT_RTC_CLK
    if (uart_config->source_clk == UART_SCLK_RTC) {
        rtc_clk_enable(uart_num);
    }
#endif
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    uart_hal_init(&(uart_context[uart_num].hal), uart_num);
    uart_hal_set_sclk(&(uart_context[uart_num].hal), uart_config->source_clk);
    uart_hal_set_baudrate(&(uart_context[uart_num].hal), uart_config->baud_rate);
    uart_hal_set_parity(&(uart_context[uart_num].hal), uart_config->parity);
    uart_hal_set_data_bit_num(&(uart_context[uart_num].hal), uart_config->data_bits);
    uart_hal_set_stop_bits(&(uart_context[uart_num].hal), uart_config->stop_bits);
    uart_hal_set_tx_idle_num(&(uart_context[uart_num].hal), UART_TX_IDLE_NUM_DEFAULT);
    uart_hal_set_hw_flow_ctrl(&(uart_context[uart_num].hal), uart_config->flow_ctrl, uart_config->rx_flow_ctrl_thresh);
    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    uart_hal_rxfifo_rst(&(uart_context[uart_num].hal));
    uart_hal_txfifo_rst(&(uart_context[uart_num].hal));
    return ESP_OK;
}
which sets up all of the things like parity, stop bits flow control and clears the buffers. 

But that is probably beyond my pay grade ($0.00) ;) 

Thoughts?  In the mean time, I am going to try turning off my debug code and build this for both ESP32-S3 and S2 and see if my test sketches are happy or not. 

```

KurtE, 2022-05-31:
Appears to work on S3 and S2, although still found it good to reset the input buffer. 
Here is current sketch on S3:
```python
import time
import board
import busio
import supervisor
from digitalio import DigitalInOut, Direction, Pull

uarts = []
board.UART().deinit()
uarts.append( busio.UART(board.IO6, board.IO7, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))
uarts.append(busio.UART(board.IO16, board.IO17, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))
uarts.append(busio.UART(board.IO10, board.IO11, baudrate=115200, stop=1, timeout=1, receiver_buffer_size=255))
for uart in uarts:
    uart.reset_input_buffer()

def CheckAndEchoUart(uart):
    waiting_count = uart.in_waiting
    if waiting_count:
        index = uarts.index(uart)
        #print("*** Uart", index, " input(", waiting_count, ") ***")
        data = uart.read(waiting_count)
        print("\t(",index, ")>>",data, "(", data.decode(), ")") 
        if (uart == uarts[0]):
            print("\tReturned len:",len(data), "data:", data.decode())
        else:
            uart.write(data)

while True:

    if supervisor.runtime.serial_bytes_available:
        text = input()
        print("*** Serial input ***", text)
        b = bytearray()
        b.extend(text)
        uarts[0].write(b)
    
    for uart in uarts:
        CheckAndEchoUart(uart)
    CheckAndEchoUart(uarts[0])
    time.sleep(0.5)
```

And run after reboot...

```
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
AbCdEfg
*** Serial input *** AbCdEfg
	( 1 )>> b'AbCdEfg' ( AbCdEfg )
	( 2 )>> b'AbCdEfg' ( AbCdEfg )
	( 0 )>> b'AbCdEfg' ( AbCdEfg )
	Returned len: 7 data: AbCdEfg
Some More Text
*** Serial input *** Some More Text
	( 1 )>> b'Some More Text' ( Some More Text )
	( 2 )>> b'Some More Text' ( Some More Text )
	( 0 )>> b'Some More Text' ( Some More Text )
	Returned len: 14 data: Some More Text
```
And I have the TX of one UART connected to RX of the next and then wrap back to RX of first...

tannewt, 2022-05-31:
> Probably the proper fix would be to change: void common_hal_busio_uart_construct(busio_uart_obj_t *self,
> to not call all of the individual setup functions within the idf, but instead call:

This sounds right to me. Thanks for digging into this!

ThomasAtBBTF, 2022-06-01:
A function like `reset_output_buffer()` would be also an option, because we already have the same for input.

PR #6453 (2022-06-13): ESP32 USE uart_param_config
Use this function instead of several individual configuration functions
to configure such things as Baud rate, transfer size, stop bits,
parity...

This function also resets both the RX and TX Hardware Fifo
reset functions are called to setup the hardware.

This is a First attempt at what was mentioned in issue.

If it works:
resolves #6254 

My initial tests are working,  
Would be good if others tried this and see if other configurations of UARTS work with this change.
closed 2022-06-13 by tannewt