adafruit/circuitpython #6255: BNO080 reboots/hangs ESP32S3 when accessing it via I2C.

mjs513, 2022-04-06:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; ESP32-S3-DevKitC-1-N8 with ESP32S3
Board ID:espressif_esp32s3_devkitc_1_n8
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""CircuitPython I2C Device Address Scan"""
# If you run this and it seems to hang, try manually unlocking
# your I2C bus from the REPL with
#  >>> import board
#  >>> board.I2C().unlock()

import time
import board
import busio

# To use default I2C bus (most boards) scl/sda
i2c = busio.I2C(board.IO10, board.IO11)
i2c1 = busio.I2C(board.IO1, board.IO2)

def i2cportscan():
    print("Wire")
    # Lock the I2C device before we try to scan
    while not i2c.try_lock():
        pass
    # Print the addresses found once
    # print(
    #    "I2C addresses found:", [hex(device_address) for device_address in i2c.scan()]
    # )
    for device_address in i2c.scan():
        print("Device found at address", end=" ")
        print(hex(device_address), end=" (")
        printKnownChips(hex(device_address))
        print(end=")")
        print()
    # Unlock I2C now that we're done scanning.
    i2c.unlock()

def i2cport1scan():
    print("Wire1")
    # Lock the I2C device before we try to scan
    while not i2c1.try_lock():
        pass
    # Print the addresses found once
    for device_address in i2c1.scan():
        print("Device found at address", end=" ")
        print(hex(device_address), end=" (")
        printKnownChips(hex(device_address))
        print(end=")")
        print()
    # Unlock I2C now that we're done scanning.
    i2c1.unlock()

def printKnownChips(address):
    # print("passed address:", address)
    if address == "0x00":
        print("AS3935", end=" ")
        return
    if address == "0x01":
        print("AS3935", end=" ")
        return
[... 248 lines cut ...]
```

### Behavior

```Wire
Device found at address 0x3c (SSD1306,DigisparkOLED )
Device found at address 0x50 (EEPROM,FRAM )
Device found at address 0x62 (LidarLite,MCP4725,AtlasEzoORP )
Device found at address 0x6f (Qwiic Button )
Wire1
Device found at address 0x29 (TSL2561,VL6180,TSL2561,TSL2591,BNO055,CAP1188 )
Device found at address 0x68 (DS1307/3231,MPU6050/9050,MPU9250,ITG3200/3701,LSM9DS0,L3G4200D )

Code done running.```

When working but if I try to add a BNO080 (Adafruit or Sparkfun) it reboots the ESP32S3 and seems to hang.  No error messages

### Description

Using a I2c test board that we used for testing the Teens 4.1:
![Image_20220406_092212](https://user-images.githubusercontent.com/5366213/162048892-0842f96e-d092-44e8-b4d0-44b64573098c.jpeg)

I attached the ESP32S3.  All worked well until I tried to add the BNO080 to either I2c or I2c1 bus (Arduino verbiage Wire or Wire1).  When I add the BNO080 it reboots the ESP32S3 and seems to hang Mu edit and I have to power cycle to get it running again.  No error messages appear.

If I run the same script using CircuitPython on the T4.1 it picks up all devices including the BNO080.  I do have a QT PY on order to test it with that as well.

mjs513, 2022-04-07:
As a follow up believe it related to a clock stretching issue with the BNO080: https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/issues/33

Will try a i2c1 = bitbangio.I2C(board.I2C_SCL1, board.I2C_SDA1, timeout = 500)
next time I have it set up

jepler, 2022-04-11:
There are known problems with esp32-s3 in i2c as well, which we are tracking. But it's true, the BNO chips have some interesting I2C behavior as well.

mjs513, 2022-04-11:
Yeah.  Was doing some research this morning and seems to be a known problem.  Unfortunately also happening on the RP2040, Teensy Micromod and Teensy 4.1 as well.

Early on in the development of the Teensy 4 we had similar issues with the BNO055 and BNO080 and required a changed to I2C.  For ref here was the fix = at least for the teensy in i2c: https://github.com/PaulStoffregen/Wire/pull/16/commits/08d870fc130b741e4fdee38798c64a28949c853d  
Maybe a clue for you all to look at

dhalbert, 2022-07-06:
Will re-test after pending S3 I2C fixes are added to ESP-IDF v4.4.x.

dhalbert, 2022-11-29:
I retested with a BNO055 on Feather ESP32-S3 with the latest build, and it still fails. It works fine on ESP32-S2, and also with `bitbangio.I2C()`. This may have the same root cause as #6311.

dhalbert, 2022-11-29:
Pushing this forward to 8.x.x because I don't see an upstream fix in sight.

dhalbert, 2023-04-11:
The BNO0xx sensors are known to violate the I2C bus protocol under certain conditions. I did some testing of the BNO055 here: https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/115. I expect similar results with the BNO080. Perhaps we can make the library do some error recovery, but given that the BNO sensors themselves are flaky, I'm not sure we should characterize this as a bug that we expect the ESP-IDF to fix. The ultimate bug is in the BNO sensors themselves.

I will close this as "can't fix" in the core for now. If you disagree let me know. I will add some warnings to the BNO080 learn guide, and have opened this library issue: https://github.com/adafruit/Adafruit_CircuitPython_BNO08x/issues/41

closed 2023-04-11 by dhalbert

dhalbert, 2023-11-05:
I retested after we upgraded to ESP-IDF v5.1 in CircuitPython 9.0.0-alpha.2. BNO055 still does not work. This is not so surprising. I added warnings to the appropriate BNO guides a while ago.