adafruit/circuitpython #6257: No microcontroller.nvm on feather M4 Express

muranyia, 2022-04-06:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-alpha.0-38-g862210b3f-dirty on 2022-03-24; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
```

### Code/REPL

```python
import microcontroller
import sys

print(str(len(microcontroller.nvm)), " bytes of nvm available.")
```

### Behavior

```python
Traceback (most recent call last):
  File "code.py", line 12, in <module>
TypeError: object of type 'NoneType' has no len()
```

### Description

According to the [docs](https://docs.circuitpython.org/en/latest/shared-bindings/nvm/index.html) I should have access nvm on the Feather M4 Express but it doesn't seem to work. Only tried `len()` so far.

anecdata, 2022-04-06:
Possibly a build error... can you try the beta?```py
Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; Adafruit Feather M4 Express with samd51j19
>>> import microcontroller
>>> len(microcontroller.nvm)
8192
```

dhalbert, 2022-04-06:
Your version is7.3.0-alpha.0-38-g862210b3f-dirty, which is a base of a ways back (though nothing about `nvm` was changed that I know of, so see if you move up, if it helps, and also check your own `mpconfigboard.*`, etc.

muranyia, 2022-04-06:
When I build from the updated git sources, it still comes out 
```
Adafruit CircuitPython 7.3.0-alpha.0-38-g862210b3f-dirty on 2022-04-07; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
```
N.B. I was able to bull the bundle OK, but pulling circuitpython I get tons of `Could not access submodule`.

How do I build the beta?

dhalbert, 2022-04-06:
Are you following the instructions in https://learn.adafruit.com/building-circuitpython ? You need to do `make fetch-submodules` at the top level.

This kind of build help is best done in our discord, https://adafru.it/discord, in the #circuitpython-dev channel.

closed 2022-04-06 by dhalbert