adafruit/circuitpython #6258: PyLeap no longer auto-reloading when code is transferred after 7.1.1

kattni, 2022-04-06:
### CircuitPython version

```python
Adafruit CiruitPython 7.2.3, 7.2.4, 7.2.5, 7.2.1 for Circuit Playground Bluefruit
```

### Code/REPL

```python
All examples supported by PyLeap were exhibiting the same behavior.
```

### Behavior

The intended behavior in PyLeap is that you pair the device (CPB or CLUE only at the moment), you choose an example, you download it (the first time only), you transfer the code to the device (in this case, CPB), and it automatically reloads and begins running the new example.

What is happening is, you follow all of the beginning steps, up through a successful transfer, however, the device does not reload, and the code does not begin running. If I manually click reset at that point, the new code begins running (which confirms a successful transfer).

### Additional information

Please let me know if I can test anything for this, I'm set up to test PyLeap already.

kattni, 2022-04-06:
@TrevKnows I wanted to tag you here so you can follow it, if you like. I will keep you posted regardless.

Note - the original issue was filed on PyLeap: https://github.com/adafruit/PyLeap-iOS/issues/18

jepler, 2022-05-03:
@kattni Is it possible to be connected to the REPL when doing this?  if so, what does the REPL show in each version?

PR #6377 (2022-05-10): Fix BLE autoreload
The trigger was happening when reload was suspended and therefore
ignored. Now, manage reload in one place in the file. This may
trigger reloads on BLE command error but that should be rare.

Fixes #6258
commit 8d739099 (2022-05-10): Fix BLE autoreload

The trigger was happening when reload was suspended and therefore
ignored. Now, manage reload in one place in the file. This may
trigger reloads on BLE command error but that should be rare.

Fixes #6258
closed 2022-05-11 by dhalbert